MarlinFirmware/Marlin #9597: [FR] TMC2130 - enable and disable stealthchop via SPI

Lyr3x, 2018-02-11:
i didnt found a request or an issue like this, so bear with me if somebody asked this.

Is it possible to switch between spreadcycle and stealthchop mode via SPI (through M-Code an/or LCD)? It would be nice to have the ability to switch to the powerful spreadcycle mode when you want to print fast as hell.

teemuatlut, 2018-02-11:
You can adjust the hybrid thresholds to achieve the same thing.

Lyr3x, 2018-02-12:
You're right. I thought about this, because for me and i think others too the hybrid mode does not work well for now.

govido, 2018-02-25:
Hi,
i'm new to customizing marlin firmware and just added tmc2130 drivers to my printer.
Would it be an option to add something like this in the custom lcd menu in configuration_adv?

  #define USER_DESC_6 "Disable Silent Mode (Stealthchop)"
  #define USER_GCODE_6 "M913 X1 Y1 Z1 E1\nM203 X400 Y400 Z15 E30"
  
  #define USER_DESC_7 "Silent Mode (Stealthchop)"
  #define USER_GCODE_7 "M913 X50 Y50 Z3 E20\nM203 X50 Y50 Z3 E20"

I haven't tested this yet. Just an idea to enable and disable the hybrid thresholds and also setting the maximum movement speeds that the spreadcycle threshold doesn't get triggered.
I'm currently using a heavily modified Ramps 1.4 with 24V power supply and tmc2130 for all motors.
I haven't tested a lot of settings to fine tune. But I can confirm that stealthchop wirh higher speed changes causes layer shifts. I haven't had many problems with x50 and y50 as hybrid threshold.
Would be nice if someone with more marlin knowledge could give some feedback.

AnHardt, 2018-02-25:
In theory this could work.
In reality you have to replace one or two of the existing 1-5 scrips or extend here:
https://github.com/MarlinFirmware/Marlin/blob/ac3fab7a9680a880fd1b113be7e758c6dabb2a57/Marlin/ultralcd.cpp#L900-L947

teemuatlut, 2018-02-25:
Don't think there's any reason why it wouldn't work but you don't want to set the max FR to be exactly the same as the switching threshold. This can cause unwanted switching back and forth. Leave some headroom.

I'll be back doing new features after v1.1.9 release and when we don't have to worry about feature freeze or feature parity between branches anymore. An LCD menu is on the list among some other things.

boelle, 2019-02-20:
@Lyr3x did you get your issue solved?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

Lyr3x, 2020-01-13:
@boelle First of all this was a FR and not a issue to be fixed. Second: A nice LCD Menu is implemented now. So the request is closed now.

closed 2020-01-13 by Lyr3x