MarlinFirmware/Marlin #9598: [bugfix-1.1.x] The name of the printer and the "ready" message are not shown [Anet A8]

FreeLuck, 2018-02-11:
### Bug Report
- Description:  
With the Marlin version 1.1.8 it's ok, I decided to go to the bugfix 1.1.x for other bugs corrected later. But now with the bugfix the name of the printer and the "ready" message, with my Anet A8, are not displayed on the main screen.
Firmware loaded all default and same problem.

[Anet A8 default settings.zip](https://github.com/MarlinFirmware/Marlin/files/1714751/Anet.A8.default.settings.zip)

DaveAhrendt, 2018-02-11:
Same issue on a FolgerTech 2020.

sh4jonny, 2018-02-11:
confirm on 2.0.x branch

GMagician, 2018-02-15:
I think this has been solved, try updated bugfix branches

FreeLuck, 2018-02-18:
I now confirm with the bugfix 1.1.x the name is shown

closed 2018-02-19 by FreeLuck

Spoons, 2018-04-10:
Interesting, having this problem on my printer on the latest bugfix 1.1.x.