MarlinFirmware/Marlin #9602: Does G29 save mesh data to eeprom?

batata004, 2018-02-12:
Hi,

@Roxy-3D is helping me clear somethings out, and I came up with a question: till today I was using bilinear ABL with 16 points (4x4). Now I was recommended to do 100 points (10x10). So surely I want it to be saved to epprom.

I already enabled eeprom usage in my confugration.h file (uncommenting the line EEPROM_SETTINGS) cause I had it disabled. So I flashed my printer and executed the G29 command.

My question is this: does G29 automatically saves the mesh bilinear data to the eeprom or I have to execute some command after that to have it saved to eeprom so next time I turn on my printer it gets loaded and I dont need to use G29 command anymore?

One more thing: if I use the G29 with 3 points (ABL) method on next boot up, will this override the mesh saved in the eeprom OR it will compensate the previously saved mesh to the 3 points recently measured?

Roxy-3D, 2018-02-12:
Bi-Linear will save the current mesh when you do a M500.   You only get to save one mesh with Bi-Linear.

If you switch from Bi-Linear to 3-point leveling (and rebuild the firmware), you may lose the mesh.   I'm not sure about that.    With UBL...   The 3-point leveling can be applied to the current mesh.

batata004, 2018-02-12:
@Roxy-3D perfect! So using the G29 will not save the mesh, I will always need to execute M500 after G29. I didnt find this anywhere in the documentation nor in http://reprap.org/wiki/G-code

Anyway, your answer was very clear! :)

closed 2018-02-12 by batata004

thinkyhead, 2018-02-14:
@batata004 This will be a better reference: http://marlinfw.org/docs/gcode/G029-abl.html