MarlinFirmware/Marlin #9603: "stop print" loosing position

kradant, 2018-02-12:
### Bug Report

I thought this bug was already fixed (see set_current_position_from_planner() after stepper.quick_stop() #3939)

- Description: when using the "stop print" item from the menu (with attached LCD), the print stops immediately but the printer loses its position (blinking ? on the display) and additionaly displays a wrong position 

- Expected behaviour: when pressing "stop print" it should stop and not loose its position

- Actual behaviour: it is stopping but loosing its position. Actually if I am doing something like 
```
G1 X100
G1 X0
```
in the print program and if I stop the print between those 2 positions, the printer will loose its position and display that he is at X100 regardless of where he really is. 

- Steps to reproduce:
  - Start a print from the sd card.
  - select "stop print"
  - check if the printer lost its position

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1716379/Configuration.zip)

AnHardt, 2018-02-12:
This stop is an emergency stop. It is intended to stop any motion immediately.
The step signal stops suddenly. No deacceleration is done. You will lose steps her if you have been above jerk-speed. No fix is possible!

Other kinds of "stop" could:
* Stop to accept new moves. That would print until the buffers are empty.
* Finish the current move - then decelerate. Here we don't know where to go during the deceleration. And we have to ask if this move is segmented or not and if we want to continue the segmented move to its end or just the segment.
* Break the current move/segment and decelerate as much as possible to the end of the move. That will fail often. If only one step is left there is no chance to break enough in time.
 * Never move an axis faster than its jerk-speed. Will print extremely slow.

For a **sudden** stop there is no acceptable way to do it without losing steps.
For a **slow** stop just stop sending moves.

How about "pause"?

AnHardt, 2018-02-12:
Or maybe rewrite the planner buffer with max deceleration until we can stop savely.

This is almost the same problem as the unsolved complicated part of the power outage problem. Stopping fast in a defined state.

kradant, 2018-02-13:
@AnHardt Thank you for your reply. I didn't know this at all. I tried the pause button but it takes an eternity to stop. The thing is that I am using a Shapeoko and I write the Gcode by hand. And my code looks like this:
```
G1 X0 Y0
G1 X200
G1 Y5
G1 X0
```
So I am doing those long lines from one end of the brint bed to the other one. I suppose those commands are still in the buffer then if I pause? I really just need a way to pause / stop the print in the middle between two commands like G1 X0 and G1 X100. It does not have to stop immediately but within say a few mms. This must be possible! How can I achieve this? thank you very much

kradant, 2018-02-13:
If I want to achieve that the print pauses faster, do I have to adjust the buffer size that there is always just one command in the queue? Will this not affect the performance/smoothness when I am doing something like a circle.?

thinkyhead, 2018-02-14:
For a long while we've had code that does its best to recover the real position after a hard stop.

```cpp
void _lcd_hard_stop() {
  stepper.quick_stop();
  const screenFunc_t old_screen = currentScreen;
  currentScreen = _lcd_do_nothing;
  while (planner.movesplanned()) idle();
  currentScreen = old_screen;
  stepper.cleaning_buffer_counter = 0;
  set_current_from_steppers_for_axis(ALL_AXES);
  sync_plan_position();
  refresh_cmd_timeout();
}
```
It's really the best we can do.

kradant, 2018-02-22:
Wow thank you @thinkyhead ! In the meantime I just changed the behavior of the "stop print" function to just pause and adjusted my gcode, so that those long lines are separated into smaller lines. By the way I found it really hard, to know how to run a specific gcode when I stop the print. For example in some programs I have to deactivate the software endstops. Now if I am aborting the print, the endstops are still deactivated! So I finally found out how to automatically reactivate them in [https://github.com/MarlinFirmware/Marlin/issues/5982]. Thank you for your help!

closed 2018-02-22 by kradant

AnHardt, 2018-02-22:
A good candidate for the startup script - if you change that often.

Else, if you want a warranty for consistency - we have a reset (on/off) button. Pushed before any print, ...

Salmonel, 2018-08-26:
Very beginner programer, I need to know where to put the void quick_stop lines in Marlin? Which tab?

thinkyhead, 2018-08-30:
Fixed long, long ago @Salmonel. Get Marlin 1.1.9.