MarlinFirmware/Marlin #9604: Change number of points of the G29 with bilinear ABL

batata004, 2018-02-12:
Hi,

I configured my printer to use by default 10x10 (100 points) mesh with bilinear ABL. But, sometimes, between prints, I would like to use less points, should I flash again marlin or could I use something like this below?

G29 P5

I read in some pages that this P5 only works with LINEAR, not BILINEAR. Could anyone of you confirm me if I could set the number of points in the mesh size without flashing it again?

I COULD easily test this myself using my printer but it's not close to me, I am traveling and I will be back today night and I would like to know if it would work so I can start making more changes to my configuration.h file before I get home.

thinkyhead, 2018-02-19:
Correct. That is not an option with Bilinear at this time. The code could be modified to make this possible, but it will require a bit more SRAM and have a slight negative impact on performance when leveling is enabled.

closed 2018-02-19 by thinkyhead

derMalla, 2020-05-24:
I am thinking about to have an adaptive grid G29 before every print taking concern of the area where the print is supposed to be on bed to have a fixed rate of cm^2 and probe points for a print area. 

I modified the G29-code so that P adapts the steps from the command line but the Correction map does not switch to the desired number of probing points. Any hints which files must be modified in that configuration ?