MarlinFirmware/Marlin #9605: Anet A8 Marlin 1.1.x don't store settings

Max1963, 2018-02-12:
Switched from Anet A8 standard config to Marlin 1.1.x last release; compile and upload without problems; LCd at start show X, Y and Z with blinking question mark; set home all, caibrate bed, then store new value with both M500 (Cura-Pronterface) and LCD; once restarted the LCD show ALWAYS same blinking X, Y and Z.

ripp2003, 2018-02-12:
I think it's a normal issue : you must do a G28 so the printer know where are all axes ?

Max1963, 2018-02-12:
Thank you. So every time I start the printer have to do an home all? Also if the axis are home already?

ripp2003, 2018-02-12:
hmmmm normally the printer made a g28 at beginning of the print so unless you need to do some test, it's not needed.
For security, Marlin got a function to forbid any mouvement if G28 was not set :by default this function is not activated

Lyr3x, 2018-02-13:
You always should have a G28 in your start script otherwise the printer does not know where the toolhead is placed, as @ripp2003 said.

When you just powered the printer and you want to move your axis, you need to perform a G28 as well.

Max1963, 2018-02-13:
Ok, thanks for hints.

closed 2018-02-13 by Max1963