MarlinFirmware/Marlin #9606: Anet A8 with E3d Bowden, NTC 3950 thermistor getting too hot/heating failed.

CurtisNull, 2018-02-12:
I'm having trouble with this. I found some code here: https://nutz95.wordpress.com/2014/04/13/marlin-firmware-thermistor-3950-table/ 

This is thermistor information. It doesn't seem to work, as I'm pretty sure the nozzle got red hot at one point. I really just want to get my 3D printer working with Marlin, but it seems that I am unable to.

Lyr3x, 2018-02-13:
May you post your configuration files

Max1963, 2018-02-13:
Don't know if this would be of any help, but with my Anet A8 and E3D clone, I'm stuck with default Anet A8 configuration.h, (extruder=5) and all seems to works fine.

Lyr3x, 2018-02-13:
this isnt helpful, because you got a completely different thermistor. He need to set up Thermistor 11, 13 or 60. I guess 11 is the right one.
I think he did not changed the configuration file for the proper thermistor, which is very dangerous.
Its also dangerous do give such wrong advices!

@CurtisNull you should read some documentation and safety information before using a cheap printer kit like the Anet A8 which have a lot safety issues.

Max1963, 2018-02-13:
> this isnt helpful, because you got a completely different thermistor. He need to set up Thermistor 11, 13 or 60. I guess 11 is the right one.

Sorry, but I've same thermistor on my E3D clone (NTC 3950) and didn't work either with 11, 13 or 60 ones... Work, for me, flawless with stock 5 in config

thinkyhead, 2018-02-23:
Good luck @CurtisNull — If you find any bugs in Marlin or have a feature request please let us know.

closed 2018-02-23 by thinkyhead