MarlinFirmware/Marlin #9607: HAL_timer_set_count Register TC_CV

MagoKimbra, 2018-02-12:
@thinkyhead 
In HAL_DUE the HAL_timer_set_count routine, even if not used, makes no sense.
The TC_CV timer register in the SAM3X8E processor is ready only.
So I think it's right to cancel the routine.

Datasheet http://wiki.amperka.ru/_media/%D0%BF%D1%80%D0%BE%D0%B4%D1%83%D0%BA%D1%82%D1%8B:arduino-due:atmel-11057-32-bit-cortex-m3-microcontroller-sam3x-sam3a_datasheet.pdf

Page 879, Table 36.6
0x00 + channel * 0x40 + 0x10 Counter Value TC_CV Read-only 0

teemuatlut, 2018-02-12:
This actually applies to the Teensy chips as well in slightly different way.

> 40.3.4 Counter (FTMx_CNT)
The CNT register contains the FTM counter value.
Reset clears the CNT register. Writing any value to COUNT updates the counter with its
initial value, CNTIN.

p3p, 2018-02-13:
It's going to be time consuming for thinkyhead or others that are not as familiar with all architectures currently supported to do necessary sweeping changes to the HAL layer at this point, although those functions were added a while ago I think, is there a way to write the current count value on those platforms?

thinkyhead, 2018-02-20:
There's noplace in Marlin where `HAL_timer_set_count` is called, so it can definitely be dropped. We used to set `TCNT0` directly, but that has been out for a long while.

If it was needed, there could be a way to set the value indirectly, such as setting a "bottom" value for the timer counter (where the "compare" value is "top") and then doing a "reset" on it. So far, reading the datasheet, this doesn't seem to be an option.

closed 2018-02-20 by thinkyhead