MarlinFirmware/Marlin #9608: [FR] Add Generic GPL Compliance Notice endorsed by Devs to Wiki/Docs

aenertia, 2018-02-12:
-----------------------------------------------------------
### Feature Request
[FR] Add GPL Compliance process information to Documentation

------------------------------------------------------------
### GPL Compliance in Products in the Wild
I recently funded the Dobot Mooz2 printer.

https://www.mooz.co/

After some basic strings analysis and looking at the board layout. It's quite Clear it's based on a 1.0.x series of Marlin/GRBL code and still includes the Copyright strings in the compiled binaries. 

The base has been confirmed to be a fork by Mooz/Dobot in Q&A threads and Forums in various places. 

http://forum.dobot.cc/t/gpl-compliance-source-code-for-dobots-use-of-marlinfirmware-in-mooz/602
------------------------------------------------------------
### Bug Report
I've requested via the official support forum nicely on several occasions for the Source for the Mooz to be provided. The Unit shipped with no GPL notice or link to the license either. 

It would be useful to have a standard 'What to do' wiki page to more strongly support requests for source which are using Marlin as their base for shipping products. IANAL but in order for Legal requests for compliance a least one copyright holder must be party to a request to have any legal standing in most Jurisdictions. 

What would be helpful is a list of steps and templated project request endorsed by one or more Copyright Holders in these instances. This would save time detracting from actual useful work by Dev's and give a little more weight to end-users requests to non-compliant organizations shipping without complying with the terms of the GPL. esp in relation to Source code changes in devices which prevent diff/evaluation and compliation of mainline in their products/boards.

thinkyhead, 2018-02-20:
Thank you for the heads-up. Others have also mentioned DoBot to me recently.

It's near impossible to get compliance from some of these manufacturers. The most effective approach we have been able to find so far is to talk to the companies in China that do properly comply and try to get them to spread the word and to push for a change in the culture.

I've also been spitballing with various 3D printing and open source advocates, and will be adding a page to the Marlin site highlighting the manufacturers that use Marlin. On that page I will include the good ones and link to their repositories and downloads. The bad ones will be marked in red and I'll provide corporate and support contact information.

I definitely want to include a guide for users on the same page, letting them know in plain terms what their rights are under our license, and I'll encourage them to play the squeaky wheel.

Some of these vendors are certainly of the mindset that they will eventually release the source, but they want to wait until they have a better foothold on the market before they give away the keys to the kingdom, as they see it. Dobot has probably done some clever things that they want to conceal from competitors. Most of my clients would probably also like to hide their so-called innovations in perpetuity. Gratitude is not a widespread value in the business world. (But we the enlightened know better.)

GameMaker, 2018-02-27:
Wheel squeaked. :)
https://twitter.com/ThePouncer/status/968330510031257600

thinkyhead, 2018-02-27:
Indeed. All mentions fly into our Slack!

aenertia, 2018-04-12:
Dobot have provided source and - Buildchain info now. Thanks in part to this ticket. Much appreciated.

They are also offering a Unit for anyone who wants to work on it; i'll open another ticket to incorporate support for the Mooz series into mainline on a separate ticket.

https://forum.dobot.cc/t/gpl-compliance-source-code-for-dobots-use-of-marlinfirmware-in-mooz/602

closed 2018-04-12 by aenertia

thinkyhead, 2018-04-12:
I don't see their link to the DoBot source. Where have they got it posted?

aenertia, 2018-04-12:
See https://github.com/MarlinFirmware/Marlin/issues/10391

Dobot source here: https://github.com/mooz3dprinter/MOOZ3DPrinter