MarlinFirmware/Marlin #9611: UBL disabled = Babystepping disabled , please make a fix

studiodyne, 2018-02-12:
- UBL , make shit when go outside of the mesh
  Solved by G29 D before pause  G29 A when back

But , if we want to print with UBL disabled ---> BabyStepping OFF too ...

Possible to have a fix ?

Roxy-3D, 2018-02-12:
I did not test things extensively...   But the DOUBLECLICK_FOR_Z_BABYSTEPPING works with UBL active or inactive.      Are you sure about what you are seeing?    Can you try turning on DOUBLECLICK_FOR_Z_BABYSTEPPING and give it another try?

studiodyne, 2018-02-12:
Totally sure .
Double click works but no action on Z 
Impossible to print if babystep not setted before.
Test with/without Grafx overlay and other compile option

Roxy-3D, 2018-02-12:
I just checked it in bugfix-2.0.0.    Is that the branch you are looking at?    It seems to be just fine in that branch.

studiodyne, 2018-02-12:
I use a one month ago  2.0 bugfix

Roxy-3D, 2018-02-12:
I believe it is working in the current version...

studiodyne, 2018-02-13:
What version is the most advanced  ? or the most updated ?
1.1.x , bugfix 1.1.x , bugfix 2.0 ?
Does bugfix 2.0 receive in realtime the same update of 1.1.x ?

Roxy-3D, 2018-02-13:
I tested it with bugfix-2.0.0.   And  the Z-BabyStepping worked when I did a Double Click.   (Using the Folger Tech i3-2020 example configuration files)   I'm trying to fix a regression to the G26 mesh validation pattern that got introduced when the trig array was collapsed to 6 elements.

studiodyne, 2018-02-13:
Roxy , the last update of marlin have ' pausing destroyed ' , so i can not update all the files , tell me just what unit i have to change , just ubl , or some other  ?

Roxy-3D, 2018-02-13:
Can you do a comparison, and find out which change broke the pausing feature?     That would be the most helpful.    If I remember right, there was BabyStepping issue where if the machine was not homed, baby steps did not happen.   But that was deliberate.   I think that condition has been relaxed.

studiodyne, 2018-02-13:
I will fork a new one , and i will my  past my firm and make a closed pr , you could check too

closed 2018-02-14 by studiodyne