MarlinFirmware/Marlin #9612: [bugfix-2.0.x] MKS SBASE 1.3 does not stop on endstop

Moorviper, 2018-02-12:
Bed doesn't stop on Endstop
------------------------------------------------------------
### Bug Report
did a G28 Y and 
G0 Y90
then 
tried to move y -10 via Octoprint (clicked 100 and move)
https://hessen.meshviewer.net/SBASE/endstop.mp4
It doesn't stop.
Is it a bug or a configuration error.

Configuration:
------------------------------------------------------------
http://hessen.meshviewer.net/SBASE/Configuration.h
http://hessen.meshviewer.net/SBASE/Configuration_adv.h

AnHardt, 2018-02-12:
Your Configuration.h
```
...
/**
 * Software Endstops
 *
 * - Prevent moves outside the set machine bounds.
 * - Individual axes can be disabled, if desired.
 * - X and Y only apply to Cartesian robots.
 * - Use 'M211' to set software endstops on/off or report current state
 */

// Min software endstops curtail movement below minimum coordinate bounds
//#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  #define MIN_SOFTWARE_ENDSTOP_Z
#endif
...
```
Your Configuration_adv.h:
```
...
// @section homing

// If you want endstops to stay on (by default) even when not homing
// enable this option. Override at any time with M120, M121.
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
...
```
All kinds of endstops are off.

Moorviper, 2018-02-12:
Ah ok thx

closed 2018-02-12 by Moorviper