MarlinFirmware/Marlin #9615: Compile Error - Using IIC Character LCD (20x4)

davigre, 2018-02-13:
I want to use ONLY 20x4 LCD with YwRobot IIC encoder ( NO Click Encoder )

So I enabled **LCD_I2C_SAINSMART_YWROBOT** option in 'Configuration.h':

```
//
// Sainsmart YW Robot (LCM1602) LCD Display
//
// Note: This controller requires F.Malpartida's LiquidCrystal_I2C library
// https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home
//
#define LCD_I2C_SAINSMART_YWROBOT
```

in file 'Conditionals_LCD.h' I found this: 

```
/**
   * I2C PANELS
   */

  #if ENABLED(LCD_I2C_SAINSMART_YWROBOT)

    // Note: This controller requires F.Malpartida's LiquidCrystal_I2C library
    // https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home

    #define LCD_I2C_TYPE_PCF8575
    #define LCD_I2C_ADDRESS 0x27   // I2C Address of the port expander
    #define ULTIPANEL  // <- ULTIPANEL was defined.

  #elif ENABLED(LCD_I2C_PANELOLU2)

    // PANELOLU2 LCD with status LEDs, separate encoder and click inputs
 ...
```

As I don't have a Click Encoder I disabled **NEWPANEL** in 'Conditionals_LCD.h' file:

```
#if ENABLED(ULTIPANEL)
    //#define NEWPANEL  // Disable this if you actually have no click-encoder panel
    #define ULTRA_LCD
    #ifndef LCD_WIDTH
      #define LCD_WIDTH 20
    #endif
    #ifndef LCD_HEIGHT
      #define LCD_HEIGHT 4
    #endif
  #elif ENABLED(ULTRA_LCD)  // no panel but just LCD
```

When I compile I get an error:

`#error "ULTIPANEL requires some kind of encoder."``

As you can see I don't have an ULTIPANEL.

Thank's

davigre, 2018-02-13:
Can I undefine 'ULTIPANEL' in Conditionals_LCD.h ?

```
/**
   * I2C PANELS
   */

  #if ENABLED(LCD_I2C_SAINSMART_YWROBOT)

    // Note: This controller requires F.Malpartida's LiquidCrystal_I2C library
    // https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home

    #define LCD_I2C_TYPE_PCF8575
    #define LCD_I2C_ADDRESS 0x27   // I2C Address of the port expander
    //#define ULTIPANEL

  #elif ENABLED(LCD_I2C_PANELOLU2)
```

It compiles now.

thinkyhead, 2018-02-20:
Thanks for pointing this one out. We've been trying to do some cleanup of the LCD options and conditionals to make it possible to use displays without a controller, but there's a lot of mixup in the code as to which of the conditions and lcd options refer to a display, which refer to a controller, and which refer to capabilities. This is just one of those things we haven't sorted out. Your solution is essentially correct. I'll make sure to patch this today.

PR #9743 (2018-02-21): [1.1.x] Sainsmart LCD 20x4
Fix #9615

Distinguish between the Sainsmart 1602 and 2004.

Companion to #9740
PR #9740 (2018-02-21): [2.0.x] Sainsmart LCD 20x4
Fix #9615

Distinguish between the Sainsmart 1602 and 2004.
closed 2018-02-21 by thinkyhead