MarlinFirmware/Marlin #9616: Help me understand probe repeat-ability (M48)?

bcsteeve, 2018-02-13:
I'm confused on what I'm seeing.  I run an M48 and I get pretty awesome looking results to me:

`Recv: Finished!
Recv: Mean: -1.743375 Min: -1.745 Max: -1.742 Range: 0.003
Recv: Standard Deviation: 0.001192`

I mean... that's pretty repeatable, no?

But then I run the exact same test again and I get:

`Recv: Finished!
Recv: Mean: -1.693625 Min: -1.697 Max: -1.687 Range: 0.010
Recv: Standard Deviation: 0.004037`

Again, pretty good results, but...

How is it that the test says I'm off by that little, yet the mean from one test to the next varies by an order of magnitude more than the standard deviation?

What am I not understanding here?

Btw, these are not one-off results.  I've been running tests for hours today trying to get a handle on this and the results from test to test are "significant" while the intra-test Standard Deviation is always well under 0.01.   Not that I'm really all that concerned about 0.04 variance in a practical sense... just don't get how it passes the M48 test with flying colors and then passes it... differently... 2 seconds later.

houseofbugs, 2018-02-13:
Did you happen to have any butterflies fly by your window during the test?

Roxy-3D, 2018-02-13:
First...   .004 mm of standard deviation is **_VERY_** good.    But different locations are going to have more or less 'slop' in the mechanics of the machine.    And if you add the 'S' option to M48 you will get a more realistic value than if you are simply making the probe go up and down.   'S' will shake the extruder around a little bit before each probe.   So any backlash and looseness in the mechanisms should be exposed.

bcsteeve, 2018-02-13:
I think you both may be missing my point.  The numbers don't matter absolutely, I'm talking about the relative difference from one set to another.  It doesn't seem likely (or even possible) to me that a butterfly flying by my window only affects the result between tests, but not within a test.  My tests - as you can see - are very good...  but 10x less good from one to the other.  The whole point of a sensor test spitting out standard deviation is to determine the repeatability of that sensor.  So while I understand the S parameter for gauging the SYSTEM's function... for the SENSOR (and that's what M48 says), it is introducing a variable for no reason.  It makes the test LESS reliable, not more (again, from the perspective of the sensor on its own, and not the system as a whole).  Regardless, that is moot because it still has nothing to do with the drift between tests.

Let me put this another way.  If I run with a P100 (I don't think that's possible due to resources, but let's just pretend) the result should even out enough that it should be identical to two subsequent P50's run quickly one after another  But that's not what's happening.   I do a set and then another set and each set is individually spot on, yet compared to each other there's a significant difference.

I get sensor drift.  And that's exactly why I started down this M48 path.  I was getting weird results from M29 and M30 so I wanted to see if it was my sensor causing the problem. But what I'm seeing and reporting SHOULD be impossible, and I'm wondering if there's some variable that's not clearing and introducing some cumulative error artificially?

Roxy-3D, 2018-02-13:
> I think you both may be missing my point.

I said:

> But different locations are going to have more or less 'slop' in the mechanics of the machine. And if you add the 'S' option to M48 you will get a more realistic value than if you are simply making [...]

DID YOU EVEN TRY THE 'S' OPTION on M48?

bcsteeve, 2018-02-13:
I've seen from your other posts that you have a particular disdain for people that you assume don't try things you suggest :)

Yes, as a regular matter of course I always use S2.  It was irrelevant to the point though.  And **for this particular exercise**, as I said, it introduces variables when the whole point here is to limit them.   But for the record, when I did it with S2 my numbers were (predictably) slightly worse than what I reported above, yet still out of whack from one test to the next.  For completeness, I also removed the glass bed because I get more consistent results directly on the aluminum.  That's not "real world" but again, the point is to eliminate variables and focus on the sensor.  Having things shake around, etc, may simulate real world... but that's not the point.  The test ITSELF needs to be reliable, and I think I've introduced evidence to suggest it may not be.

closed 2018-02-13 by Roxy-3D

bcsteeve, 2018-02-13:
Really?  wow.

AnHardt, 2018-02-13:
What is your Z's steps/mm?
Don't trust any result better than that. +-1 is always possible.
If Z is in the mid of two steps you may get a super result. 
If right at the border even the best probe will show one time the one and one time the other result.

I don't see factor 10, is see a bit more than 3.

bcsteeve, 2018-02-13:
I have to correct myself (and ask @Roxy-3D what you meant).

I did *not* try any "S" flag for M28.  I got confused (and you did?).  From the documentation, there is no S flag.  I thought you meant L and that's what I was answering.  Please confirm or clarify.

bcsteeve, 2018-02-13:
@AnHardt 
Z steps/mm is 400.  What is that... 0.0025mm, right?  When I say factor of 10, I mean a typical difference I'm seeing from one average test result to the next is about 10 times that of the intra-test standard deviation.  Let's look at the data I originally posted (which i just a micro sample of all the tests I've been doing).   Std Dev's of 0.001 and 0.004, yet the average of the first test was 1.74 and the second was 1.69, or a difference of .05... or between 12.5x and 50x the Std Dev's.  So 10x was hardly an exaggeration.

Again, the actual figures don't really matter.  The point remains... VERY tight tolerances when performing an M48, yet wildly (relatively speaking) different results from one test to the next with no changes and only seconds apart.  To be clear, that's with NO physical contact with the machine (everything remote).

AnHardt, 2018-02-13:
@bcsteeve 
I'd have more trust in the author of that function than in documentation.

There is always one definitive source, the *source*
```cpp
/**
 * M48: Z probe repeatability measurement function.
 *
 * Usage:
 *   M48 <P#> <X#> <Y#> <V#> <E> <L#>
 *     P = Number of sampled points (4-50, default 10)
 *     X = Sample X position
 *     Y = Sample Y position
 *     V = Verbose level (0-4, default=1)
 *     E = Engage Z probe for each reading
 *     L = Number of legs of movement before probe
 *     S = Schizoid (Or Star if you prefer)
 *
 * This function requires the machine to be homed before invocation.
 */
void GcodeSuite::M48() {

  if (axis_unhomed_error()) return;

  const int8_t verbose_level = parser.byteval('V', 1);
  if (!WITHIN(verbose_level, 0, 4)) {
    SERIAL_PROTOCOLLNPGM("?(V)erbose level is implausible (0-4).");
    return;
  }

  if (verbose_level > 0)
    SERIAL_PROTOCOLLNPGM("M48 Z-Probe Repeatability Test");

  const int8_t n_samples = parser.byteval('P', 10);
  if (!WITHIN(n_samples, 4, 50)) {
    SERIAL_PROTOCOLLNPGM("?Sample size not plausible (4-50).");
    return;
  }

  const bool stow_probe_after_each = parser.boolval('E');

  float X_current = current_position[X_AXIS],
        Y_current = current_position[Y_AXIS];

  const float X_probe_location = parser.linearval('X', X_current + X_PROBE_OFFSET_FROM_EXTRUDER),
              Y_probe_location = parser.linearval('Y', Y_current + Y_PROBE_OFFSET_FROM_EXTRUDER);

  #if DISABLED(DELTA)
    if (!WITHIN(X_probe_location, MIN_PROBE_X, MAX_PROBE_X)) {
      out_of_range_error(PSTR("X"));
      return;
    }
    if (!WITHIN(Y_probe_location, MIN_PROBE_Y, MAX_PROBE_Y)) {
      out_of_range_error(PSTR("Y"));
      return;
    }
  #else
    if (!position_is_reachable_by_probe(X_probe_location, Y_probe_location)) {
      SERIAL_PROTOCOLLNPGM("? (X,Y) location outside of probeable radius.");
      return;
    }
  #endif

  bool seen_L = parser.seen('L');
  uint8_t n_legs = seen_L ? parser.value_byte() : 0;
  if (n_legs > 15) {
    SERIAL_PROTOCOLLNPGM("?Number of legs in movement not plausible (0-15).");
    return;
  }
  if (n_legs == 1) n_legs = 2;
[... 159 lines cut ...]
      SERIAL_PROTOCOLPGM(" Min: ");
      SERIAL_PROTOCOL_F(min, 3);
      SERIAL_PROTOCOLPGM(" Max: ");
      SERIAL_PROTOCOL_F(max, 3);
      SERIAL_PROTOCOLPGM(" Range: ");
      SERIAL_PROTOCOL_F(max-min, 3);
      SERIAL_EOL();
    }

    SERIAL_PROTOCOLPGM("Standard Deviation: ");
    SERIAL_PROTOCOL_F(sigma, 6);
    SERIAL_EOL();
    SERIAL_EOL();
  }

  clean_up_after_endstop_or_probe_move();

  // Re-enable bed level correction if it had been on
  #if HAS_LEVELING
    set_bed_leveling_enabled(was_enabled);
  #endif

  report_current_position();
}
```

bcsteeve, 2018-02-13:
> I'd have more trust in the author of that function than in documentation.

Point taken (but they should be consistent).

Documentation exists because source is difficult to read.

Changes nothing though.  I tried it.  It moves things around (in a nice smooth pretty pattern).  If anything, that just makes it less accurate of a test **for what we're discussing**.  I fully appreciate the real world application.

bcsteeve, 2018-02-13:
lol... but "discussing" is not the appropriate term.  My point is being completely ignored.  Closed even, without a word.   Whatever.  THIS particular thing isn't much of an issue.  I'm trying to HELP diagnose a larger issue with ABL, and all I wanted to do was confirm consistancy with my probe to rule that variable out.  Apparently I can't do it with this in-built test, which obviously (by the author's comments) not actually intended to do what it is called, which is "Probe Accuracy Test".   In fact, it seems it is more of a "Probe precision test".  My results are very precise... not very accurate (the terms are often confused.  See  http://cdn.antarcticglaciers.org/wp-content/uploads/2013/11/precision_accuracy.png, for example)

AnHardt, 2018-02-13:
Let me tell you a little half offtopic story about things tat repeat and have at least similar results contrary to your M48 experience.

About the first thing when @Roxy-3D joined us, was to invent M48. All worked pretty well but 'S' behaved much more random than expected. Roxy searched for weeks but could not find the problem. Finally 'S' stayed in, to repair it later, but was likely not propper documented for that reason. Much later we discovered that Roxy tried to use sin()/cos() with grad instead of rad parameters, because (and now the clue) Arduino documentation was unclear about that, at that time.

If things are really repeatable the M48 documentation will be improved pretty soon now, like the Arduino documentation for sin/cos/tan did.

I'd really like to know why your probe system produces that big inter M48 discrepancies. I could now write about some dozen speculations how a dozen kinds of probe systems can fail to be repeatable, but a hint about _your_ system would be highly appreciated.

Roxy-3D, 2018-02-13:
> > which obviously (by the author's comments) not actually intended to do what it is called, which is "Probe Accuracy Test". In fact, it seems it is more of a "Probe precision test". 

Very literally...   It is a Z-Probe Repeatability Test.    It is documented as such.  For example here:   http://reprap.org/wiki/G-code#M48:_Measure_Z-Probe_repeatability

Or here in our source code:  https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L104  
or here:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L7257-L7273

bcsteeve, 2018-02-13:
Yeah... I get that.  Well, sort of.  To be nit-picky, it is more publicly documented as [M48 - Probe Accuracy Test](http://marlinfw.org/docs/gcode/M048.html).

But as YOU suggested to introduce shake, movement, etc... that no longer becomes about the probe, but about the probe + whatever it is attached to + what is moving it, etc.

AnHardt, 2018-02-13:
@Roxy-3D 
+
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/gcode.h#L100
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/calibrate/M48.cpp#L43
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L104
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L7345

thinkyhead, 2018-02-14:
Should I add `S` to the `M48` docs at the marlinfw website? It seems like a fun option.

AnHardt, 2018-02-14:
Yes please. Just to make my story true.