MarlinFirmware/Marlin #9617: WHO IS THE GUY THAT HAVE MADE THE PR FOR THE ACTUAL PAUSE SYSTEM

studiodyne, 2018-02-13:
**PLEASE STOP YOUR ETERNAL MODIFICATION I'M REALLY TIRED !!!!!!!!!

Not tested , not finished ,  and MADE ONLY FOR YOUR MACHINE

Every day something is modified  , and now pause system IS SHIT

YOU CAN NOT MODIFY ETERNALLY , because we built our machine and we need A FIX PSYCHOLOGY 

You change the names every weeks , You change all 

AND NOT TESTED NOT VERIFIED**

@thinkyhead   please stop to merge pause system , before asking me to check  , looks the tons of shit 

First : 
```
/**
 * Unload filament from the hotend
 *
 * - Fail if the a safe temperature was not reached
 * - Show "wait for unload" placard
 * - Retract, pause, then unload filament
 * - Disable E stepper (on most machines)
 *
 * Returns 'true' if unload was completed, 'false' for abort
 */
This code says '  retract , pause to cool and unload ' ,   HERE I UNDERSTAND  the filament is cooled to avoid to dirty the gear when go out 
But later in the code 

  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH 13   // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY 5000          // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH 8      // (mm) An unretract is done, then this length is purged.
```

Squid116, 2018-02-14:
1. If you require stability - why would you be pulling and compiling frequently? surely you would only want to use a stable branch?
2. Why don't you watch the areas of the code you are particularly interested in and add some constructive criticism on any PRs which impact this feature?  This might would help both you and the devs in a positive manner.
3. If you are frequently pulling new versions, you can commit and merge your configs in your own fork (or local repo).  There shouldn't be a need for 'dropping' a config in and out regularly if you do this.
4. The **VOLUNTEER** devs are working very hard, they are maintaining an enormous code base, adding new features and moving towards 2.0 and 32 bit compatibility.  They are going to get some stuff wrong occasionally (I'm not sure they are in this case), as they are after all human - however a long rant like this does nothing to help and only distracts them from these tasks.

studiodyne, 2018-02-14:
Yes , I know
But , the first line of the rules here is  ' RESPECT THE MILLIONS HOUR OF WORKS GIVEN gratefully by everyone here ' , the second lines is ' MAKE ALWAYS A PROGRESS AND DONT DESTROY ALL THAT HAVE ALREADY MADE

studiodyne, 2018-02-14:
To answer you
I discover an old bug of UBL , i'm talking about this , in the issue tab ,  and i discover a second bug again , and the communauty tell me that all is fixed in the last bugfix , it's why , i decide again and again , to update , but i have my own features to install over it ,  i have made ' spool swapping , my own filament loading , my own multi pin sensor runout ' and MANY MORE

So , i decide , to fuck a few hours  to make my new release , but , i discover , THE PAUSE SYSTEM IS DESTROYED , totally DEAD
And it's not the first time someone make some BAD PR about the pause system , i'm in favor of progression , but ,  there is no progression , always regression
It's not me when i scream here , but , trust me when i say , the PAUSE SYSTEM IS FUCKED on the last bug fix

AnHardt, 2018-02-14:
@studiodyne 
I would have much more trust in what you tell us if you wouldn't be that loud an seve confident.
Always you are writing in capitals using superlatives and claiming to be the best and writing error free code. That's simply unlikely. What can we trust in and what not? Reading your comments leave the impression you are not trustworthy

Ok. You want your extra attention. But if you continue that way you will end in my killfile.

By the way. Yes, we have a problem with code changed by programmers not knowing what they do. But every now and then we all do that. Marlin is too complex now to have a complete overview. A year ago i knew the code nearly by heart. But after a break for about half a year i have no chance to keep up. Marlin is moving faster than i can follow it. That's frustrating for me, but a sign for a much more active community - what makes me lucky.

thinkyhead, 2018-02-14:
http://marlinfw.org/docs/development/contributing.html#community-guidelines

I may need to add a new section for mad Frenchmen.

thinkyhead, 2018-02-14:
Did you actually create 5 sock-puppet Github accounts to upvote this nonsense?

Thin ice is breaking.

closed 2018-02-14 by thinkyhead

studiodyne, 2018-02-14:
@AnHardt  I totally don't care what you always says , i hate your style of communication and i never read all you say
@Thinkyhead , i love your style , your brain , this extraordinary brain , and every day when i use marlin , this magic firmware that is truely a great advance firmware , babystep/ubl/Fw , i think about you , and the chance to have you as boss here, and the time you spent everyday FOR ME , and all people of this world , and all the companies that love marlin. 
But i'm not engineer like you , but , i know when SHIT IS DONE !!!!!!!!!!!! and i know the last PR is not thinkyheadstyle , you trust too quickly some guys and you merge them . The last pause system is modified everyweek , it's too much ! Names are changed all the time ! Impossible to try a new release just because of a little name. What a surprise when i would help roxy to find the fix for ubl strange behavior , to see what is became the config adv ... changed again , and again , and always THE PAUSE SYSTEM !
STOP !!! too much ! And i was afraid to try it. I know it's not you , because , you don't have this writting style , but you trust too much some bad writters.
In all i have said about this topic , you will never the target of my words , only the guys you trust too much !
@AnHardt  : do not answer me , i will never read

studiodyne, 2018-02-14:
@thinkyhead   I came too late in c++world and i can contribute as the same powerfull way you do it , so i want to make a donation to you , but , in all your website or anywhere , it's totally impossible to talk to you in private. I have a modest life , it will be a modest donation , but i want to thank you , i love my firmware , it's really a GREAT ADVANCED SOFTWARE !

AnHardt, 2018-02-14:
Plonk