MarlinFirmware/Marlin #9619: [bugfix-1.1.x] Linker error since a specific commit

Zwaubel, 2018-02-13:
------------------------------------------------------------
### Compile Error
Since commit 8979990805782246e56c7eab2d39450f30d6ee42 marlin isn't compiling anymore.

I'm getting the following error message while being in the stage of Linking:
C:\Users\\*Username*\AppData\Local\Temp\ccsiZlNx.ltrans1.ltrans.o: In function `main':
ccsiZlNx.ltrans1.o:(.text.startup+0x34a): undefined reference to `Temperature::setPwmFrequency(signed char, int)'
ccsiZlNx.ltrans1.o:(.text.startup+0x362): undefined reference to `Temperature::setPwmFrequency(signed char, int)'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\megaatmega2560\firmware.elf] Error 1

Every commit prior to 8979990805782246e56c7eab2d39450f30d6ee42 compiles just fine.

I think it is related to the changes in the files Marlin.h, temperature.h and temperature.cpp

------------------------------------------------------------

Zwaubel, 2018-02-14:
Just tested the branch bugfix-2.0.x: compiles perfectly good without any linker errors.

thinkyhead, 2018-02-23:
Yes, we definitely got this one. Thanks!

closed 2018-02-23 by thinkyhead