MarlinFirmware/Marlin #9620: [bugfix-2.0.x] Cannot compile with Sublime or Atom (Windows or Linux with fresh install).

houseofbugs, 2018-02-13:
So no matter what I do I cannot get this to compile on either my Linux or Windows box. The errors are not leading to much online and coming up with other things unrelated to Marlin. I am compiling for the MKS SBASE. Thank you in advance.

[ Deviot 2.1.6 ] Starting...
Traceback (most recent call last):
  File "C:\Python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\Timothy\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\platformio.exe\__main__.py", line 9, in <module>
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\__main__.py", line 123, in main
    maintenance.on_platformio_exception(e)
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\maintenance.py", line 64, in on_platformio_exception
    telemetry.on_exception(e)
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\telemetry.py", line 347, in on_exception
    mp = MeasurementProtocol()
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\telemetry.py", line 66, in __init__
    self['cid'] = app.get_cid()
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\app.py", line 350, in get_cid
    cid = get_state_item("cid")
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\app.py", line 292, in get_state_item
    with State() as data:
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\app.py", line 91, in __init__
    self.path = join(util.get_home_dir(), "appstate.json")
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\util.py", line 237, in get_home_dir
    join(expanduser("~"), ".platformio"))
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\util.py", line 217, in get_project_optional_dir
    config = load_project_config()
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\util.py", line 349, in load_project_config
    cp.read(path)
  File "C:\Python27\Lib\ConfigParser.py", line 305, in read
    self._read(fp, filename)
  File "C:\Python27\Lib\ConfigParser.py", line 546, in _read
    raise e
ConfigParser.ParsingError: File contains parsing errors: D:\Timothy\Desktop\Marlin-bugfix-2.0.x\platformio.ini
	[line 177]: '!python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py\n'
Traceback (most recent call last):
  File "C:\Python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\Timothy\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\platformio.exe\__main__.py", line 9, in <module>
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\__main__.py", line 123, in main
    maintenance.on_platformio_exception(e)
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\maintenance.py", line 64, in on_platformio_exception
    telemetry.on_exception(e)
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\telemetry.py", line 347, in on_exception
    mp = MeasurementProtocol()
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\telemetry.py", line 66, in __init__
    self['cid'] = app.get_cid()
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\app.py", line 350, in get_cid
    cid = get_state_item("cid")
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\app.py", line 292, in get_state_item
    with State() as data:
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\app.py", line 91, in __init__
    self.path = join(util.get_home_dir(), "appstate.json")
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\util.py", line 237, in get_home_dir
    join(expanduser("~"), ".platformio"))
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\util.py", line 217, in get_project_optional_dir
    config = load_project_config()
  File "c:\users\timothy\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\util.py", line 349, in load_project_config
    cp.read(path)
  File "C:\Python27\Lib\ConfigParser.py", line 305, in read
    self._read(fp, filename)
  File "C:\Python27\Lib\ConfigParser.py", line 546, in _read
    raise e
ConfigParser.ParsingError: File contains parsing errors: D:\Timothy\Desktop\Marlin-bugfix-2.0.x\platformio.ini
	[line 177]: '!python Marlin/src/HAL/HAL_STM32F1/stm32f1_flag_script.py\n'

Peca006, 2018-02-14:
Do you have a `#define MOTHERBOARD BOARD_MKS_SBASE` in your Configuration.h? Are you in fresh pulled `bugfix-2.0.x` branch? I just managed to built it successfully (platformio core on Windows 10). However I do not know if my configuration is all right - I have yet to test it on my machine.

thinkyhead, 2018-02-23:
@houseofbugs — Should be much easier to build now. No more git dependency.

The "PlatformIO IDE" installed in either **Visual Studio Code** or **Atom** are now the recommended solutions. The **Deviot** plugin for **Sublime Text** unfortunately has bugs that make it incompatible with our `platformio.ini` file.

Before building, be sure to delete the `.poenv` and `.piolibdeps` folders.

boelle, 2019-02-20:
@houseofbugs tried what thinky wrote?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek