MarlinFirmware/Marlin #9621: TMC2130 Z driver error for unknown reason, printer halted

Sloppsta, 2018-02-13:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

using marlin 1.1.8 on a ramps with reprap full graphics. Moved CS pins to 44/64 on aux 2 and SCK SD0 SD1 to the icsp header on the mega.

after installing, I get a Z driver error every time I try to home.

also, stealthchop reads false even though I have it enabled.

READ: Z driver error detected:
READ: 		X	Y	Z	E0
READ: Enabled		false	false	false	false
READ: Set current	638	638	638	638
READ: RMS current	1104	1104	1104	1104
READ: MAX current	1557	1557	1557	1557
READ: Run current	19/31	19/31	19/31	19/31
READ: Hold current	11/31	11/31	11/31	11/31
READ: CS actual		0/31	0/31	0/31	0/31
READ: PWM scale	0	0	0	0
READ: vsense		0=.325	0=.325	0=.325	0=.325
READ: stealthChop	false	false	false	false
READ: msteps		256	256	256	256
READ: tstep		0	0	0	0
READ: pwm
READ: threshold		0	0	0	0
READ: [mm/s]		-	-	-	-
READ: OT prewarn	false	false	false	false
READ: OT prewarn has
READ: been triggered	false	false	false	false
READ: off time		0	0	0	0
READ: blank time	16	16	16	16
READ: hysterisis
READ: -end		-3	-3	-3	-3
READ: -start		1	1	1	1
READ: Stallguard thrs	0	0	0	0
READ: DRVSTATUS	X	Y	Z	E0
READ: stallguard					
READ: sg_result		0	0	0	0
READ: fsactive					
READ: stst					
READ: olb					
READ: ola					
READ: s2gb					
READ: s2ga					
READ: otpw					
READ: ot					
READ: Driver registers:
READ: 	X = 0x00:00:00:00
READ: 	Y = 0x00:00:00:00
READ: 	Z = 0x00:00:00:00
READ: 	E0 = 0x00:00:00:00
READ: 
READ: Error:Printer halted. kill() called!

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

teemuatlut, 2018-02-14:
You have a communication error for all your drivers.
The register responses should not be 0x00000000.
Re-check your wiring, the pin configurations and that you have 12V power turned on.

FUN3d, 2018-02-15:
You can also check of your tmc2130 board is configured correctly, especially the SPI enable pin.  If this pin is grounded the SPI interface is disabled and configured in standalone mode, you cannot communicate with the TMC2130 at that moment. There are also the CFG4 and CFG5 pins CFG4 must be grounded and CFG5 To Vio
Below a link from Mytechtips what illustrates very well where to look and what to do, also a schematic of the tmc2130 board

www.youtube.com/watch?v=AVHVCPcFHUs&t=348s
github.com/MarlinFirmware/Marlin/files/1728713/SilentStepStick-TMC2130_v10.pdf

teemuatlut, 2018-02-15:
You may also want to try the latest bugfix branch, bugfix-1.1.x or bugfix-2.0.x.
We fixed a communications related bug not too long ago that hasn't yet made into a release version.

Sloppsta, 2018-02-15:
Thanks guys! Redoing my cabling has now allowed it to communicate with the drivers. Making progress at least. I am now getting a short warning on the z only on M122, it's losing steps. I'll go ahead and try the bugfix branch. Thanks again!

closed 2018-02-23 by thinkyhead