MarlinFirmware/Marlin #9622: UBL and G29 J does not converge

gloomyandy, 2018-02-13:
Hi, I've been experimenting with UBL and in particular G29 J to correct an already stored mesh and have come across what I think may be an issue with the correction code. All of the following is in regard to 3 point correction. I've spent a fair bit of time digging into things and I'd like to get your thoughts on the issue. To illustrate what I think is the problem consider the following situation...
1. We have a machine with a perfectly flat bad (yeah I know...)
2. The initial probe to create the mesh results in every point being at +0.5 (this may happen if like my system the auto home position is not that reliable and ends up different to a probe of the same location). 
3. This mesh is saved using G29 S1
4. At some point later this mesh is reloaded (with G29 L1) and then a G29 J is run to correct it. The three points all return the same value but this time the bed appears to be at at +1.5.

The mesh correction code attempts to correct for both tilt and offset, in this case there is no tilt error, but there is an offset error of +1.0. My understanding is that after correction the above situation should result in a mesh that contains all values set to +1.5. However I think that with the 1.1.8 code base what you will end up with is 0.0 everywhere, if you do the same "thought experiment" with an initial value of say 3.0 and a G29 J value of 4.0 then you get a mesh containing 5.0 rather than the expected 4.0. It seems to me that something is not right.

Digging into the code it seems that tilt and offset errors are handled in two stages. The existing mesh value is first rotated to the new position (which in these examples leaves them unchanged), then the offset correction is combined to provide the final value. The line of code the combines all of this is....
z_values[i][j] += z_tmp - d;
in which z_values contains the current mesh, z_tmp is the newly rotated z value and d is the offset correction. It seems to me that this is the cause of the problem. I don't believe that the new value should be added to the existing value of the mesh. If you look at the above example you will see that in that case z_values[i][j] will contain 0.5, z_tmp will also contain 0.5 since there is no rotation error and d will contain the offset error which in this case is 1.0. So executing the above results in a new value of 0.5 + 0.5 - 1.0 giving 0.0 rather than the expected 1.5. Similarly with the second example we get 3.0 + 3.0 - 1.0 giving 5.0 rather than the expected 4.0.

It seems to me that including the current mesh value twice (the original value plus the new rotated version) is not correct and that instead just the newly rotated value should be used. It also seems to me that there may be a sign issue with d, in my tests the above situation gives d as a +ve value and I think that the correct solution would be to add this value to the rotated current mesh value to give the fully corrected result. So the above line should read...
z_values[i][j] = z_tmp + d;
With this change I get much better correction of the mesh and running multiple G29 J operations results in a consistent mesh rather than one that seems to have values that continue to grow with each new correction.

I hope the above makes some sort of sense. I may have totally misunderstood what is going on (in which case I'm sorry to have wasted your time), but either way thanks for all of the effort on UBL and for taking the time to read this.

NOTE: I see that the G29 J code has very recently been changed (in the bug fix branch) to make it common with the G29 J<n> code. I have not investigated this new implementation, but it looks to me as if it makes corrections in the same way and so the above problem may still exist.

Roxy-3D, 2018-02-13:
The 3-point  UBL Mesh tilt code is different between bugfix-1.1.x and bugfix-2.0.0.   Which one are you using.

gloomyandy, 2018-02-13:
Hi I'm actually using 1.1.8 not the bugfix. But I see that the 1.1.8 bugfix has recently changed to do make the 3 point and general case closer. However I think the actual correction code is still in effect the same between 1.1.8 and the bugfix, but I have not really studied the bugfix code that closely. 

I'll take a look at the 2.0 bugfix code as well!

Roxy-3D, 2018-02-13:
I stand corrected...  I think ThinkyHead moved the 3-point mesh tilting from 2.0.0 to the 1.1.8 branch, so they are both the same.    Assuming that operation was done correctly, it shouldn't matter which bugfix branch we use to do the work.

The thing about the new code is doing the 3-point mesh tilting...   an exact solution is possible.   So that really helps when trying to check the general case of mesh tilting for 'correctness'.

gloomyandy, 2018-02-13:
Yes being able to confirm that the three test points are correct with the new mesh is very useful. 

With the "old 1.1.8" and the change I mentioned above I added code to confirm that the new mesh results in a value of zero if you subtract the mesh value from each of the z values from the three points used to perform the correction. The results with that code were either zero or very close to it, certainly much better than those obtained before the above change. This seemed to be the case with my real world bed not just an imaginary flat one!

As I said I may not have fully understood what is going on but I'm puzzled as to why the new rotated mesh value is added to the old one. I don't think any sort of "normalisation" has been performed on the mesh, but I may have missed it.

Roxy-3D, 2018-02-14:
> As I said I may not have fully understood what is going on but I'm puzzled as to why the new rotated mesh value is added to the old one. I don't think any sort of "normalization" has been performed on the mesh, but I may have missed it.

UBL's G29 J is not used by many people.   It is very possible the logic is 'close' but not 100% correct.   The fact that the multiple iterations of G29 J do not converge suggests that the logic is not 100% correct.    If you find z_values[i][j] += z_tmp - d; produces the correct results...   I'm all for checking that out!!!

gloomyandy, 2018-02-14:
Actually it seems to me that
z_values[i][j] = z_tmp + d;
works better!  :-)

Yes I suspect most people can simply store a mesh and then just use it. My probe does not seem to be very consistent especially when doing a home. So at the moment I use multiple probes (5!) when probing the bed. Because the home position varies I've found that I need to G29 J to "fix things up" when I print. Hence my interest in it. I have a different probe on order (should arrive in the next day or so), which I hope will fix the home problem! But I'm a software eng. by trade so I couldn't resist digging into it, when I got odd results! :-) Hope I'm not wasting your time. As I said before thanks a lot to everyone for the work that has gone into this, I know how much time it takes!

Right time to sleep for me here in the UK now. Let me know if I can help further.

Roxy-3D, 2018-02-14:
> Actually it seems to me that
> z_values[i][j] = z_tmp + d;
> works better! :-)

Right....  You said that!   I cut and pasted from the wrong spot in the thread.   It will be a little bit before I can try this.   But I will.

gloomyandy, 2018-02-14:
No problem. Let me know if I can help at all. On that topic, what version of the firmware would you recommend that I use for testing/experimenting with this sort of thing, I'd like to avoiding causing you folks any extra work by using something that is out of date, but I'm not sure how stable 2.0 is or how it compares with the 1.1.x-bugfix, or have that compares with 1.1.8 I see that there are a fair few changes going into both at the moment.

Roxy-3D, 2018-02-14:
Bugfix-2.0.0 is just as stable as Bugfix-1.1.x.   The difference is the build environment for all the new 32-bit platforms.   Some of those build platforms are not fully baked.   But if you build Bugfix-2.0.0 with Arduino for your 8-bit AVR board, you shouldn't have any problems.   That is in fact what I'm doing right now for the G26 work and UBL off mesh work.

gloomyandy, 2018-02-14:
Thanks, once I've sorted things out with the new probe I'll give 2.0 a try and see how I get on with it. 

FYI I've not seen any of the issues mentioned about rising Z etc. when using UBL, but my setup is such that with a small inset my bed covers all of the range that my printer can reach, so I never end up really going off bed. 

Having read a lot of docs etc. I think it would be good to have a clearer description of how to specify the location of the bed (or printable area) within the reachable area of the machine. In my case the default 0,0 is actually just off the bed, I did consider setting things up such that 0,0 was on the bed and allowing a -ve Y co-ord to access the off bed area (which is I think how the Prusa machines are setup, but it wasn't obvious to me how exactly to do this, or if there are any assumptions about bed location etc. But that is probably for another topic!

dagnall53, 2018-02-15:
I have also been playing with this and created a very similar issue https://github.com/MarlinFirmware/Marlin/issues/9561 that got no comments.. :-(

I suspect the issue is J is not using the "mesh compensated" probing. But I am not sure of the "correct" marlin terminology. 
I downloaded 1.1.8 last week - is the "correction" you mentioned more recent? If not, where do I need to mod my code?

Roxy-3D, 2018-02-15:
The change is considered a 'New Feature' and it was only placed in the bugfix-2.0.0 branch.   (It may have been replicated into the bugfix-v1.1.x branch,   I'll update when I know the answer.)

AndreApollo, 2018-06-12:
Hello! 
I have a i3 printer type and i want to use 3 point tilt correction every time i start the printer. 
We have a correction for this issue?

Thanks a lot.

gloomyandy, 2018-06-12:
The fix discussed here is present in the bugfix-1.1.x ans bugfix-2.0.x branches.

closed 2018-06-12 by gloomyandy