MarlinFirmware/Marlin #9623: Marlin 2.0 error compilation

makskr, 2018-02-14:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `PIO Build LPC1768` I get an error:
```
Error: 
[02/13/18 19:00:03] Processing LPC1768 (platform: nxplpc)

Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: NXP LPC >
SYSTEM: 100MHz
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(light)
Collected 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
|-- <TMC2130Stepper> v2.1.4
Error: BoardConfig: Board is not defined
 [ERROR] Took 2.49 seconds 

 [SUMMARY] 
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment melzi                   	[SKIP]
Environment melzi_optiboot          	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[ERROR]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1722419/configuration.zip)

```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Spawn32, 2018-02-14:
"Error: BoardConfig: Board is not defined"!!
You need to use a config setup from the examples directory that matches you board, and work out from that..

Peca006, 2018-02-14:
Hi there. I've got the same issue. I've use example for MKS SBASE board (just a slightly modified for my printer):
```
D:\gitOthers\Marlin>platformio run -e LPC1768

[02/14/18 10:08:38] Processing LPC1768 (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py, -fmax-errors=5, -DU8G_HAL_LINKS; src_build_flags: -Wall; lib_deps: CMSIS-LPC1768, U8glib-HAL, TMC2130Stepper@>=2.1.1; extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
-----------------------------------------------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
Collected 7 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
|-- <TMC2130Stepper> v2.1.4
Error: BoardConfig: Board is not defined
============================================== [ERROR] Took 0.94 seconds ==============================================

====================================================== [SUMMARY] ======================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment DUE                         [SKIP]
Environment LPC1768                     [ERROR]
Environment LPC1768_debug_and_upload    [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
============================================== [ERROR] Took 0.94 seconds ==============================================
```
There is my configs:
[peca-sbase-cfgs.zip](https://github.com/MarlinFirmware/Marlin/files/1723304/peca-sbase-cfgs.zip)
I tried build it in a VisualStudio 2017 pro too but I've no luck there too:  "Oops!  Make sure you have LPC1768 selected". I think it was due to wrong settings in my VS and lack of knowledge.

p3p, 2018-02-14:
PlatformIO have pushed out an update that changed some required settings, specifically we need to add a board option (`Error: BoardConfig: Board is not defined`) to the LPC1768 env, .. we don't build for a specific board we build for the mcu so currently unsure of the best choice for this.

thinkyhead, 2018-02-14:
Patched by adding a `board` line.

closed 2018-02-14 by thinkyhead

jornada812, 2019-07-15:
I just uploaded bugfix-2.0, I specify in line
#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
and get an error -
Make sure you have the LPC1768 environment selected in your IDE
The compilation is trying to happen for the atmega2560 -  what should I do?