MarlinFirmware/Marlin #9629: How convert this  "\xa4\xa5\xa6\xa7" into chinese font? Marlin or how chinese converted in to  "\xa4\xa5\xa6\xa7"

Shrikant888, 2018-02-14:
How convert this  "\xa4\xa5\xa6\xa7" into chinese font? Marlin or how chinese converted in to  "\xa4\xa5\xa6\xa7"

Please somebody give me information.

thinkyhead, 2018-02-14:
Look at the raw bytes of a Unicode string in any C program. The `\x` is the prefix to the hexadecimal value of each byte.

Shrikant888, 2018-02-14:
Thanks for reply.

For Ex: I took from zh_TW as ("已就绪.")  //" ready.. After converting to hex got like E5 B7 B2 E5 B0 B1 E7 BB AA 2E.. Only we can add \x to all utf8 hex. I am copying same thing in language cn and calling that in configuration file. but displays only utf8 characters like ä...

Ok simple is how can I display zh_CN on LCD. Please tell

thinkyhead, 2018-02-14:
For graphical LCD:

```cpp
#define LCD_LANGUAGE zh_CN
```

Shrikant888, 2018-02-14:
sir, I am doing that only. 

After changing en to zh_CN. Its  coming like ("å·²å°±ç»ª.")  //ready on LCD.

cjsoong, 2018-02-14:
@Shrikant888 
#define LCD_LANGUAGE cn

AnHardt, 2018-02-14:
No. "\xa4\xa5\xa6\xa7" is WELCOME_MSG from the 'cn' language file.
This is not utf!
The hex numbers are positions in the `dogm_font_data_ISO10646_CN.h` font.
Translation is a bit difficult without having a picture of the font.
Best is to replace WELCOME_MSG "\xa4\xa5\xa6\xa7" with
first "\x80\x81\x82\x83\x84\x85\x86\x87"
then "\x88\x89\x8a\x8b\x8c\x8d\x8e\x8f"
...
and lastly to "\xf8\xf9\xfa\xfb\xfc\xfd\xfe\xff"

and writing down or making photos of what you see - to make a table.

Then use that table to write your strings.

Sorry. Utf8 support for scripts using more than 128 non ASCII glyptes was never finished.

thinkyhead, 2018-02-14:
Actually, I think that might be a "shorthand" utf-8 that still gets partial conversion, and not direct indices into the font.

AnHardt, 2018-02-14:
You will need a graphical display to make it work. There is no chance to make this work on text displays.

@thinkyhead 
Only if someone messed it up. This was made for direct indices. MAPPER NON.

AnHardt, 2018-02-14:
Tested 'bugfix-1.1.x' with standard configuration and 
```diff
@@ -1295,11 +1295,11 @@
  *    hr, it, kana, kana_utf8, nl, pl, pt, pt_utf8, pt-br, pt-br_utf8, ru, sk_utf8,
  *    tr, uk, zh_CN, zh_TW, test
  *
  * :{ 'en':'English', 'an':'Aragonese', 'bg':'Bulgarian', 'ca':'Catalan', 'cn':'Chinese', 'cz':'Czech', 'cz_utf8':'Czech (UTF8)', 'de':'German', 'el':'Greek', 'el-gr':'Greek (Greece)', 'es':'Spanish', 'eu':'Basque-Euskera', 'fi':'Finnish', 'fr':'French', 'fr_utf8':'French (UTF8)', 'gl':'Galician', 'hr':'Croatian', 'it':'Italian', 'kana':'Japanese', 'kana_utf8':'Japanese (UTF8)', 'nl':'Dutch', 'pl':'Polish', 'pt':'Portuguese', 'pt-br':'Portuguese (Brazilian)', 'pt-br_utf8':'Portuguese (Brazilian UTF8)', 'pt_utf8':'Portuguese (UTF8)', 'ru':'Russian', 'sk_utf8':'Slovak (UTF8)', 'tr':'Turkish', 'uk':'Ukrainian', 'zh_CN':'Chinese (Simplified)', 'zh_TW':'Chinese (Taiwan)', test':'TEST' }
  */
-#define LCD_LANGUAGE en
+#define LCD_LANGUAGE cn
 
 /**
  * LCD Character Set
  *
  * Note: This option is NOT applicable to Graphical Displays.
@@ -1493,11 +1493,11 @@
 
 //
 // RepRapDiscount FULL GRAPHIC Smart Controller
 // http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
 //
-//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
 
 //
 // MakerLab Mini Panel with graphic
 // controller and SD support - http://reprap.org/wiki/Mini_panel
 //
```
Looks OK!

Shrikant888, 2018-02-14:
Thank you for all you support. Just as per my logic. There is little difference in cn,zh_tw,zh_cn. So for example if I wanted display  ("已就绪.") which is there only in zh_cn but not in cn,zh_tw. 

To display  ("已就绪.") I have to call as #define LCD_LANGUAGE cn in configuration.h file but I have to change values of "\xa4\xa5\xa6\xa7" in language.cn to generate 已就绪.

Earlier what I did--->
Configuration.h:
Board declared as Sanguino
LCD Full Graphic
Enabled U8GLIB ST7920
Language as cn

On LCD Chinese I got. But if I use only Language as zh_CN. On LCD I am getting "å·²å°±ç»ª. which string means ready.
To confirm that I pasted  已就绪 in word pad which gives same string as  "å·²å°±ç»ª. Also in word pad input string is utf8 (which converts any string into utf8). Used Marlin1.1.x.

thinkyhead, 2018-02-14:
I'm wondering if a script ever existed to do the string conversion (mapping to the font) and what has become of it? We sorely miss our helpful and patient contributor @yhfudev who knows all the secrets of Chinese.

AnHardt, 2018-02-14:
zh_CN and zh_TW have been merged in the hope for #5441. Currently they can not work.

thinkyhead, 2018-02-14:
Too bad #5441 arrived during a time when we were focused on many other issues. As the focus is increasingly in improving quality and stability, and less on fancy features, we may have a chance to revive that optimization. I think we have learned that our previous concerns about regenerating language files are not so troublesome. As long as we produce them periodically and for releases no one will lose any sleep.

cjsoong, 2018-02-14:
zh_CN and zh_TW Requires LCD 12864 with a built-in Chinese font rom.
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER LCD 12864 is not built-in Chinese fonts rom.
https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79e352d6YDIgYc&id=41254515745&ns=1&abbucket=15

yhfudev, 2018-02-14:
Hi, @all,
I'm still in the mail list and always ready to help. :-)

It's really need to consider the options seriously if you want to solve the problem as soon as possible.

First, do we need to support UTF-8? If not, just forget about any changes to the source code and we are done.

If yes, then are these followings acceptable?
1. update the font data header files using the bash script when there's changes in the translation files.
I would suggest the maintainer do this job and commit the changes for the generated font header files to git repo.
2. ask user who compile the source to expose the u8glib internal c header "u8g.h"

Last, I would happy to help merging the changes if needed.

yhfudev, 2018-02-14:
@cjsoong
The device you mentioned only supports GB2312 encoding, it not unicode. You have to use another map between GB2312 and unicode.

AnHardt, 2018-02-14:
@cjsoong 
ST7920 has a cinese fontrom but we use the display as a graphic device - not in text mode. 
I think a 8x4 text display is not very useful for marlin.

Shrikant888, 2018-02-15:
Thanks for all.

I will be working on it.

Thank you.

thinkyhead, 2018-02-23:
@yhfudev — It's a pleasure to hear from you. At the moment we're trying to stamp out issues and get a final 1.1.x release out the door. Then we can focus all attention to the more organized 2.0 branch.

I can work with you to get the needed language changes in place within the 2.0 code-base. And I'm sure we'll be able to figure out a way to automate the data files generation, either by adding a script to one of our branches and having it run within Travis CI, or by setting up a cron job on one of my webservers.

The 1.1.9 release should be this weekend if all goes well. I will set up a new issue and add you to the discussion, and then we can get right to work. If you want to get started on the process based on the `bugfix-2.0.x` branch, that would be excellent.

And sorry again for the long delay. I have revisited and rebased your code many times in hope that it would get more attention, but wow things have been piling up around here….

thinkyhead, 2018-02-23:
> 8x4 text display is not very useful for marlin.

I hear it can take a lifetime to fully understand a single Chinese ideogram.

closed 2018-03-07 by thinkyhead

3dle, 2020-02-06:
> zh_CN和zh_TW需要带有内置中文字体rom的LCD 12864。
> REPRAP_DISCOUNT_FULL_GRAPHIC_CONTROLLER LCD 12864不是内置的中文字体。
> 蓝屏LCD12864液晶屏中文字库带背光12864-5V S串/并口显示器件
> [https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79 e352d6YDIgYc＆id = 41254515745＆ns = 1＆abbucket = 15](https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79e352d6YDIgYc&id=41254515745&ns=1&abbucket=15)

> zh_CN和zh_TW需要带有内置中文字体rom的LCD 12864。
> REPRAP_DISCOUNT_FULL_GRAPHIC_CONTROLLER LCD 12864不是内置的中文字体。
> 蓝屏LCD12864液晶屏中文字库带背光12864-5V S串/并口显示器件
> [https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79 e352d6YDIgYc＆id = 41254515745＆ns = 1＆abbucket = 15](https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79e352d6YDIgYc&id=41254515745&ns=1&abbucket=15)

> zh_CN和zh_TW需要带有内置中文字体rom的LCD 12864。
> REPRAP_DISCOUNT_FULL_GRAPHIC_CONTROLLER LCD 12864不是内置的中文字体。
> 蓝屏LCD12864液晶屏中文字库带背光12864-5V S串/并口显示器件
> [https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79 e352d6YDIgYc＆id = 41254515745＆ns = 1＆abbucket = 15](https://detail.tmall.com/item.htm?spm=a230r.1.14.27.79e352d6YDIgYc&id=41254515745&ns=1&abbucket=15)

我想用zh_ch,我的是LCD12864，可是显示出来一堆乱码，不知道怎么办

3dle, 2020-02-06:
> > 8x4文本显示对马林鱼不是很有用。
> 
> 我听说要完全理解单个中国表意文字可能要花费一生。

You can write more detailed tutorials.

thinkyhead, 2020-02-06:
@3dle — This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).