MarlinFirmware/Marlin #9633: MKS TFT32

tofifee, 2018-02-14:
It looks like some have managed to use the MKS TFT as a way to control their printers. How did you manage it exactly? Would you please detail how you got it to work. Especially interested in the combination with the MKS SBase 1.3.

Also is it planned to include the MKS TFT in the LCD controller list?

MagoKimbra, 2018-02-14:
This display has its own firmware that emulates the host, it must not be managed directly from the card and from the marlin firmware. It connects on serial 0 on the mainboard and manages everything via gcode. So there's nothing to add to Marlin.

tofifee, 2018-02-14:
Pinmapping - for I believe it is AUX1 to Serial 0 - would be a worthwhile addition imo.

So if you uncomment #define MKSTFT-AUX1 it would assign the AUX1 pins of the defined board.

Spawn32, 2018-02-14:
Only thing you need to do is:
#define SERIAL_PORT -1    // is the USB port
#define SERIAL_PORT_2 0  // is the serialport for the TFT
#define BAUDRATE 115200 // for the serialport

and make sure that the firmware on you TFT 3.2 is sett to 115200 and Marlin..   :)

MagoKimbra, 2018-02-14:
Correct @Spawn32!

Spawn32, 2018-02-14:
@MagoKimbra 
I just ordered a 5.0" Nextion Enhanced NX8048K050 HMI after seeing it on you MK3duo facebook page, they look brilliant :)

Will it work with Marlin the same way it works with your software or does it need some modification ?

MagoKimbra, 2018-02-14:
I do not like talking about MK here, it's not right ...
I only say no, because it needs a completely different management from normal lcd.
Excuse me, but it seems fair to me ...

Spawn32, 2018-02-14:
You are absolutely correct, slipped my mind, sorry dev's :)

tofifee, 2018-02-14:
@Spawn32 

First of all thank you for pointing me in the right direction!

When I try to compile I run into this error:

`...
Marlin/src/module/configuration_store.cpp:1881:5: note: in expansion of macro 'CONFIG_ECHO_START'
CONFIG_ECHO_START;
^~~~~~~~~~~~~~~~~
Marlin/src/module/configuration_store.cpp:1892:26: error: 'port' was not declared in this scope
SERIAL_ECHOLNPGM_P(port, "  G21    ; Units in mm");
^
...`

tofifee, 2018-02-14:
Had to define SERIAL_PORT as 0 and leave SERIAL_PORT_" commented out for it to compile.

Thanks again!

Spawn32, 2018-02-14:
Huh ? that shouldn't be possible, are you sure you are not using an old release of bugfix-2.0x ?
Multihost (2 ports) has been in there for a while now...

Peca006, 2018-02-14:
@Spawm32: I've build Marlin for MKS-SBASE too with the same issue as @tofifee . I'm pretty sure I have fresh pulled 2.0.x branch.

tofifee, 2018-02-14:
Clone and check-out just before compiling.

Spawn32, 2018-02-14:
downloaded a fresh copy now, no compile errors, strange.....
How do you guy's compile it ? i use the command line option:  "platformio run -e LPC1768"

@tcm0116 any ideer what's going on here ?  😄

tofifee, 2018-02-14:
Same command as you in the platform.io terminal window.

tofifee, 2018-02-14:
OT - slightly ;)

Has anybody tried yet to flash an ESP8266 with the MKS Wifi firmware to get serial over Wifi?

Peca006, 2018-02-14:
@Spawn32: The same command at mine console... I tried checked src code and I think that it should be ok. Now I'm quite confused :(

tofifee, 2018-02-14:
Was just a shot in the dark :)

Peca006, 2018-02-14:
Here are compiler messages:
```
C:\Data\gitrepo\marlin>platformio run -v -e LPC1768

[02/14/18 20:48:11] Processing LPC1768 (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py, -fmax-errors=5, -DU8G_HAL_LINKS; src_build_flags: -Wall; lib_deps: CMSIS-LPC1768, U8glib-HAL, TMC2130Stepper@>=2.1.1; extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; board: lpc1768; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
--------------------------------------------------------------------------
Library Storage: C:\Data\gitrepo\marlin\.piolibdeps
Looking for CMSIS-LPC1768 library in registry
Warning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
LibraryManager: Installing id=1932
U8glib-HAL @ 0.4 is already installed
LibraryManager: Installing id=1493 @ >=2.1.1
TMC2130Stepper @ 2.1.4 is already installed
Collected 9 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <CMSIS-LPC1768> v0.0.0 (C:\Data\gitrepo\marlin\frameworks\CMSIS)
|-- <U8glib-HAL> v0.4 (C:\Data\gitrepo\marlin\.piolibdeps\U8glib-HAL_ID1932)
|-- <TMC2130Stepper> v2.1.4 (C:\Data\gitrepo\marlin\.piolibdeps\TMC2130Stepper_ID1493)
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\module\configuration_store.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -Wall -DF_CPU=100000000L -DPLATFORMIO=30501 -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin\src\HAL -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src Marlin\src\module\configuration_store.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\module\stepper.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -Wall -DF_CPU=100000000L -DPLATFORMIO=30501 -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin\src\HAL -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src Marlin\src\module\stepper.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\module\stepper_indirection.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -Wall -DF_CPU=100000000L -DPLATFORMIO=30501 -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin\src\HAL -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src Marlin\src\module\stepper_indirection.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\module\temperature.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -Wall -DF_CPU=100000000L -DPLATFORMIO=30501 -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin\src\HAL -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src Marlin\src\module\temperature.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\module\tool_change.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -Wall -DF_CPU=100000000L -DPLATFORMIO=30501 -DTARGET_LPC1768 -DU8G_HAL_LINKS -IMarlin\src\HAL -IMarlin\src\HAL\HAL_LPC1768\include -IMarlin -Iframeworks\CMSIS\LPC1768\include -Iframeworks\CMSIS\LPC1768\lib -Iframeworks\CMSIS -I.piolibdeps\U8glib-HAL_ID1932\src -I.piolibdeps\TMC2130Stepper_ID1493\src Marlin\src\module\tool_change.cpp
arm-none-eabi-g++ -o .pioenvs\LPC1768\src\src\sd\Sd2Card.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -Os -ffunction-sections -fdata-sections -mthumb -nostdlib -mcpu=cortex-m3 -Os -mcpu=cortex-m [... cut ...]
Environment LPC1768                     [ERROR]
Environment LPC1768_debug_and_upload    [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment STM32F1                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
======================= [ERROR] Took 8.17 seconds =======================
```
Here are config files:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1725457/configs.zip)

Spawn32, 2018-02-14:
Just for the fun of it, test my config's but dont move any steppers if it compiles, it's for a corexy :)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1725532/config.zip)

Peca006, 2018-02-14:
@Spawn32: Same error. Hmm - Might be it a PlatformIO issue? Today I hit one already: #9623 
Would you mind to send your platfdomio version and `platformio.ini`? However `platformio.ini` should be the same as it is commited into 2.0.x branch.

Spawn32, 2018-02-14:
PlatformIO, version 3.5.1

and i use the platformio.ini that's included if you downloaded the latest bugfix branch , strange....

Peca006, 2018-02-14:
Update does not solved it. Still the same issue :/

tofifee, 2018-02-14:
I had several updates today. One was xonxerning the LPC.

Can you try to comment out the SERIAL_PORT_2  and assign 0 to SERIAL_PORT?

Maybe something is out of whack and only this combination gets through in certain configs.

tofifee, 2018-02-14:
Also make double sure you hit save while in Configuration.h after changing something.

I was tearing my almost non-existing hair out at some point because nothing changed after uploading. But maybe that‘s only me ;)

Spawn32, 2018-02-14:
you'r like me then :)
I always delete the .pioenvs and .piolibdeps directory's if i change something major to make sure it re compiles everything.

That's something i started with after using the arduino compiler, it dosent always recompile files it should have done :)

Peca006, 2018-02-14:
@tofifee: I already made a successful build with just one serial port and it succeed again. I try delete directories that mentioned @Spawn32 and "let's see"...

Peca006, 2018-02-14:
Deleting directories not helped.

Peca006, 2018-02-14:
After another round of code inspection ( - I've been looking for various serial port dependencies ) I've uncomment `#define EEPROM_CHITCHAT` and now it builds! It was commented in sample config for MKS SBASE - so I do not know if it will work. I'm n00b and my printer is fresh new and is not tested yet. I'll continue tomorrow by testing steppers and then I'll continue by calibrating printer - if it will works.
@tofifee: Try it please - `#define EEPROM_CHITCHAT` is on line 1140 or so in `configuration.h` file.

tofifee, 2018-02-14:
Good!

My problem was solved by my above mentioned steps :) As I don‘t want to run a Computer as host and the SBase 1.3 goes into flash mode as soon as it detects a firmware.bin file on the SD Card.

Let‘s see what tomorrow brings. Night, folks!

Spawn32, 2018-02-15:
Nice that it worked out for you guy's
But this is still some kind of bug, i have EEPROM_CHITCHAT defined and it compiles just fine..

@thinkyhead any ider what this is about ? :)

tofifee, 2018-02-15:
@Spawn32 

Was commented out in my config. Did uncomment it and it did compile. So it might be something else.

Hm. Just decided to try two serial port definitions again.

SERIAL_PORT as -1

SERIAL_PORT_2 as 0

And it compiled as well.

BUT

After commenting EEPROM_CHITCHAT out again it threw the previous error again and didn't compile.

So went back and enabled it again ... Surprise! ... didn't compile.

Decided to try your hair-pulling-prevention-technique and deleted the .pio directories:

Compiles!

My take is that there is a problem with platform.io not cleaning up properly before compiling and reusing something it shouldn't.

One solution could be a little batch file like this:

`#!/bin/bash

# CleanUpAndCompile script for platform.io to prevent zombies in the .pioenvs and .piolibdeps directories from introducing errors

echo "Deleting .pio directories in current folder ..."
rm -rf .pio*
echo "Compiling ..."
platformio run -e LPC1768
ls -thal .pioenvs/LPC1768/`

Pentelfrost, 2018-02-19:
I was/am seeing the same issue. From a fresh git clone and just copying the example config files from the Mks/Sbase directory I can't compile with the same errors from above. There are no platformIO files at that point so it doesn't seem to be just a clean up issue. Enabling EEPROM_CHITCHAT gives a successful compile as suggested.

thinkyhead, 2018-02-25:
It is now patched:
```cpp
  #if ADD_PORT_ARG
    #define CONFIG_ECHO_START do{ if (!forReplay) SERIAL_ECHO_START_P(port); }while(0)
  #else
    #define CONFIG_ECHO_START do{ if (!forReplay) SERIAL_ECHO_START(); }while(0)
  #endif
```

closed 2018-02-25 by thinkyhead

GhassanYusuf, 2020-01-11:
I have got the answer, 
in order to make the mks TFT work for me

`/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200`

keep in mind I choose the 115200 baud rate because I configured the TFT screen software to use 115200

it was exactly opposite of what was mentioned before @Spawn32

T-Risely, 2020-01-23:
> I have got the answer,
> in order to make the mks TFT work for me
>
> `/**
>
> * Select the serial port on the board to use for communication with the host.
> * This allows the connection of wireless adapters [...]

When I compile with #define SERIAL_PORT_2 set to -1 I get a massive error ending with:
"expected initializer before '-' token"

It only compiles with SERIAL_PORT_2 commented out and still my MKS FT32 doesn't communicate :( 
Any ideas?