MarlinFirmware/Marlin #9634: [2.0.x] STEPPER_DIRECTION_DELAY exists, not implemented

thinkyhead, 2018-02-14:
While there is a `STEPPER_DIRECTION_DELAY` setting, currently it does nothing.

Is it needed, or can we drop it?

p3p, 2018-02-14:
Looking at a few (drv8825, 4988, tmc2130) datasheets worst case (drv8825) for Dir setup time is 650ns which is high enough to start causing issues in 32bit land depending on instructions between the pin toggle operations and mcu specifics, This also applies to Enable but I assume that's not as tightly coupled with the stepping. so .. in theory its needed at least.

smoki3, 2018-02-15:
is this maybe the problem for my not moving z axis with drv8825 on radds?

p3p, 2018-02-15:
@smoki3 It's unlikely, if this is an issue I think it will just be a single microstep in the wrong direction when changing directions, I'm sorry you haven't manged to solve that problem yet.

smoki3, 2018-02-15:
i dont know whats the problem. now i orded some new stepper drivers (tmc2130) i will try to replace the drv8825. maybe than its working for me. but the problem still exists

closed 2019-07-06 by shitcreek