MarlinFirmware/Marlin #9636: [BR] 1.1.8 - Progress Bar Not Working with SD Prints

houseofbugs, 2018-02-14:
- Description: 1.1.8 - Progress Bar Not Working with SD Prints
- Expected behaviour: Progress bar updates when printing from SD
- Actual behaviour: Nothing happens
- Steps to reproduce:
Flash 1.1.8 with SD and Reprap smart full graphics enabled. Print from SD

martinbogo, 2018-02-15:
Confirmed, the progress bar does not update on 1.1.8, when flashed onto my CR-10S

mylife4aiurr, 2018-02-17:
houseofbugs that avatar looks a lot like th3dstudio's logo - I'm telling.....  :)

On a serious note, they are going to ask if you see this problem in latest bug fix version. Because if not, its already fixed.

Nume1977, 2018-02-20:
Not working on latest 2.0.x!
SD Card percentage is also not showing anything (#define DOGM_SD_PERCENT)
Using Reprap smart full graphics

thinkyhead, 2018-02-20:
I'm unable to reproduce the issue using `bugfix-1.1.x` or `bugfix-2.0.x`. I see a working progress bar on the graphical display and the percentage is also displayed when printing a file from SD.

Compare my configurations to yours using a diff tool. Maybe I've got something in mine that's missing from yours. Here they are: 
[Configs-CR10S.zip](https://github.com/MarlinFirmware/Marlin/files/1739820/Configs-CR10S.zip)

I'm testing with this G-code:
```gcode
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
M300 S880 P50
G4 S1
M300 S660 P50
G4 S1
M300 S880 P50
G4 S1
M300 S660 P50
```

houseofbugs, 2018-02-22:
Its the lines in the ultralcd files that set the uint8_t string.

>           const uint8_t progress_bar_percent = card.percentDone();

I think we need to have that be dynamic based on the M73 being used or not. Like have it be on the SD and then when M73 is issued then swap it over to the manual mode.

closed 2018-02-23 by thinkyhead