MarlinFirmware/Marlin #9639: Is M500 really saving mesh?

batata004, 2018-02-15:
Hi,

I am using BILINEAR to map a mesh of my bed. I never had eeprom enabled so I enabled it now in MARLIN. So I booted up my printer, executed G29 and after it finishes I executed M500 to save data.

Then I turn the printer off. When I turn on my printer I only home the 3 axis and start printing and it's not good result. Do I need to execute any code, after turning on my printer, to make sure the mesh is loaded from the eeprom and used?

GMagician, 2018-02-15:
G29 S0 I suppose

batata004, 2018-02-20:
kkkk thanks!

closed 2018-02-20 by batata004

Roxy-3D, 2018-02-20:
Did you save your mesh with M500 ?     If not...   You probably don't have good mesh information.

batata004, 2018-02-20:
YEAP I followed your advice from another thread @Roxy-3D ! I saved then I loaded ;) thanks!