MarlinFirmware/Marlin #9640: RIGHT_PROBE_BED_POSITION  bigger than X_BED_SIZE + X_PROBE_OFFSET_FROM_EXTRUDER

batata004, 2018-02-15:
Hello!

In marlin firmware (using arduino ide) If I try to set RIGHT_PROBE_BED_POSITION to a value bigger than X_BED_SIZE + X_PROBE_OFFSET_FROM_EXTRUDER the arduino compiler prevents me from flashing the code. However I would like to probe 5cm more to the right cause I have room to move the x-carriage to the right. Of course the nozzle will be out of the bed but the probe will still be over the bed (my probe is mounted to the left of the nozzle). 

What is strange is that if I use the code below with a number in the R parameter bigger than X_BED_SIZE + X_PROBE_OFFSET_FROM_EXTRUDER I get no error. So what is the point of preventing user to set a bigger value in the firmware if later he can do that anyway?

G29 R180

MarioStoilov, 2018-02-15:
And what is the value of X_PROBE_OFFSET_FROM_EXTRUDER? If it is on the left, then it should be a negative number.

AnHardt, 2018-02-15:
> What is strange is that if I use the code below with a number in the R parameter bigger than X_BED_SIZE + X_PROBE_OFFSET_FROM_EXTRUDER I get no error. So what is the point of preventing user to set a bigger value in the firmware if later he can do that anyway?

It's not that strange to me.
It forbids to set wrong/unlikely _default_ settings, but allows to make an _exception_.
(If that will work is another thing.))

batata004, 2018-02-15:
Actually I am wrong, even sending the commands using USB connected to arduino mega it still does not allow me to probe further points!

MikeLud, 2018-02-17:
You need to increase your #define X_MAX_POS by 5mm located in Configuration.h see code below

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -38
#define Y_MIN_POS -7
#define Z_MIN_POS 0
#define X_MAX_POS 230
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 240

batata004, 2018-02-17:
That helps @MikeLud  BUT if a print is too big it will print out the area. If the print is too big but I have the X_MAX set right, it will print wrongly BUT inside the print area. It would be nice to have this poblems olved.

MikeLud, 2018-02-17:
The printer should only print in the area defined in the code below and also what your G-code is telling the printer. 
// The size of the print bed
#define X_BED_SIZE 220
#define Y_BED_SIZE 220

Also the code below needs to be set right so the printer knows where X 0 and Y 0 is on the print bed
// Travel limits (mm) after homing, corresponding to endstop positions.
 #define X_MIN_POS -38
 #define Y_MIN_POS -7

batata004, 2018-02-17:
@MikeLud YOU NAILED IT!!!!!!!!!!! THAT'S EXCTLY WHAT I NEEDED!!!!! THANK YOU SO MUCH!!

closed 2018-02-17 by batata004