MarlinFirmware/Marlin #9644: G29 response "OK" but hotend is not moving

raymondlml, 2018-02-15:
I flashed marlin 1.1.8 on my CR-10S with bltouch installed. Other commend such as G28, G29 P1 works great but not G29. 

When I use the terminal on Octoprint to send "G29" right after "G28", the hotend is not moving. According to what I found, the probe should move around an perform auto bed levelling. I am not sure what went wrong.

Configuration.h and Configuration_adv.h can be found here. Thanks in advance for all your help.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1727610/Config.zip)

GMagician, 2018-02-15:
UBL and only G29 does nothing, as I know...

http://marlinfw.org/docs/gcode/G029-ubl.html

Roxy-3D, 2018-02-15:
Correct.   A G29 with no other information is basically a NOP.

tcm0116, 2018-02-16:
@Roxy-3D is there any reason why UBL's G29 with no parameters couldn't go ahead and probe the entire bed and activate UBL? This with allow UBL to be compatible with the standard definition of G29. 

I realize a lot of the purpose of UBL is to provide a more sophisticated approach to bed leveling, but seeing as it's supposed to be "Unified", it should also encompass the basic function of the other leveling methods. 

I brought this up with @thinkyhead, and he seemed surprised that UBL's G29 with no parameters had no function.

thinkyhead, 2018-02-23:
> probe the entire bed and activate UBL

That would be a good default behavior. Is there any technical reason why it should not?

Roxy-3D, 2018-02-23:
Do you mean probe the portion of the bed that the probe can reach and then activate?

We could do that...    But is that a useful shortcut?   If that is what they wanted to do we only saved them from typing a P1 after the G29.     It maybe a more useful thing to assume is they want to do G29 P0 J4.   That would behave like the old school grid based leveling.   We could always add a #define to specify what the default G29 (with no parameters) is.      My vote would be it is G29 P0 J4 A

thinkyhead, 2018-02-23:
> But is that a useful shortcut?

It is a reasonable user expectation, as they say in the software biz.
Most users know about `G29` … but ask them what "`P4`" means.

Roxy-3D, 2018-02-25:
>> We could always add a #define to specify what the default G29 (with no parameters) is.

> Sure, I suppose so. People are clamoring for more options. This one is pretty concise.

How about we change the focus a little bit...   We can use this as a test case.   How about we have macro or a super small function that can tell us how many parameters have been specified for the command.   In this case, if the answer is zero, we just parse the specified, default command.

thinkyhead, 2018-02-25:
```cpp
if (!parser.seen_any()) SERIAL_ECHO("No arguments.");
```

Roxy-3D, 2018-02-25:
OK!

closed 2019-02-20 by thinkyhead