MarlinFirmware/Marlin #9648: M914 (TMC Bump Sensitivity) returns incorrect values when negative.

marcio-ao, 2018-02-15:
### Bug Report
- Description: M914 (TMC Bump Sensitivity) returns incorrect values when negative.
- Expected behaviour: It should print a negative value
- Actual behaviour: It prints a positive value
- Steps to reproduce:
  > M914 X-1
  < X driver homing sensitivity set to 127
  < Y driver homing sensitivity set to 4
  < E0 driver homing sensitivity set to 0

I've implemented a straightforward fix in our FW and am hoping it can be incorporated:

```
#define SIGN_EXTEND_SGT(sgt) int8_t(sgt | ((sgt << 1) & 0x80))

  template<typename TMC>
  static void tmc_get_sgt(TMC &st, const char name[]) {
    SERIAL_ECHO(name);
    SERIAL_ECHOPGM(" driver homing sensitivity set to ");
    MYSERIAL.println(SIGN_EXTEND_SGT(st.sgt()), DEC);
  }
```

marcio-ao, 2018-02-15:
Hummm, actually, looking back on this code I wrote long ago, I have no idea what the bit-shift is doing. I think it does nothing and all that matters is the int8_t ...

teemuatlut, 2018-02-15:
This really is something that should be done at the library level. I can't have the `sgt()` reporting values that aren't correct and I believe I've already fixed this once already.
Either way, I'm getting a very different output and that led me to [this oversight](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/tmc_util.cpp#L239).

marcio-ao, 2018-02-15:
@teemuatlut: I'm not sure why you would consider sign-extending a 7-bit value into an 8-bit value to be incorrect. I agree that this would be better done in the library, but no matter where you do it it is not incorrect.

teemuatlut, 2018-02-15:
"I can't have it report values that aren't correct"
As in if the current implementation is bugged and reporting bad values. It was not a comment on your suggestion!

marcio-ao, 2018-02-15:
@teemuatlut : Ah, gotcha :)

teemuatlut, 2018-02-15:
After fixing the argument data type: 
```
SENDING:M914 X-1
X driver homing sensitivity set to -1
>>> M914 X-60
SENDING:M914 X-60
X driver homing sensitivity set to -60
>>> M914
SENDING:M914
X driver homing sensitivity set to -60
>>> M914 X10
SENDING:M914 X10
X driver homing sensitivity set to 10
```

closed 2018-02-23 by thinkyhead