MarlinFirmware/Marlin #9649: X and Y parameter does not work in G29 with bilinear

batata004, 2018-02-15:
Hi,

I am using BILINEAR however when I execute the code below to probe squares of 5mmX5mm it does not work, it only probes 3 points in each row and 3 points in each column (cause I configured marlin firmware to probe only a mesh of 3x3).

G29 X5 Y5

Why? The documenttion at http://marlinfw.org/docs/gcode/G029-abl.html says I can set X and Y options.

filipgoc, 2018-02-20:
I'm afraid the documentation was wrong @batata004.

The P and X and Y flags are only for `Linear ABL.` 

It only mentioned the the `Bilinear ABL` as capable of this once, in the example code at the end. 

We have fixed the documentation now with commit [4808f4b](https://github.com/MarlinFirmware/MarlinDocumentation/pull/131/commits/4808f4bead41e2f3fb693116461de4230041166f)

filipgoc, 2018-02-20:
I suggest to close this issue.

batata004, 2018-02-20:
THANKS! I will close.

closed 2018-02-20 by batata004