MarlinFirmware/Marlin #9652: Quick_Home and Sensorless Homing TMC2130 not working together

workheart, 2018-02-15:
First: Great Work everybody involved. I'm using Marlin since... ever. realy Happy.

SETUP:
DIY prusa alike reprap Printer, Ramps 1.4 With TCM2130 X+Y+Z and DRV8825 E0+E1 All axis Belt Driven. Marlin Bugfix 2.0.x from 13.2.18
EXPECTED BEHAVIOR:
G28 XY Would Home X and Y
OBSERVED BEHAVIOR:
XY Crashing in the End till power pluged or software endstops stopp movemend
REPRODUCE:
Working Marlin 2.0.x configuration With Working Sensorless Homing 
uncoment //#define QUICK_HOME  Line356 in Configuration_adv.h
SOLUTION:
none. yet.
WORKAROUND:
Don,t Use QuickHoming with Sensorless Homing. its not too Importend I Think, so first I would Add a Comment to Sensorless Homing like: 
'`//#define QUICK_HOME                     // If homing includes X and Y, do a diagonal move. Does not work With Sensorless Homing TMC2130'`

my config:
https://pastebin.com/bZkdfBcS
https://pastebin.com/H8MsH8Hi

https://github.com/MarlinFirmware/Marlin/blob/0640bd028215a01ff9a01a46d4dcb859ba208a20/Marlin/Configuration_adv.h#L356

teemuatlut, 2018-02-15:
Yes, I was made aware of this some time ago. I'll try to remember to take a look at this whenever I get to fixing the Y movement when homing for the first time.

thinkyhead, 2018-02-23:
This should now be patched. Please try with the latest code.

closed 2018-02-23 by thinkyhead