MarlinFirmware/Marlin #9658: Printers almost always, crash, or reset after the first couple layers

Jasonpgr44, 2018-02-15:
-----------------------
### Bug Report
- Description: --- I just finished my first big fancy printer build after playing with an old folgertech 2020 I bought used about a year ago. I've swapped out parts and recomplied and edited the firmware for the folgertech several times with minimal issues, and it's helped me to learn a lot about how it all works, but I'm not a pro by any means, so please feel free to respond as if speaking to child so I can be sure to wrap my head around what is going on.

The new printer is a diy build of a scaled up version of the D-Bot design on thingy verse using a 500x500 printer area on a 3/8" thick sheet of aluminum, with a 1500 watt silicone heated bed (120v AC power with a solid state relay from the heated bed output) stuck to the bottom and 4 lead screws,  using one 92 oz Nema 17 stepper per pair of screws.

I initially tried running it using two Z endstops, with one on each side, controlling them independantly and adding a BL touch to to finalized leveling, but nothing I tried could get that mess of Z axis bits to work together properly. So I tried simplifying everything and I am now using the BL touch as the Z endstop as well as leveling probe for bilinear leveling. and that seems to work beautifully, I got the z offset all dialed in, and the printer was off and running fantastically, first layer was the most consistent and beautiful thing I've ever printed. I left for a couple hours and came back and saw the machine sitting totally idle and cold. I did some research and I figured maybe I was using too much processing power or memory or something by having so many features going. So I cut the microstepping from 32 down to 16 on the Z axis, and removed the SDcard support and the display from the code to free up as much as possible. I also previously broke my full graphics controller display and never really use it anyway because I control the printer through Octoprint anyway.

So I recompiled last night, using the latest 1.1.8 firmware along with those changes, and the thing still froze up on me after the first few layers. Once again, I wasn't at home when it stopped so I don't know if anything popped up at the time, but I didn't see anything in the terminal on Octoprint. It was just sitting there, heaters still on, not moving at all, and the power had been cut to the steppers. 

This is an issue I've been having intermittently with all the printers I've ever had. Certain files will slice up and print just fine, and others just wont. No matter how many things I try to change, those print jobs just can't ever make it past the first handful of layers.  

- Expected behaviour: --- Complete the print job
- Actual behaviour: --- Takes a nap, or resets mid print
- Steps to reproduce:
  - No idea how others could reproduce it, I'm going to include a copy of the gcode file I was trying to print, as well as the config and config_adv files for you to look at though. This issue has been ongoing for about a year now off and on, and now that I have this giant and super impressive printer working, I really need to get rid of this bug.
  - Do that
[Marlinfw zip.zip](https://github.com/MarlinFirmware/Marlin/files/1729242/Marlinfw.zip.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

workheart, 2018-02-15:
what are you printing with? best thing for this is some LOG file, 
you can enable logging in most programms octoprint etc (dont forgett to turn off)
2. Double Check all cables. While moving the Printer arround
3. Try to reproduce.. make big Prints without Fillament (so you dont waste some) and set printingtemp. so 30°C or so

Jasonpgr44, 2018-02-15:
Slicing with cura, then sending the cgcode file to the printer via octoprint. I just did a print of a small single piece with no issues at all. There's a little more stringing than I like, but I can dial that in later. I inspected the cables and movement during that print and didn't see anywhere they could be shorting out so everything should be fine on that front, it's all very well secured and nothing is showing that shouldn't be. 

How do I go about enabling and doing the logging? I've seen it mentioned in the forums a lot, during my research of the problem, but I've never done it before nor seen it as a clearly described option. 

Is there a gcode command I issue and it turns it on? Or do I need to go back and edit the config file and recompile and upload? If that's the case, where abouts would I find the option I need to enable? 

Seeing as how the small prints seem to work just fine, I'm still leaning towards it being a processor or memory issue. I've seen some big fancy 32bit board like the smoothieware and duetWifi things that look like they should be able to handle the load a lot better, but I'm hoping to avoid making that change because I just finally got my head wrapped around the electroncis for the Rumba board I'm using and the arduino stuff and I really like the way Marlin is organized and makes sense of everything. So changing out the board will be a worst case scenario.

Jasonpgr44, 2018-02-16:
I did some digging around in Octoprint and I found an option for "log communication to serial.log (might negatively impact performance)" Is that the option I need to check? If so, then I try a big print, wait for the machine to fail, download that log and upload it here? 

I also looked around in Marlin and the only logging feature I could find was a debug logging for auto leveling and homing I think. I didn't see a general log thing in there.

workheart, 2018-02-16:
You do that in octoprint. 
In it loggs everything you sent live in the terminal tab in octoprint. Wasn't there some error in octoprint like "error mintemp triggerd" or stuff? 
Printer stopped mid air or print continued without extruding? And no error code?

landodragon141, 2018-02-18:
Are you having a thermal run away from your part cooling fan? I've had this happen to me more times than I can count when I forget to reduce the max fan speed in my slicer.

Jasonpgr44, 2018-02-18:
Printer just stopped on the print right where it left off. It looks as if it was printing, and there was a power outage, and the power came back on after a couple minutes. Or someone hit reset on the board or something. 

I haven't seen any errors on the screen or in the terminal, but I've never been right next to it when it does the freeze/reset thing. I don't think it's a thermal runaway situation because it doesn't have any issues at all will smaller or simpler prints. But when I do big prints, or prints with lots of parts spread out on the print bed, it can't make it through the print. I also tried a big spiral vase the other day to try out some new PETG filament, and it couldn't finish that print either. I was reading about another person having issues with prints with too many big arcs, or something like that. I didn't really understand it all, but their fix was freeing up memory and processing power by shutting off the LCD. Didn't fix my problem, but it shows a lot of the same symptoms.

I'm going to try a big ole print tonight/tomorrow sometime with logging enabled it Octoprint, see if that sheds any light on the situation.

Jasonpgr44, 2018-02-18:
So while trying to finish up a small print before doing my test with the big item and logging. And it made it ALMOST all the way through the print. Octoprint says it was around 94% percent. The status in Octoprint still said printing, but it wasn't sending any new movement commands. It was just echoing back all the standard temp messages and all that. 

I scrolled up to the point where it looks like it stopped sending movement commands and switched over to the little send/receive temp status stuff when the printer is sitting idle. Oddly, the heaters are still on, it's just not moving or feeding anymore. I copied that section from the terminal and this is what it said. No error messages, nothing in Octoprint, I can't figure it out.

Send: N203868 G1 X253.247 Y228.999 E14062.94063*88
Recv: ok
Send: N203869 G1 X252.964 Y228.999 E14062.9431*101
Recv: ok
Send: N203870 G1 X252.681 Y228.999 E14062.94557*92
Recv: ok
Send: N203871 G1 X252.398 Y228.999 E14062.94804*91
Recv: ok
Send: N203872 G1 X252.115 Y228.999 E14062.95051*86
Recv: ok
Send: N203873 G1 X251.832 Y228.999 E14062.95298*95
Recv: ok
Send: N203874 G1 X251.55 Y228.999 E14062.95545*102
Recv: ok
Send: N203875 G1 X251.267 Y228.999 E14062.95792*92
Recv: ok
Send: N203876 G1 X250.984 Y228.999 E14062.96039*93
Recv: ok
Send: N203877 G1 X250.701 Y228.999 E14062.96286*89
Recv: ok
Send: N203878 G1 X250.418 Y228.999 E14062.96533*84
Recv: ok
Send: N203879 G1 X250.135 Y228.999 E14062.9678*101
Recv: ok
Send: N203880 G1 X249.852 Y228.999 E14062.97027*88
Recv:  T:199.93 /200.00 B:77.38 /75.00 T0:199.93 /200.00 T1:29.56 /0.00 @:56 B@:0 @0:56 @1:N2ok
Recv:  T:200.18 /200.00 B:77.11 /75.00 T0:200.18 /200.00 T1:29.66 /0.00 @:50 B@:0 @0:50 @1:0
Recv:  T:200.36 /200.00 B:76.77 /75.00 T0:200.36 /200.00 T1:29.53 /0.00 @:46 B@:0 @0:46 @1:0
Recv:  T:200.18 /200.00 B:76.37 /75.00 T0:200.18 /200.00 T1:29.62 /0.00 @:52 B@:0 @0:52 @1:0
Recv:  T:199.97 /200.00 B:76.04 /75.00 T0:199.97 /200.00 T1:29.50 /0.00 @:58 B@:0 @0:58 @1:0
Recv:  T:200.09 /200.00 B:75.63 /75.00 T0:200.09 /200.00 T1:29.72 /0.00 @:53 B@:0 @0:53 @1:0
Recv:  T:199.93 /200.00 B:75.30 /75.00 T0:199.93 /200.00 T1:29.44 /0.00 @:57 B@:0 @0:57 @1:0
Recv:  T:199.80 /200.00 B:74.85 /75.00 T0:199.80 /200.00 T1:29.37 /0.00 @:60 B@:0 @0:60 @1:0
Recv:  T:200.18 /200.00 B:74.57 /75.00 T0:200.18 /200.00 T1:29.47 /0.00 @:49 B@:0 @0:49 @1:0
Recv:  T:200.49 /200.00 B:74.17 /75.00 T0:200.49 /200.00 T1:29.59 /0.00 @:42 B@:0 @0:42 @1:0

Jasonpgr44, 2018-02-18:
Didn't even make it through the first layer when I tried a big print. I attached the two logs that said were modified today. Hopefully someone can make some sense of what's going on. It ended up freezing again with the heaters still on, just stopped all movement.

[octoprint.log](https://github.com/MarlinFirmware/Marlin/files/1734545/octoprint.log)
[serial.log](https://github.com/MarlinFirmware/Marlin/files/1734546/serial.log)

AnHardt, 2018-02-18:
From the posted serial log:
```
2018-02-18 08:57:41,278 - Recv: ok
2018-02-18 08:57:41,281 - Send: N7806 G0 F4800 X334.651 Y210.2*124
2018-02-18 08:57:41,295 - Recv:  T:204.87 /205.00 B:77.73 /75.00 T0:204.87 /205.00 T1:33.69 /0.00 @:55 B@:0 @0:55 @N780ok
2018-02-18 08:57:43,285 - Recv:  T:204.96 /205.00 B:77.27 /75.00 T0:204.96 /205.00 T1:33.71 /0.00 @:53 B@:0 @0:53 @1:0
2018-02-18 08:57:45,285 - Recv:  T:204.82 /205.00 B:76.81 /75.00 T0:204.82 /205.00 T1:33.71 /0.00 @:55 B@:0 @0:55 @1:0
```
From the example above:
```
Recv: ok
Send: N203880 G1 X249.852 Y228.999 E14062.97027*88
Recv: T:199.93 /200.00 B:77.38 /75.00 T0:199.93 /200.00 T1:29.56 /0.00 @:56 B@:0 @0:56 @1:N2ok
Recv: T:200.18 /200.00 B:77.11 /75.00 T0:200.18 /200.00 T1:29.66 /0.00 @:50 B@:0 @0:50 @1:0
Recv: T:200.36 /200.00 B:76.77 /75.00 T0:200.36 /200.00 T1:29.53 /0.00 @:46 B@:0 @0:46 @1:0
```

Octoprint could not see the ok at the end of the third row and for that did not send the next command.

Why the 'ok' isn't at a seperate line is much harder to find.
Line 3a is not finished.
We have a strange 'N780' ('N2').

AnHardt, 2018-02-18:
There is a merged workaround for this in the bugfix branches. Do they help?

briareos77, 2018-02-18:
@AnHardt Could you give a link to the merged workaround for this? I was not able to find it.

Thanks.

workheart, 2018-02-18:
Maybe bad USB cable try lower baudrate...  I got communication timeouts once

AnHardt, 2018-02-18:
Pauses after printing the temperatures. Can't remember the numbers.

Jasonpgr44, 2018-02-18:
I doubt it’s a usb cable. I’ve tried 4 different ones, ranging from 12” up to 5 feet. I tried a small print again last night, just one of the pieces I crammed onto the build plate in the big print, and that one did the same thing after just a couple hours. Got up to around 1 1/2 inches tall out of 5. Just stopped moving, and left all the heaters on.

workheart, 2018-02-19:
its a long shoot... corrupted Gcode? can you Find the posision where the crash happens? file corruption Bad micro sdCard?
What about Driver temp? 
you can also try using "fake ok" option in octoprint. its in the Terminal tab under "advanced" on the verry bottom

Jasonpgr44, 2018-02-19:
It has done it on multiple sets of gcode, so I doubt it's corrupted since I've tried exporting from cura and repetier with same results. I can try bypassing octoprint and the pi all together and do a big print straight from the laptop via USB rather than Octoprint. Though I have another Octoprint running my little printer that has been doing a print for almost 30 hours straight now with no issues at all. It could be something weird in the micro SD though I suppose. 

I think I'll try the "fake ok" option first, then try the laptop to stream the print if that fails. I was looking at that option in there but it has a warning like "don't do this, it's dangerous" or something like that under it. I'll give that a go, and report back shortly.

thinkyhead, 2018-02-25:
Try lowering your max feedrates a bit and see if it frees up some resources and makes the machine more reliable. Do some tests moving Z as fast as it will go and see if this increases the error rate. Do some fast diagonal moves with X, Y, and Z together and see if it crashes or reboots the machine.

If you don't need Z to be accurate to 1.3µm (750 steps per mm) try cutting your Z micro-steps in half with the jumpers and then use `M92 Z375`.

Jasonpgr44, 2018-02-28:
And I finally thought that would do the trick. I lowered my max feed rates by about 30%, cut my microsteps in half on the z axis, and tried running a print off the laptop bypassing the pi all together. but alas. Frozen again mid print. Made it about 1 inch total up the print.

At this point I'm feeling like it has to be some kind of problem with the board. I'm seriously debating getting a duet wifi. Just sucks because I already have so much time and so many hours into this beastly printer.

[image]

thinkyhead, 2018-02-28:
If Marlin gets low on CPU resources it will always reboot. If there's a freeze, it does usually indicate something more weird is going on.

Jasonpgr44, 2018-02-28:
Well, it looks like last nights print failure was actually due to user error. I didn't plug the power supply to the laptop in all the way, so the printer didn't fail, the computer just went to sleep. I touched the keyboard and it took right off and moving again. So it made it all the way through this print, which is fairly large, but since it's just an outer spiral, it's not a lot of movement or material. I'm going to try another bigger one tonight with the new settings and see if my luck continues.

[image]
[image]

workheart, 2018-02-28:
pretty clasic user error ^^

thinkyhead, 2018-02-28:
@Jasonpgr44 — Thanks for the update. Good luck with tonight's print!

boelle, 2019-02-20:
@Jasonpgr44 still having issues?

closed 2019-05-16 by thinkyhead