MarlinFirmware/Marlin #9659: Curiosity question - high step rate system info/examples

Bob-the-Kuhn, 2018-02-15:
I keep seeing the 32 bit controllers being touted as being able to achieve higher step rates.  I'm wondering what is needed to take advantage of the higher step rates.

Does anyone have real world examples?

I did a little playing with the LPC1768 and it appears that higher acceleration rates are needed or else the steppers rarely get to cruising speed before they need to decelerate.

How are they getting the higher steps per mm?  Please assume no micrstepping.

What steppers are being used?  The usual hobbyist sources seem to top out about 2K RPM (6.7K steps per second).  50K steps per second would need a 15K RPM stepper.

The higher speeds mean lower torque.  Does that mean delta systems are needed because of the lower mass of the head assembly?

teemuatlut, 2018-02-15:
To put it into another perspective
150 mm/s = 12000 steps/second = 225RPM (80 steps/mm, 200 fullsteps/rev).
I'm fairly confident in those numbers. How did you get yours?

TMC2130 has a max f_FS of around 24kHz.

I always thought that higher stepping rates would allow us to use higher microstepping settings. Whether or not this is necessary is another matter however.

thinkyhead, 2018-02-23:
> rarely get to cruising speed before they need to decelerate

Makes sense. To reach a higher speed in the same (or less) time requires higher acceleration. As rates climb to crazy levels it becomes preferable to use an S-shaped rather than trapezoidal ramp.

AnHardt, 2018-02-24:
What does it need to print fast: (random order, incomplete)
1 Low mass to accelerate (F = m*a)
2 Gearing to balance torque with RPM
3 Steppers with high RPM
4 Stepper drivers allowing high frequent step signals
5 'S', cos() or multistep acceleration profiles to really get the motors max RPM
6 CPU able to output high step rates, without bad timed double/quad stepping.
7 Well shaped stepper pulses with ~50/50 duty cycle.
8 CPU able to plan long buffers. Long buffers means, enough to plan the top speed to null deceleration even with short segments. Currently we need `MIN_STEPS_PER_SEGMENT` because we can't plan faster. (improved resolution). Currently planning/optimizing more moves in the buffer needs more time - usable buffer length is limited by CPU.

Currently, for me, a faster CPU is most valuable in planning not in stepping. We have so many 'artificial' limits we can shift a step further.

AnHardt, 2018-02-24:
9 Constand and low friction.
10 Extruder able to melt high amounts of material.

Hywelmartin, 2018-02-24:
12 cooling the printed part accurate

closed 2018-05-20 by Bob-the-Kuhn