MarlinFirmware/Marlin #9660: CR10 Compile Errors (Debian)

plainpylut, 2018-02-15:
### Compile Errors
When I compile for `CR10` I get errors:
```
Printer: Creality CR-10 Mini (300mm x 220mm x 300mm)

These are the Components in my CS-10 Mini
https://www.ebay.ie/itm/Creality-3D-CR-10-3D-Printer-Control-Mainboard-Upgrade-Circuit-Board-Control-/253193655402
https://www.aliexpress.com/item/Universal-LCD-12864-3D-Printer-Display-Screen-With-Encoder-For-CR-10-CR-7-Model/32833148327.html

For Creality CR-10 Printer in Marlin-bugfix-1.1.x:

Subdirectory:  /Marlin-bugfix-1.1.x/Marlin/src/config/examples/Creality/CR-10/

File:  _Statusscreen.h contains improperly formatted structures, which cause compile errors (Debian):

Line 54:
B00000000,B00111110,B00001111 B11110000,...........

Change to:
B00000000,B00111110,B00001111,B11110000,...........

Line 75:
B00000000,B00111110,B00001111 B11110000,...........

Change to:
B00000000,B00111110,B00001111,B11110000,...........

File:  _Bootscreen.h for my CR-10 Mini is missing from CR10 and CR-10S subdirectories:
Add it back to the subdirectories.  It will be a Generic screen; changeable by the user.
Add README.md in both subdirectories to provide information where the two header files get copied (Marlin/src).

Subdirectory: /Marlin-bugfix-1.1.x/Marlin/src/lcd/dogm/
 
File:  ultralcd_st7920_u8glib_rrd_AVR.cpp contains an optimize statement that causes errors on the LCD display.

Line 46:
#pragma GCC optimize (3)

Change to:
// #pragma GCC optimize (3)

I'm new to GitHub and 3D printing, however I have over 23 years experience in real-time software engineering (Flight Simulation).  I'd like to make the pull request and get these changes applied to the Marlin bugfix1.1.x repo.

I have a CR-10 to test the compiled code.

My 'first small step on using Git'!

Bob-the-Kuhn, 2018-02-15:
You can make those changes yourself in your local copy of Marlin and then compile & test.

To move the tested code into Marlin you'll need to follow the PR process.  See the [Marlin docs site](http://marlinfw.org/) for details on how to do that.  It's under Documentation -> Development.

plainpylut, 2018-02-16:
Mr. Bob,

I already have compiled and tested the code changes.
I thought I was doing this right?  I have been watching youtube videos on Git processing of files.
I'll read the Marlin docs site.  Thanks for the comment.
I have 23+ years experience using Configuration management is a much stricter environment.  Git isn't what I'm used to using in a CM environment; but that is not an excuse..

Bob-the-Kuhn, 2018-02-16:
There's lots of ways to get to a useful result.  Don't worry about being perfect when starting into the 3D printer software.  We've all been there.

> /Marlin-bugfix-1.1.x/Marlin/src/config/examples/Creality/CR-10/

That's not the file structure used by Marlin for the bugfix-1.1.x branch.  It is the one used for the bugfix-2.0.x branch.  Did you get your code from the bugfix-2.0.x branch or the CR-10 site or ...?

I tried the CR-10 example configuration in our bugfix-1.1.x.  I saw the line 54 & line 75 errors. and made the same changes as you.  Now it's complaining about dimensions in another file. I'll look into it further tomorrow.

plainpylut, 2018-02-16:
Mr. Bob,

Well, I think the code may have come from the 2.0 branch.  Git must be a lot smarter than me.
I’m struggling with trying to learn ‘GitKracken’ and how Git works.
I ‘forked’ the bugfix-1.1.x code from Chrome (at least, I think I did).  
I use Windows 10 and today I’m going to attempt using the Git shell and GitHub to do a pull request.

If you want to make the changes, go ahead and make them.  I can provide the _BootScreen.h file that ‘looks’ good running in the CR10.

Garry

From: Bob-the-Kuhn 
Sent: Thursday, February 15, 2018 10:32 PM
To: MarlinFirmware/Marlin 
Cc: UnknownRider ; Author 
Subject: Re: [MarlinFirmware/Marlin] CR10 Compile Errors (Debian) (#9660)

There's lots of ways to get to a useful result. Don't worry about being perfect when starting into the 3D printer software. We've all been there.

  /Marlin-bugfix-1.1.x/Marlin/src/config/examples/Creality/CR-10/

That's not the file structure used by Marlin for the bugfix-1.1.x branch. It is the one used for the bugfix-2.0.x branch. Did you get your code from the bugfix-2.0.x branch or the CR-10 site or ...?

I tried the CR-10 example configuration in our bugfix-1.1.x. I saw the line 54 & line 75 errors. and made the same changes as you. Now it's complaining about dimensions in another file. I'll look into it further tomorrow.

Bob-the-Kuhn, 2018-02-16:
Yes, please post the corrected file.

plainpylut, 2018-02-17:
[CR-10.zip](https://github.com/MarlinFirmware/Marlin/files/1733178/CR-10.zip)

Mr. Bob,
My twin brother has me all confused.  He was using the 'easyconfig' files and telling me they were Marlin-1.1.8!

He has compiled the real Marlin 'bugfix-1.1.8' code using these 2 files.  The display looks good and the code just needs calibration for his printer.  The file ultralcd_st7920_u8glib_rrd.h worked fine with the
#pragma GCC optimize (3) in it.

I'm slowly understanding Git, but I haven't found a youtube video that unlocks it for me.
That is a terrible thing to say for me, since I worked as a Lead SW Engineer and thought i understood CM!  :(

thinkyhead, 2018-02-19:
@plainpylut — A good introduction to Git branching and merging… https://learngitbranching.js.org/

Bob-the-Kuhn, 2018-02-26:
They've made some CR10 changes.  Please try the latest & let me know if this has been resolved.

thinkyhead, 2018-02-26:
It is patched now!

closed 2018-02-26 by thinkyhead

plainpylut, 2018-02-27:
Bob,

Larry relayed this:

“I downloaded the Marlin bugFix-2.0.x tonight 02/26/2018 around 18:00 Phoenix time.

I just copied it to a testM20 subdirectory in my Sketchbook for Arduino 1.8.

I also copied both configuration files and both _Bootscreen.h & _Statusscreen.h.

from the ~/Creality/CR-10mini folder.

The code built properly, uploaded to my CR-10, and booted properly, displaying both

boot screens.  The SDCard even works, so that error is corrected too.  I'm heating 

my Hot End to do a test print.

THANK YOU GUYS!

Larry” 

He also downloaded the Bugfix-1.1.8 and compiled it, uploaded it, and tested.  It displays a lower case ‘t’ on the _Bootscreen;
everything appears to work.  He did see a ‘low memory’ warning during compile.  The Bootscreen with ‘t’ is Tevo Tarantula (I think). 

Garry

From: Bob-the-Kuhn 
Sent: Monday, February 26, 2018 9:45 AM
To: MarlinFirmware/Marlin 
Cc: UnknownRider ; Mention 
Subject: Re: [MarlinFirmware/Marlin] CR10 Compile Errors (Debian) (#9660)

They've made some CR10 changes. Please try the latest & let me know if this has been resolved.

Bob-the-Kuhn, 2018-02-27:
Great!

thinkyhead, 2018-02-27:
> The Bootscreen with ‘t’ is Tevo Tarantula

I dunno. I'm sure I've seen that 'T' someplace else… Hmm…

commit 114e9b0f (2018-02-26): Fix CR-10 status screen missing commas

Fix #9660
commit cd7dc495 (2018-02-26): Fix CR-10 status screen missing commas

Fix #9660