MarlinFirmware/Marlin #9662: FR: Force Home X before Home Y

smoki3, 2018-02-15:
I have a feature request: Can we create an option to force home X before Y (and Force home Y before X)?
I have a fix endstop on my core xy which need to home y before x.

GMagician, 2018-02-15:
Have you tried HOME_Y_BEFORE_X in configuration_adv.h?

smoki3, 2018-02-15:
Yes but if I only home x my printer is crashing, because the y axis has to be homed first that my hotend can reach the x endstop

GMagician, 2018-02-15:
I don't understand... if homing X before Y crash define is for homing Y before X...

workheart, 2018-02-15:
he means like "safe Homing" feature for Z
X can only be homed if Y is homed already

GMagician, 2018-02-15:
Ah ok you want to inhibit x homing until Y has been done, not home one before the other...

smoki3, 2018-02-16:
to change the order when i home ALL there is the option HOME_Y_BEFORE_X. And everything is fine. 

but if i want to home X my printers Y axis has in to be in the home position. otherwise the x axis cant have contact to his endstop and it will end in a crash. 

I now simply added a 'HOMEAXIS(Y);' to 'Marlin/src/gcode/calibrate/G28.cpp' in home (x) part. so its working now but it would be nice to have it in the config

studiodyne, 2018-02-16:
You have  G28 X or G28 Y 
Create a custom user command ' in config adv ' you create 5 scripts

And you make the script you want , in your situation , you can make '  G28 X\nG28Y\G28Z '

smoki3, 2018-02-16:
yes thats a workaround :D @studiodyne but if you click home x on display the printer crashes. So I integrated it to the firmware. maybe i upload the code.  mk4duo also has this funktion

studiodyne, 2018-02-16:
Ok To home x , the printer need to have a special Y POS , and the pos is the Y=homed
And if Y homed , but not a the pos , if you want to home x , THE WORLD EXPLODES
oK , but homing is just before the print  , and just made once 

Rewrite G28 CPP , and   Home Y + Home x but ' if Y homed --> Y go to home_x_Pos 
But it's individual coding , don't need to be in config files , because each new updated release you will have to rewritte G28cpp---
Not a good idea , because each week a new release have so many fixes , and you will never follow us , if you need to add your own feature

I repeat ,  PLACE YOUR SWITCH a the better place it need to be lolll

smoki3, 2018-02-16:
i created a pull request for this

https://github.com/MarlinFirmware/Marlin/pull/9676

Studiodyne, 2018-02-17:
If the PR is pushed , your problem is closed 😸

closed 2018-03-07 by thinkyhead