MarlinFirmware/Marlin #9673: [FR] IO pin Multiplexing

Iamturbo1978, 2018-02-16:
I'm just throwing this idea out there

With the increasing number of advancements in 3D printing hardware, I think we could hit a wall when it comes to Arduino IO pins. For example, I have an Azteeg x3 PRO with a VIKI2 display and (6) TMC2130 stepper drivers. That is a total of 8 parts on the SPI bus. I have run out of easily available (no soldering) IO pins on my board to add additional hardware. 

So my thought was to multiplex the chip select pins with either a logic decoder (a 3:8 decoder for example), or use a shift register. If a shift register is used, it could also be used to control the stepper motor enable lines. I would say any Adrunio output pin that does not need a fast response (using the term lightly) could be controlled by a shift register instead of taking up an Arduino pin.

Maybe there is something already in Marlin, and I just don't know about it.

Thanks
Anthony

tcm0116, 2018-02-16:
I've actually been thinking about doing something like this. The issue is that the CS pins are actually enabled/disabled in the TMC2130Stepper library and not within Marlin itself. As such, this capability would have to be incorporated by @teemuatlut into the TMC2130Stepper library directly.

teemuatlut, 2018-02-16:
A bit niche but not unreasonable.
Perhaps the interface could be something like
```cpp
TMC2130Stepper stepperX(dataPin, clkPin, latchPin);

// Tell the library instance the current value in the shift register.
stepperX.shiftRegister(value_held_by_Marlin);
// Library bit bangs the new register value and latches it.
```
Then Marlin could read back the shiftRegister value but that should be optional as the library will toggle the CS pin LOW and then back HIGH again, meaning no change happens.

archi, 2018-02-20:
I would "tentatively" second this - I'm planning (long term) to do some work on a cheap 32 bit board and would like to add good support for TMC2130 and similar, but the MCU (STM32F1) doesn't have enough IO pins without using some tricks - though I would use a 16 port I2C I/O expander instead of a shift register. Those can be programmed to send interrupts via a separate pin, which might be useful if endstops are "hidden behind" that IC.

Shawnb03, 2018-02-21:
Please take a look at this board and tell me what you think. It has the same exact footprint as the Mega 2560 and the Arduino Due. It also comes with a ethernet port and a ton of extra pins that possible could solve this problem.

Here is the link to it. I am only posting this because I am interested in checking this out and giving it a try.

https://www.digikey.com/products/en?mpart=ATSAME70-XPLD&v=150

archi, 2018-02-21:
@Shawnb03 That's directed at whom? In case you asked me to take a look, I wouldn't personally compare a 37US$ board with those <3US$ STM32F1 boards. Of course it's always possible to use a more expensive IC with more IOs ;-)

Shawnb03, 2018-02-21:
It was really just a open comment. I am just looking for peoples opinion on this idea of mine. I would love to try this board out, due to it having massive power to run pretty much anything 3D Printer related.

Bob-the-Kuhn, 2018-02-21:
FYI - the Due's processor already has a hardware mux function for SPI selects.  It'll put out a 4 bit code on I/O pins so it can drive up to 15 SPI devices off of the hardware SPI.

 I'm hoping to see LPC1778 based boards come out. More I/O pins and lower cost than the LPC1769/17678 along with EEPROM.

Bob-the-Kuhn, 2018-02-21:
@Shawnb03 - the problem with a board like that is hooking up all the I/O.  You're money ahead to select an existing controller with the features you want.

As to processing power, I haven't heard any computing power complaints about any of the 32 bit processors.

Bob-the-Kuhn, 2018-02-22:
The Due has four total UARTs of which one is dedicated to the USB programming port.  The others are on logical pins 14-19.

According to the RADDS schematic I have, only one of the remaining three are available.  The I/O pins for the other two are dedicated to other functions on the board.  The **pins_RADDS.h** file confirms this.

[image]

boelle, 2019-02-20:
@thinkyhead if this is a feature request (hard to judge) maybe add the label?

boelle, 2019-06-22:
missing label

gDanix, 2021-05-12:
I would like to resucitate this FR, because it would be really useful to have a shift register with all the mostly-static, output-only things connected there: RGBW leds (4 pins), beeper, driver's EN pins, BLTouch servo pin (maybe?), etc...

I've taken a look at the code, and maybe it can work by modifying the FastIO's _WRITE(IO, V) macro to include a new case in the conditional to manage the shift register by writing values in one pin, and clock pulses in another pin.

Could this work? After "calling" WRITE it's assured that the physical pin is already updated, or it's expected to be a delay?

enwi, 2022-04-11:
I also have a lot of interest in this feature. I designed a Pick and Place machine PCB using the ESP32 to control everything. Due to the limited IOs I had to add 4 shift registers to be able to control 6 TMC2130 motor drivers and an additional 4 MOSFET outputs. I would be awesome to be able to use Marlin to control my PnP.

hsyngndz85, 2024-05-15:
The output pin for ESP32 has been resolved. Has the input multiplexing issue with the I2S port been resolved? It would be better to use endstops and external inputs.