MarlinFirmware/Marlin #9677: TMC2130 driver issues

martind-n, 2018-02-16:
Hi,

I found a coule of inputs for similar problems but I couldn't find the same I'm having.

I decided to change my old Pololu drivers for the new TMC2130.

Everything seemed to be fine but while trying to start moving the printer, the problems started.

Basically, I made all the marlin set up which means enabling TMC2130 support, changing CS pins (I'm using an LCD screen on AUX3), setting currents, etc.

Everything seems to be fine while sendig M122

		X	Y
Enabled		false	false
Set current	0	0
RMS current	1767	1767
MAX current	2491	2491
Run current	31/31	31/31
Hold current	31/31	31/31
CS actual		31/31	31/31
PWM scale	1	1
vsense		0=.325	0=.325
stealthChop	true	true
msteps		16	16
tstep		1048575	1048575
pwm
threshold		0	0
[mm/s]		-	-
OT prewarn	false	false
OT prewarn has
been triggered	false	false
off time		5	5
blank time	24	24
hysterisis
-end		2	2
-start		3	3
Stallguard thrs	0	0
DRVSTATUS	X	Y
stallguard
sg_result		0	0
fsactive
stst		X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
	X = 0x80:1F:00:00
	Y = 0x80:1F:00:00

First of all, I don't know why "set current" is 0, since I set this up in 600. Then I saw the following in the start routine from pronterface.

echo:Stepper driver current:
echo:  M906 X 0 Y 0

I don't know why this is being set to 0.

Anyway, I tried to move my Y and X axis. Here the real problems started. Only one motor moves (It's a core XY printer). And after a two/three movements the printer crashes

00:00: X driver overtemperature warning! (0mA)
00:00: Y driver overtemperature warning! (0mA)
X driver error detected:
		X	Y
Enabled		true	true
Set current	0	0
RMS current	1767	1767
MAX current	2491	2491
Run current	31/31	31/31
Hold current	31/31	31/31
CS actual		31/31	31/31
PWM scale	255	225
vsense		0=.325	0=.325
stealthChop	true	true
msteps		16	16
tstep		1048575	1048575
pwm
threshold		0	0
[mm/s]		-	-
OT prewarn	true	true
OT prewarn has
been triggered	true	true
off time		5	5
blank time	24	24
hysterisis
-end		2	2
-start		3	3
Stallguard thrs	0	0
DRVSTATUS	X	Y
stallguard
sg_result		0	0
fsactive
stst		X	X
olb
ola
s2gb
s2ga
otpw		X	X
ot			X
Driver registers:
	X = 0x84:1F:00:00
	Y = 0x86:1F:00:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

This happens everytime. No matter if I move X or Y axis, only one motor moves.

I tried changin CS pins several times, I almost used every free pin in my RAMPs.

What else I'm missing here? What could be the problem I'm having?

Thanks in advance!
Martin.-

KrX3D, 2018-02-16:
it happens for me too. sometimes i can start a print and after some minutes i get a Printer halted message... for now im back to my Pololu

teemuatlut, 2018-02-16:
It looks like your communication should be working alright.
Do you have EEPROM enabled? If so, send `M906 X600 Y600` followed by `M500`. You can then powercycle the printer.

RogueBolo, 2018-02-25:
I second teemuatlut's comment, when I ran into this a week ago resetting my eeprom got the drivers working.

KrX3D, 2018-03-03:
for me it is solved. on one of my tmc2130 i resoldered the diag pin and i also changed the marlin firmware from 1.1.8 to the bugfix firmware

martind-n, 2018-03-06:
Hi, finally I could solve my problem, for some REASON I don't know, mega 2560 didn't get updated after flashing it so the values never got refreshed. I changed the arduino board and now it's working flawlessly.
Big thanks @teemuatlut for your help.

closed 2018-03-06 by martind-n