MarlinFirmware/Marlin #9680: LCD_PROGRESS_BAR dependent on SDSUPPORT

Zer0t3ch, 2018-02-17:
### Bug Report
- Description: Unable to enable an LCD progress bar without enabling SDSUPPORT
- Expected behaviour: Compile with LCD_PROGRESS_BAR defined and SDSUPPORT undefined
- Actual behaviour: Fails to compile with LCD_PROGRESS_BAR defined and SDSUPPORT undefined

Personally, I have an ANET A8 with the stock mainboard, meaning I do have an SD port, but I wanted to leave it disabled to save memory, but I still want to be able to view a progress bar by sending M73 automatically mid-print via OctoPrint. I'm planning on upgrading to an ATMEGA2560 with RAMPS1.5 soon, which I don't believe has an SD port, but will keep using my display, hence I *really* won't have a reason to leave it in, soon.

benlye, 2018-02-17:
I created #9208 to address this a little while ago.

thinkyhead, 2018-02-17:
Will merge that shortly…

closed 2018-02-17 by thinkyhead