MarlinFirmware/Marlin #9683: When I compile with `FAST_PWM_FAN` I get an error

Iamturbo1978, 2018-02-17:
### Compile Error
When I compile with `FAST_PWM_FAN` I get an error with current BugFix 1.1.x
```
C:\Users\****\AppData\Local\Temp\arduino_build_166868\sketch/temperature.cpp:1064: undefined reference to `Temperature::setPwmFrequency(signed char, int)'

collect2.exe: error: ld returned 1 exit status
```

If I comment out `FAST_PWM_FAN` it compiles fine. For my setup, I have to use `FAST_PWM_FAN` for my setup. I have a low-pass filter on my fan that I set up just for the fast PWM frequency.

Bob-the-Kuhn, 2018-02-17:
Try replacing the temperature files with these.  It compiles but I can't test it so I'd like your feedback before it gets implemented

[temperature.zip](https://github.com/MarlinFirmware/Marlin/files/1734277/temperature.zip)

Iamturbo1978, 2018-02-18:
I will try it out today and let you know. 

Thank you for all the work.

Iamturbo1978, 2018-02-18:
I replaced temperature file with the one you provided, and it worked. I have full control of my part fan. I don't have my O-scope right now (I let a friend barrow it), so I cant see if it is providing the same frequency as before. But from the way my fan acts and sounds, I think it should be correct. Thank you

Bob-the-Kuhn, 2018-02-18:
Great!

The fix has been merged into bugfix-1.1.x

AlienTux, 2018-02-22:
I have a small issue with Fast PWM as well.

I can compile, but homing on the LCD breaks. If I try individually homing X, for example, the endstops don't seem to work. If I home using Repetier on my PC it works fine. Even checking M129 works and endstops report properlly. It's just the LCD homing that breaks.

thinkyhead, 2018-02-23:
@AlienTux — Thanks for visiting. Please open a new issue if the bug still exists in today's `bugfix-x.x.x` branches.

closed 2018-02-23 by thinkyhead