MarlinFirmware/Marlin #9684: recovery atmega1284p

manotroll, 2018-02-17:
After making some recordings I get this error now

avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

Bob-the-Kuhn, 2018-02-17:
Does it respond to the host program (Repetier Host, Pronterface, etc.)?

If not then the flash is probably corrupted.  One way to recover is use something like the [USB Tiny ISP](https://www.ebay.com/itm/USBtinyISP-USB-Tiny-Meag2560-USBTiny-AVR-ISP-Programmer-For-Arduino-Bootloader/162869073377?hash=item25ebc0d5e1:g:odcAAOSw6ShZXLav).

Roxy-3D, 2018-02-17:
This happens when Arduino tries to flash the part with a successfully built firmware image...   But it can't do it because some other program (like Pronterface) has control of the serial port.    You need to disconnect the program from the printer electronics...   Then rebuild the firmware and probably it will flash OK.

manotroll, 2018-02-18:
> Does

 it respond to the host program (Repetier Host, Pronterface, etc.)?

If not then the flash is probably corrupted. One way to recover is use something like the USB Tiny ISP.

I have an arduino nano mini and r3
can I do this with any of these or only with the one with that indicated board?

Bob-the-Kuhn, 2018-02-19:
I know you can use other Arduinos to burn a new bootloader but I haven't found one that specifically uses a nano.

I think this is a good starting point to learn about the process.  https://learn.sparkfun.com/tutorials/installing-an-arduino-bootloader

manotroll, 2018-02-24:
Okay, I'll try to bring him back today.

closed 2018-02-27 by thinkyhead

manotroll, 2018-03-01:
:)
http://www.instructables.com/id/HOW-TO-FIX-ANET-BRICKED-BOARD-USING-AN-ARDUINO-UNO/
with this tutorial I got it back

Bob-the-Kuhn, 2018-03-01:
FANTASTIC!!!