MarlinFirmware/Marlin #9685: 2.0 +1.1.x  BIG FONT EDIT IS DOWN .... doesn't work ... Snifff i'm crying

Studiodyne, 2018-02-17:
The EDIT BIG FONT  not working on all release version

I want to die ...

Roxy-3D, 2018-02-17:
@Studiodyne    Is it possible to find the commit that broke it?   If we know where it got broken...  Usually it is easy to fix.

Studiodyne, 2018-02-17:
@Roxy-3D , i have enough difficulties to print something with marlin , strong difficulties to debug something wrong , and now , you want i learn this website to understand how to find the tons of commits , and search something like ' BIG EDIT FONT breaker commits '

Lol
Too much for me

Studiodyne, 2018-02-17:
I TRUST YOU 😸

Studiodyne, 2018-02-17:
I will wait

Studiodyne, 2018-02-17:
My wish for UBL
- off mesh Z secure height 'you know all '
- off mesh z secure active only under current z 
- lcd babystepping active is UBL disabled '  seems to be solved '
- And a new wish ..... FADE HEIGHT in LCD , because  fade height change when the object to print change , and there is no possiblity to adjust it by lcd , and if we print by sdcard ,  to set a fade height need usb connection and will reboot the board 
Fade height have been forgotten , but it's a VERY IMPORTANT DATA
I use one millimeter fade height , because i d'ont understand why10mm
if you print a plate of 3 i understand that we want same thickness everywhere , but it's the only object that need a high fade height , all other object want to smooth the plan the more quick as possible

Studiodyne, 2018-02-17:
The code seems to be very simple , just some line in ultralcd , that increase size , but there is a ton of conditionnals about the CHARSET compatbility 
I have looked for this in commits , and i have found nothing , and I prefer , that the last author of something about LCD char size will quickly fix it

Roxy-3D, 2018-02-17:
I'm sure they will fix it...   If they see that there is a problem.

Roxy-3D, 2018-02-17:
@Studiodyne    What font are you using?   I saw this code:
```
// Only Western languages support big / small fonts
#if DISABLED(DISPLAY_CHARSET_ISO10646_1)
  #undef USE_BIG_EDIT_FONT
  #undef USE_SMALL_INFOFONT
#endif
```

Studiodyne, 2018-02-17:
My old release , one month ago , works fine , and have UTF8 fr 
It's why , i searchin anything else that the font type
and this code , is the same too , i have already check the basic
But , may be the way to define DISPLAY_CHARSET_ISO10646_1 can be the source 
I have looked at this too , but , i was fallen in a jungle of conditionnals
i have stop to think i'm the greatest engineer here , and i have continued to cry ! 😢

Roxy-3D, 2018-02-17:
.ZIP up the last version you have that works and post it here.   I'll do a visual diff on the Character Set stuff and the BIG_EDIT stuff.    It maybe with a working version to compare against that it is easy to find the problem.

Studiodyne, 2018-02-18:
You're a very cool girl ! Yes !
[Multo 3.1.zip](https://github.com/MarlinFirmware/Marlin/files/1734524/Multo.3.1.zip)

Roxy-3D, 2018-02-18:
There are a lot of changes for how character size is handled in ultralcd_impl_DOGM.h   My guess is those changes were not tested with the Big Edit fonts.   It is going to be some work to find the bad code.   You may be able to take the old (and working) ultralcd_impl_DOGM.h and replace the current file with it.   That is probably worth a try.   If it compiles, probably you will have the Big Edit Fonts back.   And that would give us confirmation of where the problem is.

thinkyhead, 2018-02-20:
Patched!

closed 2018-02-20 by thinkyhead

Studiodyne, 2018-02-20:
@Thinkyhead  From what planet do you come from  ? how it's possible to solve so quickly all the issues here ......... Respect !

thinkyhead, 2018-02-20:
@Studiodyne — I've got a pile of things to follow up on just clearing bugs for the release. I'll look at new features more closely after that.

[image]

commit 35800eec (2018-02-20): Fix big edit font width calculation

Fix #9685
commit 88d9de64 (2018-02-20): Fix big edit font width calculation  Scott Lahteine committed

Fix #9685
commit 9f0907d4 (2018-02-20): Fix big edit font width calculation

Fix #9685