MarlinFirmware/Marlin #9689: Crazy Z axis bug

ModMike, 2018-02-18:
### Bug Report

**Environment:**
I am using the Anet 1.0 firmware (not optiboot) and Marlin Instant configurator.  I disabled SD card support to get auto leveling.  I am using an SN04 sensor on cold bed.

**Issue:**
After successfully configuring Marlin for an Anet A2 with Version 1.5 board, I printed a 12 hour part without issue.  I then flashed the firmware to update the probe Z offset. 

Following the flash, any Z axis move command above 10, sends the carriage up until it jams at the stops.  Homing also cause the Z axis to raise to 10 after homing is complete.  The printer seems to understand it did this because it correctly reports position as X0 Y0 and Z10.  Sending a G1 Z0 brings it back to zero 

**Attempted fixes:**
I reverted to an earlier backup that had worked but still had same issue.  For testing purposes I removed auto level, simplified all options as much as possible, cleared eeprom (M502) and finally disabled EEProm and re-flashed.  Still no luck.

The configs I attached are the last ones I tried.  I hope you have an idea because I have been at this for 40 hours!

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/1734376/Archive.zip)

This is the first time I post here, please excuse me if I missed formatting or description conventions.

ModMike, 2018-02-18:
More information:

It turns out I was able to resolve the issue by setting the DEFAULT_AXIS_STEPS_PER_UNIT to 400.

What I don't understand is that when I use that rate, the motor is loud and gets hot.  I used the Reprap and Prusa calculators to calculate the correct steps and it gave me 1600 for Z.  

I was able to use 1600 for the Z axis before.  It seems to me it worked and was way smoother.  Is there another setting I need to balance it with?  What did I miss here?

workheart, 2018-02-18:
I Had a similiar problem long time ago on another axis, it somehow refers to speed and the problem was a bad conection where (that was my thinking) some intefferences caused the driver to revice a "step" signal from the motor wires. rerouting the cable solved the prob. (its totaly gueswork tho.. ) but in my case it only happend on one speciffic speed ..

Lenbok, 2018-02-19:
Do you have home offsets set? 

I'm seeing weird z axis movement behaviour on my mendel90 too and I think it's related to the fact that I have a z home offset in play.

If I set the home offset to 0 the z axis moves as it should. 

If I set it to my normal value (approx -18mm, since the compiled firmware just has an approximate Z_MAX_POS and it depends on the length of the extruder I'm currently using what the actual offset needs to be), and then home Z, it goes to the max Z as it should, except the LCD reports a Z_MAX_POS rather than (Z_MAX_POS + Z home offset). If I tell octoprint to drop Z by 100mm, the printer Z physically goes down by approx 80mm. If I then tell octoprint to drop Z by a further 10mm, the Z actually *raises*.  

I suspect there's a bad interaction between a Z offset and homing to max Z as the mendel90 does -- when you home to max Z, setting a Z offset should result in a change in the total height, rather than a simple shift in coordinate.

I can't really pinpoint when the behaviour changed, as I've done upgrades through several marlin releases recently and I haven't been paying close attention to the LCD or doing manual Z movements.

Lenbok, 2018-02-19:
Yeah, I've played with this some more and this behaviour is just weird. Paging @thinkyhead please?

I recompiled (release 1.1.8), with my Z_MAX_POS set to 180.
M206 is initially set to Z0. Homing and manually moving the z axis works as it should. 
Now I manually jogged to ~8mm above my bed so the nozzle touched a piece of 8mm rod (actually 7.96mm). The LCD said ~3.1mm, so I issued M206 Z4.86. However M114 says Z is now at 7.96 (which it is) but the LCD hasn't updated.  Why doesn't the LCD display the correct position?

Now I homed all axes, which homes Z to max as it should, however, after homing the LCD Z says it is at 180 (it should say 184.86, which we have now determined is the real distance from the bed to the Z MAX endstop). The printer serial output does think it is at 184.86 though:
````
Send: G28
Recv: X:100.00 Y:100.00 Z:184.86 E:0.00 Count X:8000 Y:8000 Z:576000
````

Now I jog down 100mm in octoprint and check the coordinates:
````
Send: G91
Send: G1 Z-100 F200
Send: G90
Send: M114
Recv: X:100.00 Y:100.00 Z:80.00 E:0.00 Count X:8000 Y:8000 Z:240448
````
(LCD reports Z is 75.14). What?  At this point Z should be 84.86 on both M114 and LCD. Something is screwy.

Now I jog Z *up* by 1mm - however the machine physically moves *down*!:
````
Send: G91
Send: G1 Z1 F200
Send: G90
Send: M114
Recv: X:100.00 Y:100.00 Z:76.14 E:0.00 Count X:8000 Y:8000 Z:228096
````
(LCD reports Z is 71.28).

That's just rooted!

ModMike, 2018-02-19:
Lenbok.  I agree.  I did verify what I stated and it is a fact that I should be using 1600 for the DEFAULT_AXIS_STEPS_PER_UNIT for the Z-Axis.  I confirmed this in the Prusa calculator and by seeing that the Tevo Tarantula (of which the Anet A2 is a clone) also uses 1600.

I can't understand why it worked once and then went crazy.  I don't think its a home position setting at zero because each 10mm step I move the z axis results in a move that is farther and farther off, suggesting it's the wrong stepping.

I suspect that my version may have incorrect settings that are not revealed in the Marlin Instant Configurator from akaJes, specifically the Nema 17 settings that allow you to define the motors, pullys and what not that allow the compiler to calculate the proper default axis steps.

I will advise if I am on to something.

ModMike, 2018-02-19:
This thread may be helpfull:

https://github.com/MarlinFirmware/Marlin/issues/7149

ModMike, 2018-02-19:
I just realized I forgot to give my calculations:

[image]

ModMike, 2018-02-19:
Issue MAY be resolved by setting MINIMUM_STEPPER_PULSE to 1 In configuration_adv.h.

Set to 1 for A4998
Set to 2 for A8998

Will test some more and advise.

ModMike, 2018-02-19:
I tried setting the MINIMUM_STEPPER_PULSE to 1 and also raised the vRef to no avail.  I've been on this for 3 days, anyway someone can help?

Lenbok, 2018-02-20:
(Since my issue sounds separate to yours, I've created a new issue)

ModMike, 2018-02-20:
Please close this issue as it is resolved.  Anet uses a 4 start lead screw so 400 is the correct value.

thinkyhead, 2018-02-25:
W00t! That's good. 1600 micro-steps on an AVR board leads to instability and reboots.

closed 2018-02-25 by thinkyhead