MarlinFirmware/Marlin #9702: BLtouch/3DTouch not working

sshmooss, 2018-02-18:
Hey all.

For some reason my Sensor goes instantly into a alarm mode after it probes or even startrs. If i reset it with my LCD it stays normal , but as soon as i deploy it or selftest it goes into alarm and does not work. Homing the Z axis does not work either with the sensor.

I added my configuration file.

Im using a CR-10S and my Touch is black with a blue light inside of it. With the 3 pin cable (Black/Red/blue) and the 2 pin (yellow/black)

Thank you for reading and hope we can find an answer for me :)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1734796/Configuration.zip)

boubouexpress, 2018-02-18:
salut
[image]

va voire
https://github.com/MarlinFirmware/Marlin/issues/9386#issuecomment-365618552

sshmooss, 2018-02-18:
@boubouexpress yes i have wired my cables like following in the picture with only exception that my sensor has different colors on the cables . If i compare the picture and my Sensor cables its like following => Orange =RED , Brown = BLACK , Yellow = BLUE , White = YELLOW, Black = BLACK. And as stated before 3pin is red/black/blue and 2pin is yellow/black

boubouexpress, 2018-02-18:
option BLtouch
#define BLTOUCH 
#define X_PROBE_OFFSET_FROM_EXTRUDER -42 
#define Y_PROBE_OFFSET_FROM_EXTRUDER -11 
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.1

#define MULTIPLE_PROBING 2

#define GRID_MAX_POINTS_X 7

#define LEFT_PROBE_BED_POSITION 40 
#define RIGHT_PROBE_BED_POSITION 455 
#define FRONT_PROBE_BED_POSITION 30 
#define BACK_PROBE_BED_POSITION 470

#define AUTO_BED_LEVELING_BILINEAR

// # define LCD_BED_LEVELING

#define LEVEL_BED_CORNERS

#define Z_SAFE_HOMING

sshmooss, 2018-02-18:
@boubouexpress ill try those settings and ill see what it says after that! :)

boubouexpress, 2018-02-18:
for this model

www.thingiverse.com/thing:2493610/#files

#define X_PROBE_OFFSET_FROM_EXTRUDER -42 
#define Y_PROBE_OFFSET_FROM_EXTRUDER -11 
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.1

boubouexpress, 2018-02-18:
if you do not have the last marlin

#define SERVO0_PIN 11

sshmooss, 2018-02-18:
https://streamable.com/8jlj9

Bob-the-Kuhn, 2018-02-18:
I hadn't seen the newer BLTouch units.  Black with a bright blue LED & a connector.  I like that.

My understanding is that the BLTouch only goes into alarm when commanded to deploy/stow but the move fails.

Here's a sequence that should help narrow down what is happening.  It was written for the classic so the LED color info is different.
1. Connect BLTouch, power up controller and upload the firmware
2. Blue LED will be ON if a servo signal is present.  Blue LED is faint.
3. M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used. Also check that Z_MIN_ENDSTOP_INVERTING and Z_MIN_PROBE_ENDSTOP_INVERTING are both set to "false".
4. M280 P0 S10 ; should deploy the probe and the orange LED will be OFF.
5. M119 ; should return OPEN
6. M280 P0 S90 ; should stow the probe and the orange LED will be ON
7. M280 P0 S60 ; puts it into the M119 test mode.
8. The probe should remain stowed and the blue LED should be OFF. If it's ON then the unit needs to be adjusted.
8. M119 ; should return TRIGGERED
10. M280 P0 S160 ; returns it to normal operation
11. M119 ; should return OPEN.

sshmooss, 2018-02-19:
@bob-the-kuhn ill try that send you the report after! 

Here is a video when i had 2pin wires like in the picture but z axis did not even go down the with auto home. 

https://streamable.com/iveg8

So as i wrote earlier i have the wirings now. What do you think about them?

boubouexpress, 2018-02-19:
it's a 3D touch is not a bltouch

sshmooss, 2018-02-19:
Or did i have the wiring right first time but the z axis juat goes up because of the alarm?

sshmooss, 2018-02-19:
@boubouexpress . Does that actully matter? Dont they both do the same thing?

boubouexpress, 2018-02-19:
the wiring

boubouexpress, 2018-02-19:
Zmin
white = yellow
black = black

brown = black (ground)
red= red  (5v+)
orange = blue ( D11)

boubouexpress, 2018-02-19:
in pins_ramps.h 

there is

 #define SERVO0_PIN       11

boubouexpress, 2018-02-19:
otherwise it takes the meter in configuration.h

#define BLTOUCH
#if ENABLED(BLTOUCH)
#define SERVO0_PIN 11

sshmooss, 2018-02-19:
So qhat do you want me to change in my marlin?

sshmooss, 2018-02-19:
@bob-the-kuhn i did all the steps .

Steps:

2. Blue led is ON
3.Open
4. Only blue blinking light
5.triggered
6. Does not stow , blue blinking
7. Deploys and blue blinking
9. Triggered
10. It goes up and blue light is on
11. Open

Bob-the-Kuhn, 2018-02-19:
Before changing anything please do the following:
1) In Configuration_adv.h enable **PINS_DEBUGGING**
2) From the host issue a **M43 S** command & follow the on screen prompts.
3) Post the results of the M43 command
4) Redo the 11 steps & report the results of every step.  I'm not understanding the results you just posted.

The last time I played with a 3D Touch it behaved the same as a BLTouch so I'm not concerned that you have a 3D Touch.

sshmooss, 2018-02-20:
`Send: M43 SRecv: Servo probe testRecv: .  using index:  0Recv: .  deploy angle: 10Recv: .  stow angle:   90Recv: . probe uses Z_MIN pin: 18Recv: . uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)Recv: . Z_MIN_ENDSTOP_INVERTING: falseRecv: . deploy & stow 4 timesRecv: WARNING - INVERTING setting probably backwardsRecv: please trigger probeRecv: trigger not detectedRecv: ok`

sshmooss, 2018-02-20:
`Send: M119Recv: Reporting endstop statusRecv: x_min: openRecv: y_min: openRecv: z_min: openRecv: filament: TRIGGEREDRecv: ok[...]Send: M280 P0 S10Recv: ok[...]Send: M119Recv: Reporting endstop statusRecv: x_min: openRecv: y_min: openRecv: z_min: TRIGGEREDRecv: filament: TRIGGEREDRecv: ok[...]Send: M280 P0 S90Recv: ok[...]Send: M280 P0 S60Recv: ok[...]Send: M119Recv: Reporting endstop statusRecv: x_min: openRecv: y_min: openRecv: z_min: TRIGGEREDRecv: filament: TRIGGEREDRecv: ok[...]Send: M280 P0 S160Recv: ok[...]Send: M119Recv: Reporting endstop statusRecv: x_min: openRecv: y_min: openRecv: z_min: openRecv: filament: TRIGGEREDRecv: ok`

sshmooss, 2018-02-20:
Ok so let me try to explain again the steps and how to 3Dtouch behaves :)

- When power on it is stowed and BLUE light does not blink.
- M119 says OPEN

-M280 P0 S10 , the pin drops down and immideatly BLUE light starts blinking
-M119 says TRIGGERED

-M280 P0 S90 , DOES NOT WORK. The pin does not go up
-M119 says TRIGGERED
-Still BLUE light blinking

-M280 P0 S160 , returns the pin up and BLUE light does not blink anymore
-M119 says OPEN

Hope this clarifies something to you @Bob-the-Kuhn

Bob-the-Kuhn, 2018-02-20:
I don't understand.

When you did the **M43 S** test, did it deploy & stow four times?

Maybe the wiring is wrong.  Try disconnecting the white & black wires and then see if the **M280 P0 S10** and **M280 P0 S90** commands work as expected.  If they do then REVERSE the two wires & re-connect.

sshmooss, 2018-02-20:
It did not deploy and stow four times. It only deployed once and then started blinking the blue light. Ill try it and return to you with the answer :)

sshmooss, 2018-02-20:
https://streamable.com/k9o8v

Here is a video. 
FIRST. i did "M43 S" and as you can see it does not go up and down. only down and then starts blinking.
THEN i did "M280 P0 S10", and as you can see it drops down and starts blinking
THEN i did "M280 P0 S90", and it does not go up . it only stays down
THEN i did "M119" to show it says TRIGGERED
AND for last i did "M280 P0 S160" to show how it goes up and resets
AND as you can see i unconnected the Z min cables :)

@Bob-the-Kuhn

Bob-the-Kuhn, 2018-02-20:
I'm starting to wonder exactly what you have.  The Geeetech site still shows the translucent white case and, on EBay, the black one is just "3D Touch TE871", not "Geeetech  3D Touch".

I'm going to email Geetech & invite them to comment on this thread.

technical@geeetech.com

sshmooss, 2018-02-21:
@bob-the-kuhn thank you :)

sshmooss, 2018-02-21:
@bob-the-kuhn wich one is the white working one that everyone is using if i may ask? :)

Roxy-3D, 2018-02-21:
> I'm going to email Geetech & invite them to comment on this thread.

This is going to be fun.   I  doubt anybody at Geetech cares enough to come here and help a user.   But if they do, I have a few questions to ask them...

Bob-the-Kuhn, 2018-02-21:
The ones I know are 100% compatible are the Antclabs BLTouch and the Geeetech 3D Touch.

sshmooss, 2018-02-21:
Well lets hope for the best that we can find an answer for me and the rest of the people out there with a black 3dtouch :)

GMagician, 2018-02-23:
OT @Roxy-3D 
>  I have a few questions to ask them...
:-D  It sounds like a threat

Bob-the-Kuhn, 2018-02-26:
There was another person with what sounds like the same issue.  He solved his problem by adjusting the set screw in the top of the 3D touch.

Please try that.

Bob-the-Kuhn, 2018-02-27:
Just got an email from Geeetech.  They said that the black one is not theirs.  

Were you able to get it to work?

sshmooss, 2018-02-28:
@bob-the-kuhn no i was not. Tried adjusting and everything else that i could find on the web :D i also ordered a geeetech one to see if its the probe thats faulty. BUT all suggestions are welcome :)

Bob-the-Kuhn, 2018-02-28:
Adjusting the set screw was the last suggestion I had.

Geeetech makes a good product.  I don't remember any hardware problems with them.

baslbro3, 2018-03-02:
Have you test setting Z_MIN_ENDSTOP_INVERTING and Z_MIN_PROBE_ENDSTOP_INVERTING both to TRUE just for testing? Maybe the non geetech 3dtouch is inverted.

sshmooss, 2018-03-06:
@baslbro3 same thing even if you change the true and false values back and front. I already ordered the real one. So we will see when that one arrives :)

baslbro3, 2018-03-06:
@sshmooss yeah, I was actually experiencing the same thing and it was actually a bad probe. Once I got my new probe, I realized a big difference. Even when not powered, the pin should be able to stay up in the sensor via magnetism. My original sensor was just dropped. The servo would bring it up and down but the magnet would not hold the pin up and would keep the sensor in alarm mode. I was able to fiddle with the top screw and eventually get the magnet to hold the pin up but any movement in my xaxis would make the pin drop. The new sensor works perfect.

Split7fire, 2018-04-01:
Hello everyone! I'm too having this black semi-transparent 3d Touch sensor, and I can't figure out how to make it work.
I' ve tried a lot of configuration options.
@Bob-the-Kuhn Here is output from M43 S:
`BLTouch compatible probe detected - pulse width (+/- 4mS): 12`
But I can't reproduce this message, because somehow sensor is stuck (not physically) and I can't control via commands (stow and trigger). Also my config.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1866077/Configuration.txt)

thinkyhead, 2018-04-02:
Does the device have a small screw that can be adjusted?

Split7fire, 2018-04-02:
@thinkyhead Yep, I tried adjusting that. No luck.

Bob-the-Kuhn, 2018-04-03:
Did the probe extend & retract during the M43 S test?

I don't think it matters but the following lines can be commented out.  Marlin automatically sets these when BLTouch is enabled.
```
#define Z_ENDSTOP_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {10,90}  // Z Servo Deploy and Stow angles
```

Split7fire, 2018-04-03:
@Bob-the-Kuhn 

> Did the probe extend & retract during the M43 S test?

Nope.
What this sensor does:

- When powering up, it stow and trigger 2 times, then it can flash or can dim (depends on invert setting in Marlin, I think)
- No commands from terminal can stow or reset the sensor
- If I set invert Z axis, and manually pull probe from the sensor, M119 prints "Z-axis - Trigger".

As for now, I dont know where to search for workarounds.

P.S. Being curious what is inside this sensor, I openned it and found soldered out element 01x02 or something like that. It was black color and may be that was resistor, someone here said before.

Bob-the-Kuhn, 2018-04-05:
Probe extends & retracts after a power up so we know that works.

M43 S reports a reasonable pulse length so we know the signal is wired up properly.

What doesn't work is the servo command interface.

If you can't find any wiring problems with the servo signal then it's a bad probe.

liviudd, 2018-09-13:
Hello, 

It's a bit late, but i was struggling with an 3D Touch for the last week with the red led blinking after extend/retract and only reset from menu worked. After reset, any retract giving me the same error. After i tried everything on this issue and nothing worked i remembered moving the top screw just after mounting it. So my pin was about 3mm out in retracted position.

It seems stupid but it has an sweet spot about 9mm out. After this everything is working and the sensor does not remain blocked after extend and retract. 

So maybe this will help others not to waste time like me.

bob808, 2018-10-19:
@liviudd is it the black variant?

liviudd, 2018-10-19:
@bob808  black sensor ? No, is the transparent one,

BKroef, 2019-11-24:
Maybe play with these settings:
//#define ENDSTOPPULLUP_ZMIN
//#define ENDSTOPPULLUP_ZMIN_PROBE

I had to change my settings to NOT use an ENDSTOPPULLUP resistor for the BLTouch pin (ZMIN and ZMIN_PROBE).
Only enabled it for the XMIN, because YMIN is a inductive sensor and therefor also doesn't use an ENDSTOPPULLUP resistor.
These are my settings in Configuratin.h:
// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  #define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE

Koetjeboeh, 2019-12-21:
Working on it here, but this doens't do the trick at my Ender-5