MarlinFirmware/Marlin #9706: Creating a new interrupt on Marlin

MarlonBrandoII, 2018-02-18:
Suspect this topic has come up in the past, but did not find a satisfactory answer in the prior discussions, so here it goes...

Looking to count some very high rate pulses coming in from a circuit (such as a set of encoders). Wish to do this on the same board that is running Marlin.

Specific frequencies and other details less important. The main question: Could one define an interrupt on an unused, compatible pin, pause all other interrupts, count pulses on that interrupt for a duration (say, 100 ms) and then restart all other interrupts on Marlin? Will probably define a new Gcode command on "Marlin_main.cpp". 

Don't really need temperature controls running during this phase of the operation, so could turn off PID controls if necessary.

The basic functionality would be similar to that succinctly shown here: https://gist.github.com/omsai/2363047. Just need to accomplish it within Marlin, if possible.

Thanks in advance.

Sebastianv650, 2018-02-18:
> Could one define an interrupt on an unused, compatible pin, pause all other interrupts, count pulses on that interrupt for a duration (say, 100 ms) and then restart all other interrupts on Marlin?

No way. If you block Marlin for 0.1s, there will also be no stepper movement in this time. 
But you can do technicaly the same as in your linked example: Create an ISR which listens to a pin and increase the counter on every event. Then do something with that value inside the Marlin main loop for example. You just have to find a suitable free pin. Usualy all ports can trigger a level change interrupt, but only some special pins can be configured to trigger only on a rising or falling edge. While the last ones are easy they belong to a specific pin, the generic pin (port) change interrupts are a bit more tricky: Any pin  which changes it's level from 0 to 1 or 1 to 0 belonging to this port will trigger the ISR. So you have to check inside the ISR if the right pin is responsible for triggering, only if so increase the counter.

Bob-the-Kuhn, 2018-02-18:
Another possibility is to clock one of the counters from the pin.  This will catch every transition but there's only a few pins that will do this and most of the counters are already used.

What CPU/controller are you using?

MarlonBrandoII, 2018-02-19:
Awesome comments, @Sebastianv650 and @Bob-the-Kuhn. Much appreciated.

Implemented it as suggested on the Marlin_main.cpp file (using attachInterrupt() function) on the interrupt pins (tried 2, 3, 18, and 19 on the Mega 2560). Used delay(1000) to simply return the frequency of the pulses coming in (total count/s).

Works very well. Counts quite accurately, up until about 125-130 kHz square waves input on the setup pin. Higher frequencies than that start returning garbage. Actually, 200 kHz and higher simply resets the controller.

Is it conflicting with another interrupt? No steppers are running at the time, and does not matter whether PIDTEMP is enabled or not. All the pins I tried (listed above) gave me the same result.

Sebastianv650, 2018-02-19:
Why you use a delay at all?
Maybe your counter overruns at a higher frequency than 130kHz during the 1 second? If you do just a "++" inside the ISR, I think you should be able to go well above 200kHz, assuming no other ISRs are enabled.

If you have all other ISRs disabled no conflict is possible. Try to eleminate the delay first, I don't think that's a good way of coding..

AnHardt, 2018-02-19:
16MHz/0.2MHz = 80 ticks for jumping into to ISR,, saving the registers, doing something useful, pulling the registers back and returning is a bit optimistic.

delay() needs the timer interrupt to work. If something blocks the timer interrupt for longer than a millisec a tick gets lost - delay will be too long.

@MarlonBrandoII 
Did you disable the watchdog? Delaying a second and disabling the temperature IRQ, both makes the watchdog very unhappy.

MarlonBrandoII, 2018-02-19:
delay() function, as I understand it, does not impact other interrupts. That's why no watchdogs are unhappy about it, I think. Marlin is unhappy for higher frequency pulses coming in, though.

What's a better option to count pulses for a given period without using delay()? Another interrupt that fires every chosen duration?

Sebastianv650, 2018-02-19:
@AnHardt is right, 200kHz might be too high.

> What's a better option to count pulses for a given period without using delay()? Another interrupt that fires every chosen duration?

I built an oboard computer for my motorcycle some years ago for calculating milage, fuel consumtion and so on. For this, an Arduino counted the pulses of the speed sensor from the front wheel and the pulse length from the fuel injectors.
The principle is the same I would use in your case: I did a main loop which was run every second. If this loop finished in < 1s (which it always did of course), I told the Arduino to do nothing. Basicaly looping NOPs, but I think delay is doing more than that.
In the mean time, while the main loop is running and while it idles, the ISRs counted the amount of speed pulses and the duration of the injection opening.

What I want to say is: Let the ISR run in background. Do the main loop meanwhile or other tasks. Check the results when a suitable amount of time has passed, but use the time ;)

AnHardt, 2018-02-19:
> delay() function, as I understand it, does not impact other interrupts.

Causality error! You are right. delay() does not affect other interrupts. But if you shut off the ticker interrupt, delay() does not tick. So your wanted second will not be a second but longer.

The watchdog counter is pretty independent from all the other processor interna. So if you are looking for an unexpected reset, ...

MarlonBrandoII, 2018-02-19:
Thanks folks. So in the end, what is the best way to run the pulse incrementer ISR every second within Marlin? Is there a global ticker that one could piggy-back on? Or does one need to create said global ticker to time 1 s intervals?

Outside Marlin, within the Arduino environment, I have used the Timer1 library to create such tickers. Not sure what to do within Marlin, though.

AnHardt, 2018-02-20:
Don't switch off **all** interrupts. Let the ticker on.
Don't count pulses/time. Measure time for a number of pulses.
Don't use the normal Arduino interrupt header. Write your own, saving only the used registers (assembler).

What do you want to count? Does the exact number of pulses matter? If it's a position encoder stripe it matters. If fan rotations, likely the magnitude is exact enough.

Can the pulse rate be lowered in hardware (counter-TTL -  just counting overflows - 74xxx)?

MarlonBrandoII, 2018-02-20:
A couple of questions:

1) @Sebastianv650 suggested that maybe the counter overruns more than 125k-130k pulses arrive during the counting duration. Is there a way to find out if this is indeed happening?

2) Marlin specific: Assuming we keep well below the 130k pulses per count duration, how should that count duration be determined within Marlin? Looking for a specific implementation suggestion here. As already mentioned, Timer1.h library offers a convenient and accurate way to implement such a "ticker" interrupt outside of Marlin, but I am not sure if I can use it inside Marlin (that counter is probably being used for other Marlin-specific tasks already).

I looked into the Stopwatch class in Marlin. It uses millis() to count time. Not as accurate as the Timer1.h library, which ticks in microseconds. Of course, delay() function is in milliseconds. At any rate, could the Stopwatch class be used as a ticker?

Are there any other "tickers" that can be used in Marlin to accurate gauge pulse counting intervals?

Sebastianv650, 2018-02-20:
> @Sebastianv650 suggested that maybe the counter overruns when more than 125k-130k pulses arrive during the counting duration. Is there a way to find out if this is indeed happening?

Just by worst-case calculation. Frequency * max time between counter reset = needed value size. 130kHz * 1s = 130000 counts. So bytes and integers are too small, a long is sufficent.

For the duration, you have to find a timer that is not used by Marlin or a used one that runs at a frequency which fits your needs. The only way to find a propper one is to read the Marlin code and make notes which timer used used at which settings for what. Then just get the timer value after last counter reset and a timer value before you take the frequency counter value and calculate the difference.

wabbitguy, 2018-02-21:
I'm also interested in this topic. My reasoning is that while there are thermal routines in Marlin to prevent over/under thermal issues, there's no simple monitoring of the hot end fan to see if it's even running or not.

This might or might not be an issue for other printers (I only have delta's) but for a delta during a print if the hot end fan goes out of warranty, the hot end  tends to melt or severely distort the effector and entertainment value goes way up.

The fans I've seen are all two wire, but my thinking is for safety they should be three wire with simple monitoring to darn sure they are running. And if not, shut'er down and get the head out of there. And yeah, I've had two fans fail in 3K hours and the hot end dancing like a lure on the end of a fly rod...

MarlonBrandoII, 2018-02-22:
Sorry, folks, but don't think we actually received a precise solution to the question. Here is the basic question, repeated here for convenience:

* What can one create a timer ticker inside Marlin (without using "delay()")? Is there a counter that is currently unused that can be used as a timer ticker? If so, please provide a simple, specific example configuration.

Thanks.

thinkyhead, 2018-02-23:
Timer 5 is probably free.

boelle, 2019-02-20:
@MarlonBrandoII is this a question or feature request?

closed 2019-07-06 by shitcreek