MarlinFirmware/Marlin #9707: Y axis on TMC2130 in wrong microstepping / driver registers WRONG

mamama1, 2018-02-18:
I have updated all libraries to the most recent version and now the Y axis moves in the wrong microstepping mode. 244 real millimeters are shown as 122 millimeters on the display.

all drivers are TMC2130 in SPI mode.

everything has worked until I uploaded a new-compiled 1.1.8 (had 1.1.8 before that already) after I updated all libraries, so the bug must have been brought in by one of the updates.

as of now, I have:
- arduino ide 1.8.5
- SD lib 1.2.2
- Stepper 1.1.3
- TMC2130Stepper 2.1.4
- TMC2208Stepper 0.0.3
- U8g2 2.20.13
- U8glib 1.5.2
- Ucglib 1.5.2

not sure if that are all libs used by marlin and which of the u8* libs is used (u8glib i guess).

the driver registers are reported by M122 as follows, AFTER the newly compiled firmware with all the library updates:

Recv: Driver registers:
Recv: 	X = 0xA0:08:00:00
Recv: 	Y = 0x81:08:01:EC
Recv: 	Z = 0x80:0C:00:00
Recv: 	Z2 = 0x80:0C:00:00
Recv: 	E0 = 0x80:08:00:00

BEFORE that, when everything worked, (unfortunately I don't know which lib versions I had but I guess some of them were quite old) the driver registers were reported like this:

Recv: Driver registers:
Recv: 	X = 0x80:08:00:00
Recv: 	Y = 0x80:08:00:00
Recv: 	Z = 0x80:0C:00:00
Recv: 	Z2 = 0x80:0C:00:00
Recv: 	E0 = 0x80:08:00:00

I'll now try to downgrade some libs (starting with the TMC2130) to see if I can narrow the issue down.

mamama1, 2018-02-18:
ok I have narrowed it down to the TMC2130 lib, beginning with version 2.1.1.
prior to version 2.1.1 (2.0.2 and downwards) works.
very old versions don't work, obviously, as there has been too much change to the public interface of the library.

teemuatlut, 2018-02-18:
Try the latest bugfix-1.1.x branch.

mamama1, 2018-02-20:
seems to work with the latest bugfix branch. when will there be a new stable release?

thinkyhead, 2018-02-25:
Soon!

closed 2018-02-25 by thinkyhead