MarlinFirmware/Marlin #9709: Stepper motor is VERY HOT with TMC2130

Gaben5224, 2018-02-18:
Hey Guys!

I need a litle help! Today I replaced the old motors and the old stepper drivers too.
Now I use 2 TMC 2130 (X,Y axis) and 3 TMC2208 stepstcks (Z, Z2, E axis), and the machine is super silent, but SUPER HOT.
The X, Y, and Z motors are around 50°C and the extruder motor is around 80°C!!! 
Could anyone give me some advice?

Thanks for the help!!!

Here is specification of the motor: 
https://www.omc-stepperonline.com/hybrid-stepper-motor/nema-17-bipolar-09deg-36ncm-51ozin-09a-54v-42x42x40mm-4-wires-17hm15-0904s.html?search=17HM15-0904S

And here is the TMC debuging datas:

Recv: 		X	Y	Z	Z2	E0
Recv: Enabled		false	false	false	false	false
Recv: Set current	550	550	550	550	550
Recv: RMS current	520	520	939	520	939
Recv: MAX current	733	733	1324	733	1324
Recv: Run current	16/31	16/31	16/31	16/31	16/31
Recv: Hold current	8/31	8/31	8/31	8/31	8/31
Recv: CS actual		8/31	8/31	0/31	0/31	0/31
Recv: PWM scale	0	0	0	0	0
Recv: vsense		1=.18	1=.18	0=.325	1=.18	0=.325
Recv: stealthChop	false	false	false	false	false
Recv: msteps		16	16	256	16	256
Recv: tstep		1048575	1048575	0	0	0
Recv: pwmRecv: threshold		0	0	0	0	0
Recv: [mm/s]		-	-	-	-	-
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		5	5	0	5	0
Recv: blank time	24	24	16	24	16
Recv: hysterisisRecv: -end		2	2	-3	2	-3
Recv: -start		3	3	1	3	1
Recv: Stallguard thrs	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E0
Recv: stallguardRecv: sg_result		0	0
Recv: fsactiveRecv: stst		X	X
Recv: olbRecv: ola
Recv: s2gb
Recv: s2gaRecv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:	X = 0x80:08:00:00Recv: 	Y = 0x80:08:00:00Recv: 	Z = 0x00:00:00:00Recv: 	Z2 = 0x00:00:00:00Recv: 	E0 = 0x00:00:00:00

teemuatlut, 2018-02-18:
Unfortunately it's currently hard to interpret the TMC2208 output when used with SW serial as there is a limitation from SoftwareSerial implementation that we haven't taken into account just yet.

Gaben5224, 2018-02-19:
Dear teemuatlut! Thanks for your help, but I think I found the solution. 

I left the R_SENSE value on 0.11, despite I should use a value around 0.24 because my motor current is 0.9A. This is just a theory yet, I have to test it, but I think it can work.

I will share the result after the test.

teemuatlut, 2018-02-19:
The R_SENSE value should match the physical hardware component in your driver board or your motherboard.

ikarisan, 2018-02-19:
@Gaben5224 Since I am using TMC2130 drivers my motors are getting hot, too. I measured the temperature on X and Y and it is always around 54°C. So I bought some of these cork dampers to isolate the hot motor against my PLA printed mounts.

Gaben5224, 2018-02-19:
So, @teemuatlut you had right of course. I checked the documentations and the resistors too. Because the motors are 0.9A I set up the current to 636mA. Now it works but I had to decrease the jerk from 20 to 10 because I had some layer shifting. Now the motors are around 50C except the extuder what is around 70C. This is quite high I think. The question is that this is normal or not!? What should I do...I don't know. May I should put the old A4498 driver back to the extruder or is there any option what I should try to?

Thanks!

workheart, 2018-02-19:
you can autotune current via M906 S1
but```
 
Recv: X	Y	Z	Z2	E0
Recv: Enabled	false	false	false	false	false
```
says ony X has TMC2130 anabled or am i Wrong? 
im running 1.2A montors on 650mA and they are cold..  can you paste conf_adv?

Gaben5224, 2018-02-20:
@workheart thanks for your help!

There is the part of the config file:
`// @section TMC2130, TMC2208

/**
 * Enable this for SilentStepStick Trinamic TMC2130 SPI-configurable stepper drivers.
 *
 * You'll also need the TMC2130Stepper Arduino library
 * (https://github.com/teemuatlut/TMC2130Stepper).
 *
 * To use TMC2130 stepper drivers in SPI mode connect your SPI pins to
 * the hardware SPI interface on your board and define the required CS pins
 * in your `pins_MYBOARD.h` file. (e.g., RAMPS 1.4 uses AUX3 pins `X_CS_PIN 53`, `Y_CS_PIN 49`, etc.).
 * You may also use software SPI if you wish to use general purpose IO pins.
 */
#define HAVE_TMC2130

/**
 * Enable this for SilentStepStick Trinamic TMC2208 UART-configurable stepper drivers.
 * Connect #_SERIAL_TX_PIN to the driver side PDN_UART pin with a 1K resistor.
 * To use the reading capabilities, also connect #_SERIAL_RX_PIN
 * to PDN_UART without a resistor.
 * The drivers can also be used with hardware serial.
 *
 * You'll also need the TMC2208Stepper Arduino library
 * (https://github.com/teemuatlut/TMC2208Stepper).
 */
#define HAVE_TMC2208

#if ENABLED(HAVE_TMC2130) || ENABLED(HAVE_TMC2208)

  // CHOOSE YOUR MOTORS HERE, THIS IS MANDATORY
  #define X_IS_TMC2130
  //#define X2_IS_TMC2130
  #define Y_IS_TMC2130
  //#define Y2_IS_TMC2130
  //#define Z_IS_TMC2130
  //#define Z2_IS_TMC2130
  //#define E0_IS_TMC2130
  //#define E1_IS_TMC2130
  //#define E2_IS_TMC2130
  //#define E3_IS_TMC2130
  //#define E4_IS_TMC2130

  //#define X_IS_TMC2208
  //#define X2_IS_TMC2208
  //#define Y_IS_TMC2208
  //#define Y2_IS_TMC2208
  #define Z_IS_TMC2208
  #define Z2_IS_TMC2208
  #define E0_IS_TMC2208
  //#define E1_IS_TMC2208
  //#define E2_IS_TMC2208
  //#define E3_IS_TMC2208
  //#define E4_IS_TMC2208

  /**
   * Stepper driver settings
   */

  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #define X_CURRENT          638  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256

  #define Y_CURRENT          638
  #define Y_MICROSTEPS        16
[... 115 lines cut ...]
   *
   * Use M915 Snn to specify the current.
   * Use M925 Znn to add extra Z height to Z_MAX_POS.
   */
  //#define TMC_Z_CALIBRATION
  #if ENABLED(TMC_Z_CALIBRATION)
    #define CALIBRATION_CURRENT 250
    #define CALIBRATION_EXTRA_HEIGHT 10
  #endif

  /**
   * You can set your own advanced settings by filling in predefined functions.
   * A list of available functions can be found on the library github page
   * https://github.com/teemuatlut/TMC2130Stepper
   * https://github.com/teemuatlut/TMC2208Stepper
   *
   * Example:
   * #define TMC_ADV() { \
   *   stepperX.diag0_temp_prewarn(1); \
   *   stepperY.interpolate(0); \
   * }
   */
  #define  TMC_ADV() {}

#endif // TMC2130 || `TMC220/`

workheart, 2018-02-20:
The Config looks right. I would say ( you can add sonsorless Homing if its Cartesian but thats another theme. If you have problems with Layershifting I would first of all **disable**
`  #define STEALTHCHOP`
as I have different steppers I cant say anything about current. How have you set them before? calculated or "tuned" by ear? 
for testing: first do a M502 to resets your Eprom (it should report back its erased)
Than Do a M509 it should report the set current from your Config File. (638mA) <-- why 638?

Gaben5224, 2018-02-20:
@workheart 

So, 638mA because the motors are 900mA. RMS current * 1.41 = peak current, so 638 * 1.41 = 899,58. But it is possible that I misunderstand the comment in the config file. However I used to calculate the current to the A4988, but as I said I replaced all the motors and the drivers too.

about STEALTHCHOP:

I tried this, but it was very loud. May I sould to calibrate the frekvency somehow.

Sensorless:

I tried this... Doesn't work. I tried to set up the sensitivity t 0 too, but the sound of the motor was terrible. I thought that if anything else will work, than I will test this function again.

NeelD87, 2018-02-21:
How to find the right R_SENSE value

teemuatlut, 2018-02-21:
Either look at the board itself or check the schematic. [Here's Watterott TMC2130](https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC2130_v10.pdf). The sense resistors are the ones going from `BRB` and `BRA` to `GND` and are marked as `R2` and `R4`.

Gaben5224, 2018-02-22:
Hey Guys!

So, I think I know what causes the problem. (may I'm wrong...) The motors are 5.4V and 0.9A, but they get 12V and 0.9A from the motherboard. Could you think this will be the problem? If the answer is yes, can I modify the voltage with software (with pwm), or I should buy some new 12V motors?

Thanks!

ikarisan, 2018-02-22:
@Gaben5224 The voltage is not that important. The driver limits the current through the coils. And the current is what the motor makes heat up.

Gaben5224, 2018-02-22:
Okay. Then I really don't understand. The current is under 0.9A...

psavva, 2018-02-23:
You're running your motors with the wrong voltage, and you're running then close to their limits.

As per the datasheet, this motor can go up to 80deg max, and that's within 2 mins.  My suggestion would be to get 12v compatible motor, probably 1.7A to get a strong and cooler running setup...

S1CAR1US, 2018-02-24:
Also, which TMC2130s do you have?  Watterott or the shit YSETC.com ones....   I have seen folks reporting these to make for much hotter motors even when tuned to the same current

itnAAnti, 2018-02-24:
I have this same issue. I swapped my A4988s with TMC2130s, (and updated my Marlin config) and my steppers immediately got burning hot. It’s not a stepper or voltage issue - my power supply and steppers are unchanged. It must be something between the drivers and Marlin.

Other than the stepper temp, my printer is running great, very quiet and perfect printing.

Regarding sensorless homing, mine is working fine. I have the HOMING_SENSITIVITY set to 5 (on an i3). Make sure you have the diag pins from the 2130s connected to your homing switch input pins, it won’t work without those.

Gaben5224, 2018-02-24:
@S1CAR1US

I user the original watterot stepsticks. Today I will replace the IGUS bearings to the ordinary LM8UU on the X and Y axis  because they are a little bit looser, and I will try to decrease the current. And because I don't have any better idea I will put the A4998 back to the extruder. We will see what happens.

itnAAnti, 2018-02-24:
@Gaben5224 I'm curious to hear what happens when you put the A4988 back in. I might do that for my extruder stepper also if you have good results.

Gaben5224, 2018-02-24:
@itnAAnti 
I replaced the stepper driver, and now it's working with the A4988 driver. The temp of extruder motor was around 70°C, now this is around 45°C, so different is around 25°C.  So be honest, this is not a little difference. I measured the current and this is almost the same at the first and the second case too.

On the X and Y axis with the ordinary LM8UU bearings I could reducire the current to 500mA. Now I don't see any layer shifting or any other similar issue, but I'm not satisfied because it's noisy. - Not loud but compare to the IGUS bearings it's much louder. -

itnAAnti, 2018-02-26:
I ended up swapping in my A4988 on the extruder, like you did. It’s so much cooler it’s astonishing, though I didn’t take measurements. It’s good enough for now, but I hope someone figures this out!

thinkyhead, 2018-02-26:
In my testing of TMC2130s I have also encountered the fact that they get ridiculously hot. I presume you have all seen this thread…?

https://discuss.toms3d.org/hardware-f6/tmc2130-and-motor-overheating-t409.html

itnAAnti, 2018-02-27:
@thinkyhead 

Yep, I saw that thread, I'm in it! (Thanks though!)

The interesting part is my _drivers_ aren't overheating (at least not to the point of automatic current limiting, or any other side-effects, perhaps they are getting hotter than my A4988s, I haven't checked TBH), just my steppers are getting ridiculously hot. With A4988's, my steppers never get above slightly warm to the touch. With the 2130s, they are so hot they will burn you if you leave your finger on them for more than a quick touch. 

Is it possible that I have .9A steppers that work fine with A4988s, and overheat with 2130s, and changing to 1.7A steppers could help? I have an original Prusa i3 Mk2s, with an after-market board that supports stepper driver swaps.

lemmingDev, 2018-02-28:
Stealthchop mode makes by drivers heat up and give me temperature warnings

Spreadcycle mode keeps them cool and is still very quiet

Sensorless homing doesn't work right for me if I disable Stealthchop so I leave it enabled and set the HYBRID_THRESHOLD to 1 on all drivers to effectively disable Stealthchop and keep sensorless homing working

Try it out and let me know if it's working better

G3DBuilder, 2018-03-27:
5.4V Stepper Motors @ 900ma = 4.86 watts. to run the same steppers at 12 V you must reduce the current to 405ma or you will overheat the steppers. my steppers would run hot until I reduced the current to the proper level. (ohms Law)

adras, 2018-05-06:
I also had some overheat issues and step losses yesterday . I decided to lower the acceleration values, since I believe a fast accelleration needs a high current. I also lowered the motor current. Right now I'm running successful prints in Silentchop.

I'm using TMC2130 from Watterott.

Here are my new values:
Acceleration xyz lowered from 3000 to 300
Motor current xyze lowered from 800 to 600 mAh
Print speed is 30 perimeter, 40 infill, 60 travel.

Even with 600 mAh I don't have any step losses now. 

The drivers are still getting pretty hot (I'm measuring around 60-70°C), but no longer shut down due to overheating. I'll install a fan next week, then I'll crank them up, I still need to test how fast I can go with my printer. But for now, these settings work a treat.

I hope that helped

closed 2018-05-08 by thinkyhead

lakyluke98, 2019-05-02:
Hello everyone,

i have the same issue. I've build myself an AM8 Mgn out of my old Anet A8 nad also upgraded the electronics to an MKS gen 1.4 with 4 TMC2130s for X, Y, Z and E0. 

I never had any issues with the stepper drivers overheating. They are all equipped with heatsinks and a 120mm fan is blowing fresh air on them. I also never had any issues with layer skips or similar. I'm running in Stealthchop and have Sensorless Homing enabled on X and Y.

The X, Y and E0 Steppers get ridiculously hot. I can't touch them for more than a second. I'm using the 'normal' motor settings that Thomas Sanladerer also provided in his setup video:

R_SENSE = 0.11  
HOLD_MULTIPLIER = 0.2  
INTERPOLATE = true  

I run the drivers with 800mA of current with 16 microsteps. 
My acceleration settings are 
X = 2000 mm/s
Y = 2000 mm/s
Z = 100 mm/s 
E0 = 10000 mm/s

with an DEFAULT_ACCELERATION = 400 mm/s 
DEFAULT_RETRACT_ACCELERATION = 1000 mm/s and 
DEFAULT_TRAVEL_ACCELERATION = 1000 mm/s.

My Jerk settings are 
X and Y = 10 mm/s
Z = 0.3 mm/s and 
E0 = 5.0 mm/s

Do any of these settings seem off to you? As mentioned i don't have any issues with the printing quality of my printer but could quite possibly fry an egg on the steppers.

teemuatlut, 2019-05-02:
Coil current is what mostly affects the motor temperature.
You hand is also a very bad temperature measuring device. 60C or 80C is perfectly fine for the motor (perhaps not the motor mount) but will feel really hot on your skin.

lakyluke98, 2019-05-02:
Thanks teemuatlut!

Is there a way to lower the coil current?
I will measure the temperatures today and share the results.

teemuatlut, 2019-05-02:
http://marlinfw.org/docs/gcode/M906.html

lakyluke98, 2019-05-03:
So i measured around 60 °C on the stepper motors after like an hour of printing. Seems fine to me then

Tank you!

Blisk, 2020-07-04:
I have the same problem. But it is interesting, sometime it is OK sometime motors are getting so hot I can not touch it and it start to smell.