MarlinFirmware/Marlin #9710: [FR] Possible support for ATSAME70-XPLD-ND board.

Shawnb03, 2018-02-19:
Dev's, Please check out this board and tell me what you think. It has the same exact footprint as the Arduino Mega 2560 or the Arduino Due, and it also very inexpensive at 37.69 per board.

Here is the website,
https://www.digikey.com/products/en?mpart=ATSAME70-XPLD&v=150

If I am not allowed to post the website, I am sorry in advance. I am only posting it because this board could be the answer to a lot of our problems. This board has massive amount of power in it, along with having Ethernet, 300Mhz Processor, 2048k  Embedded Flash and Dual 16K Cache Memory.

No, I don't work for this company.

I just thought this would be a awesome board that I would like to see if it is a viable option. Thank you Dev's for all the work you have put into Marlin 2.0.

thinkyhead, 2018-02-28:
Thanks for pointing out this board and chipset. We don't generally support incomplete solutions until there is associated 3D printing hardware to go with it. If a board appears which is fully configured and has pin assignments to all the usual functions of 3D printing, we will consider supporting it.

Since we use PlatformIO we have a ready-made list of dozens of interesting and untested new chips to choose from, and we can generally build Marlin for any of them, as they all share common traits. But we need to have pin assignments and then we have to implement a HAL entry. But these are difficult and meticulous to develop, and we are already struggling with the set of boards we are targeting for 2.0 initial release. We would prefer to get the list we have already into working order and then clean up the mess and make it more organized and sane.

The board you point to will come into Marlin in the natural course of things as it gains in popularity, but we can't take on every new comer without yet having a targetable user and tester base. We prefer that people who have an interest in hardware integrate support into Marlin themselves and then bring us their fruits, as we are already overwhelmed with the load we have.

boelle, 2019-02-20:
@Shawnb03 i agree with thinky here, getting the board added should come in the form of a PR from either the maker of the board or the ones that wants it added

marlin dev's have enough on their plates

@thinkyhead maybe add the feature request label?

boelle, 2019-06-22:
missing label

sjasonsmith, 2020-11-10:
Based on the age of this request and lack of continuing activity I'm going to close this out.
We have a multitude of 32-bit boards available today from multiple 32-bit HALs. I agree with earlier comments that adopting new architectures would come in the form of somebody actually contributing it, rather than a feature request.

closed 2020-11-10 by sjasonsmith