MarlinFirmware/Marlin #9713: bed probe is ignored if autohome is started while probe is active

truglodite, 2018-02-19:
### Bug Report
- Description: ---
The title describes the problem. My anet a8 printer works great except for this bug. The printer is configured (stock A8 board) with 3 point leveling, bed pid, no z pullup (in circuit pullup), inverted z probe logic, and otherwise default a8 config files.
- Expected behaviour: ---
Normally when I hit autohome or bed level (and the probe is not active), the carriage finds x and y home, then it moves to the safe z-home position, and moves downward until the probe is activated (indicated by the red LED on the probe). It then reverses upward, comes down slower for a second probe, then either raises 10mm (autohome), or continues on with the bed leveling.
- Actual behaviour: ---
If I have the printer configured such that the X carriage is low enough to activate the Z probe (LED on), then I hit autohome, the nozzle raises up, homes X and Y, moves to the safe z-home location, then crashes into the bed despite the probe properly signaling to stop (LED on, also verified at the board with a meter). The same behavior occurs when bed leveling is started with the probe already active (bed leveling starts with the autohome routine, so hopefully one bug to fix).
- Steps to reproduce:
  - Anet A8 board with the latest Marlin 1.1.x bugfix, and a z probe (inverted logic using zmin plug)
  - Configure your x carriage such that the probe is low and activated
  - Hit autohome
  - Be ready to unplug!!!

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1735718/Configuration.zip)

thinkyhead, 2018-02-24:
Does the probe trigger at that distance, or does the probe show up as not-triggered when it's turned on close to the bed (because internally it's looking for state-change)? If it shows up as triggered, then I can devise a patch to check for a triggered probe when first starting a probe, and raise it until it is no longer triggered, plus a mm or 2.

In the meantime you can set `Z_HOMING_HEIGHT` to a safe distance (if you trust your Z won't be at the top very often) and that will make sure the probe isn't triggered when starting to home.

Note that since the Z isn't known at startup, it can be "dangerous" to raise the axis (or lower the bed) in all cases. That's why we make  `Z_HOMING_HEIGHT` optional and default 0.

thinkyhead, 2018-02-24:
Does the probe show up as **triggered** or **not-triggered** when it's turned on close to the bed? If it showed up as **triggered**, then I could devise a patch to check for a triggered probe when first starting a probe, and raise it until it is no longer triggered, plus a mm or 2.

However, your probe is **not-triggered** when turned on close to the bed, because it's looking for a change in capacitance or inductance, and it doesn't get such a change. This is obvious, because if it _was_ triggered then it would stop moving right away.

Since it **_isn't showing up as triggered_** you have to use `Z_HOMING_HEIGHT`.

By setting `Z_HOMING_HEIGHT` to a safe distance (if you trust your Z won't be at the top very often) you can be sure the probe won't be triggered when you're starting to home.

Note that since the Z isn't known at startup, it can be "dangerous" to raise the axis (or lower the bed) in all cases. That's why we make  `Z_HOMING_HEIGHT` optional and default 0.

thinkyhead, 2018-02-24:
I see you already have `Z_HOMING_HEIGHT` set to 10, which is the same as your `Z_CLEARANCE_DEPLOY_PROBE` and higher than your `Z_CLEARANCE_BETWEEN_PROBES`, so you should already be getting enough height from homing moves and when the probe is deployed for leveling too.

I also see that your `Z_PROBE_OFFSET_FROM_EXTRUDER` is set to 0. Did you adjust this to a more accurate value using `M851 Z` or have you actually set up your probe to trigger when the nozzle is touching the bed?

truglodite, 2018-02-25:
thinkyhead, thank you very much for sharing your time by posting the replys. Fortunately I discovered it is was not a bug in the software, just an intermittent probe wire (*sigh*). Not that it matters now, but yeah I mechanically adjust my z-probe so the nozzle touches the bed.
Thanks again, Kevin

closed 2018-02-25 by truglodite

thinkyhead, 2018-02-25:
No problemo. Sometimes all you need is someone to bounce things off till the dawn arrives.