MarlinFirmware/Marlin #9715: Marlin 1.1.8 issue with G28 before G29 with Auto Bed levelling

simonboydfoley, 2018-02-19:
------------------------------------------------------------
### Bug Report
- Description: -- All the guides state you need to G28 home before you set your Z offset calibration. However if you have a printer that homes at the corner of the build plate (like the CR10's) ... issuing a G28 after flashing 1.1.8 will grind the hotend through the buildplate because devices like the BLTouch replace the z end stop and the sensor could be outside the build plate and can't activate! 
The home location is only reset/overridden by a G29 which needs to be done 1st after installing.
- Expected behaviour: -The code needs to either check that a G29 has been done before a G28 is allowed and block it and prompt the user to do a G29. Or better still because the code knows the Bed size and probe offsets ... it can change the home location without needing a G29 in the 1st place.
Indeed as we know the probe offset and bed size  .... why does the user have to define the left / right / front / back probe bed limits ? This can all be worked with some simple maths ... and is rendered confusing / irrelevant by the 1.1.8 auto bed levelling code as it now seems to be doing this anyway because it compensates and Centers on the probe not the nozzle when probing anyway. Even if you wanted user overrides .... 4 parameters can be replaced by a simple PROBE_MARGIN_BOUNDARY_DEPTH
- Actual behaviour: --- allows users to break their printers by issuing a G28 before a G29 after initial flash.
- Steps to reproduce:
  - Flash with new 1.1.8 marlin with a probe set to a negative x offset and a home location that puts the probe outside the bed. 

  - issue a G28 before you do a G29

locutus40, 2018-02-19:
im confused? i g28 then g29.... i uploaded the latest and it crashed my boot loader somehow i had to use another arduino to reload the bootloader on the mega.. i uploaded .18 when i put it back tomorrow ill see if i have the same issues.. when i g29 the probe would shoot behind the bed far beyond the bed.. might be a bad mega failing but this sounds similar

adamnock, 2018-02-19:
Sounds like you need to be enabling Z_SAFE_HOMING. Thats what its there to prevent.

locutus40, 2018-02-19:
yea i already did that before it corrupted my boot program...

ive had this arduino for about 4 years now its probably failing... so 2 nights ago when i uploaded.18 even with safe homing enabled  it  probed outside of the bed.. then i said hmm let me see if its the actual issue with the code.. so i re uploaded and poof failed code  500 boot loader error.. then the arduino seized up.... and stopped working i had to reinstall the boot loader and it finally took the code..

so im not sure if its the code.. or the arduino thats failing.. .. i dont have another mega to test it.. so ill put it back together tomorrow aka today since its 1am.. and go from there if it still does it.. its probably the code.... i cant really confirm it till i buy another mega and i need to earn some money first.. so might be a week or so.. but i saw your post and was like what the that sounds soo familiar... so if it does the same thing yours does again... ill just scrap the mega i have and buy a new one if it does it with that one its the code....,... maybe failing arduinos act funny.. like this i dunnoo.. since everything is made in china now.. its funny i have an official mega too

simonboydfoley, 2018-02-19:
Z_SAFE_HOMING is enabled ... but it dosnt  prevent people from issuing autohome from the Marlin Firmware menu which smash's the nozzle into the bed.  Incidentally  Z_SAFE_HOMING dose not seem to  work either as it forbids G28 Z when the probe is in the Center of the bed and within the probe area for the build plate.... butbi need Hondo more testing tonight

simonboydfoley, 2018-02-19:
FYI  I also want to double check to see if the fix once implement Is Persistant in EEPROM  as in my use case I came from stock Cr10s firmware that did not use/expose eeprom to 1.1.8 that did initiate EEPROM. If I can't recreate the problem I'll need to find a way of clearing down the eeprom to a default Cr10s state. I know that setting the z offset in EEPROM did not fix the issue between flash's ( I put the old z end stop Microswitch back on the melzi v1 cr10s mega 2650 board and disabled Z_SAFE_HOME so I could see if the problem was not having an accurate z offset .... but it did not stop the nozzle heatbed smash).

simonboydfoley, 2018-02-19:
I need to look at the code and find out what the g29 (ABL issued from the menu) Actually does to override the default home position of 0,0,0

simonboydfoley, 2018-02-19:
Crap .... I've done some 2hr more testing  and i could not re-create the problem  with the original stock firmware ....... then.i reversed the polarity on the End stop connextor to the BL Touch  from Z min .... bingo  .... I had exactly the same problem.

When I was double checking the cabling after all the voltage's checked out .... I replaced the JST Connector on my end stop extension cable. I don't remember making a note of the polarity it was originally set to.  So I can only assume in had  it reversed on the JST Connector . On  Microswitch polarity dose not matter but on the BL Touch it does ...... 

My bad .. apologies for being a dick

closed 2018-02-19 by simonboydfoley