MarlinFirmware/Marlin #9716: Latest 2.0.x (2-19-2018) compilation error due to new undocumented u8glib requirement.

SS396, 2018-02-19:
------------------------------------------------------------
### Compile Error
When I compile the latest bugfix 2.0.x, I get an error:
```
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\TMP\sketch\src\lcd\dogm\u8g_dev_st7920_128x64_HAL.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\TMP\sketch\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega" "C:\TMP\sketch\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp" -o "C:\TMP\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\TMP\sketch\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:27:20: fatal error: U8glib.h: No such file or directory

 #include <U8glib.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

Why does my build now require u8glib to be installed?  It never required it previously, and the printer has a http://reprap.org/wiki/RepRapDiscount_Smart_Controller not a full graphic one.  

Pretty confident the error is related to https://github.com/MarlinFirmware/Marlin/pull/9624 dated 5 days ago as I never had any issues building previous versions and even a 2.0.x version I pulled last week.

https://github.com/MarlinFirmware/Marlin/commit/9a423e9a891381b41ca31f0dd04b39002f99c7e2#diff-a3acb5fe7daf75c60548289a5afd49b7L43

--proposed fix--
"#include <U8glib.h>" needs to be moved back to line 43 or at least underneath the "#if ENABLED(U8GLIB_ST7920)" at line 29, and removed from line 27 in Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp.

After I've made the proposed changes and put the include back to line 43 the build succeeds without u8glib installed as it did previously.   However it will succeed if I install the u8glib library, but I don't feel that is overall the best solution for the community and the future.   The u8glib library should only be required to be installed if you actually have the full graphic display.   The main issue I'm trying to prevent in the future is that nowhere in the documentation I used to build marlin for this specific printer is it mentioned that u8glib is required.  So if the code remains the way it is right now, some new user simply blindly following the instructions they found will fail with a compilation error as documented above and they will be stuck not understanding why it won't build or even how to fix it.  So all of the various build instructions for non full graphic controllers all across the internet are essentially broken and have to be modified/updated to install the u8glib library before attempting to build.  The issues area is going to be overrun with frustrated people posting the same error as I did, because the instructions they are following are leaving out a very critical step.  If you put it back to the way it was, that critical step is no longer needed.

diaconuliviu85, 2018-02-19:
I have the same problem.

Bob-the-Kuhn, 2018-02-19:
I can't reproduce the problem.

Please ZIP & post your Configuration.h and Configuration_adv.h files and tell me what compiler you're using (Arduino 1.8.5, PlatformIO, ...).

AnHardt, 2018-02-19:
Bob.
I assume you have u8glib installed.

The problem is pretty obvious.
`ultralcd_st7920_u8glib_rrd_AVR.cpp` isn't an ".h" fila anymore - so comiled always.
At the current position, before the `#if ENABLED(U8GLIB_ST7920)` u8glib is alway required.

Bob-the-Kuhn, 2018-02-19:
Obvious?  I'm really good at missing the obvious!  ☹️

OK - so this happens when using Arduino, no LCD enabled and U8G not installed in Arduino.  

I just looked at PR #9624.  Moving **#include <U8glib.h>** was not an intended change.

I'll have the fix merged in an hour or so.

closed 2018-02-20 by thinkyhead