MarlinFirmware/Marlin #9718: CoreXY reverting to regular cartesian?

gregspruce, 2018-02-19:
I've had a very strange bug(?) occur to me twice now, both times using bugfix 1.1.x, once from an october build and then once from a build from just 2-3 weeks ago.

My CoreYX printer seems to suddenly think it is a regular cartesian setup. This last time, I saw it happen during the purge line at the beginning of a print. Halfway through a G1 X160, one motor stopped and the "X" motor kept going, which sent the carriage off the front of the bed and against the front frame. I rushed to reset with the button on the board. From that point forward, all movements do NOT use CoreYX kinematics. Instead, for Y axis moves, the left motor runs and for X axis moves, the right motor runs - just as if it were a normal cartesian setup.

The last time this happened, I was looking for a reason to re-flash and enable UBL anyway, since I was getting the bilinear z-offset issue so many people seem to be having. Loading a newly compiled version made the printer function as configured. I haven't yet reflashed after last night's incident.

Azteeg X3 pro, Viki2 screen, UBL 7x7 via BLTouch, .9deg steppers, Titan extruders on E3d chimera, CoreYX inspired by Hypercube Evolution. I attached all the files I have to modify for my setup: configuration.h, configuration_adv.h, pins_ramps.h, and pins_Azteeg_X3_pro.h
[bugfix_1.1.x_modified_files.zip](https://github.com/MarlinFirmware/Marlin/files/1736672/bugfix_1.1.x_modified_files.zip)

thinkyhead, 2018-02-19:
CoreXY is baked into the low-level stepping routines, so it's not possible for it to stop doing CoreXY kinematics in response to line segments. Linear moves always distribute steps to the two motors according to core kinematics.

To get a better idea of what's going on, enable the `M114_DETAIL` option and re-flash, preferably with the current `bugfix-1.1.x` branch. During good and normal motion, do `M114` at various points around the bed and save the results in a text file for later reference. When/if the bad behavior reappears, do `M114` at various points (even though the steppers may grind and not move the nozzle to the right place) and add those results to the text file.

We can then compare those results to see if something is corrupting positions at the high or low level.

Also, when the issue appears, check to see if the steppers or their drivers are heating up more than usual. Thermal overload can cause stepper drivers to shut down, though usually this only lasts a fraction of a second before they recover.

thinkyhead, 2018-02-19:
> bilinear z-offset issue so many people seem to be having

All testing so far has narrowed this down to gantry misalignment throwing off probe measurements. You can make a fake mesh by setting the points yourself and test whether the Z axis is properly compensating. You can also use `PROBE_MANUALLY` and see if the resulting mesh differs from the automatic probe results. If the mesh differs then it is a clear indication of a probe issue.

gregspruce, 2018-02-19:
Thanks for the responses.
For the ABL issues, I switched to UBL and now have perfect leveling. Side question since I'll be reflashing again anyway: is it still recommended to disable segmented moves for UBL?

For this weirdness I'm experiencing now, I'll do the M114 logging when I reflash and see if it happens again down the road. I agree, it makes zero sense for it to "stop" doing CoreYX kinematics, but that is the best description for the behavior I am watching. Steppers and drivers were not noticeably warm, but I had been printing TPU all day - I'll see how it's working tonight before I reflash.

thinkyhead, 2018-02-25:
> is it still recommended to disable segmented moves for UBL?

It is forced to be off now. As we continue to develop and fix stuff, it may be worth revisiting.