MarlinFirmware/Marlin #9719: [FR]dual z rumba don't work[FR]

amirgaziel0, 2018-02-19:
Hello everyone
I tried everything
I do not know what to do anymore
I'm adding photos of settings I've set up
My board is RUMBA BIQU

1. my board configuration.h
[IMG]http://www.interload.co.il/upload/5673096.jpg[/IMG]
2. configuration_avd.h
[IMG]http://www.interload.co.il/upload/4422559.jpg[/IMG]
3. pins_RUMBA.H
[IMG]http://www.interload.co.il/upload/908096.jpg[/IMG]

  
please help me
Thank you

revilor, 2018-02-23:
The image links are broken. But please don't replace the images, attach your configuration files instead.

thinkyhead, 2018-02-24:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

thinkyhead, 2018-02-24:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We'll check them over and see if anything is amiss.

amirgaziel0, 2018-02-24:
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/1754692/conf.zip)

thinkyhead, 2018-02-27:
We have a form that needs to be filled out for issues so that we can understand your problem. If you don't want to fill out the form then we can't help you.

closed 2018-02-27 by thinkyhead